SmingHub/Sming #1354: Sming 3.5.1 rBoot problem

kwis2, 2018-04-02:
I have tried to run Basic_rBoot example and I have infinite wdt reset loop. There is problem with Sming or I make sth wrong. I don't undrstand what hapened - why there are mode : sta + softAP ? and why serial not work on 115200 baudrate. Finally why there is infinite wdt loop?

Sming 3.5.1 (201a746...) with `make SMING_RELEASE=1` , ESP8266-01 with 1MB memory
To run example I have modified:
copied `rboot.rom0.ld` into `rom0.ld` and `rom1.ld`
in Makefile-rboot.mk (most safe I think)

> DISABLE_SPIFFS = 1
> RBOOT_ENABLED ?= 1
> RBOOT_BIG_FLASH ?= 0
> RBOOT_TWO_ROMS  ?= 1
> SPI_SIZE        ?= 1M
> RBOOT_ROM_0     ?= rom0
> RBOOT_LD_0      ?= rom0.ld
> RBOOT_ROM_1     ?= rom1
> RBOOT_LD_1      ?= rom1.ld
> RBOOT_SPIFFS_0  ?= 0xf0000
> RBOOT_SPIFFS_1  ?= 0xf0000

and my application modification :

```
    Serial.begin(SERIAL_BAUD_RATE); // 115200 by default
    Serial.systemDebugOutput(true); // Debug output to serial

    WifiAccessPoint.enable(false);

    WifiStation.enable(true, true);
    WifiStation.config(WIFI_SSID, WIFI_PWD); // Put you SSID and Password here
    WifiStation.connect();

    Serial.println("\n Test \n"); 
    Serial.println();

```

With terminal I got  with baudrate 74880 not 115200 :

>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Writing default boot config.
> Booting rom 0.
> rf_cal[0] !=0x05,is 0xFF
> rf cal sector: 251
> rf[112] : ff
> rf[113] : ff
> rf[114] : ff
> w_flash
> 
> SD
>  Test 
> 
> 
> mode : sta(a0:20:a6:06:b8:d7) + softAP(a2:20:a6:06:b8:d7)
> add if0
> add if1
> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
> Fatal exception 28(LoadProhibitedCause):
> epc1=0x40101bf6, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000004, depc=0x00000000
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> rf_cal[0] !=0x05,is 0xFF
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> rf_cal[0] !=0x05,is 0xFF
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> rf_cal[0] !=0x05,is 0xFF
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> rf_cal[0] !=0x05,is 0xFF
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> rf_cal[0] !=0x05,is 0xFF
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
>

avr39-ripe, 2018-04-02:
may be 'make flashinit' helps?

kwis2, 2018-04-02:
@avr39-ripe thx
flashinit helps with wdt, but still there is mode : sta + softAP 

after reset at 74880 i got 
> ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 0.
> rf cal sector: 251
> rf[11Ę[03]1®[0C]
> 

then on 115200

> Test 
> 
> 
> mode : sta(a0:20:a6:06:b8:d7) + softAP(a2:20:a6:06:b8:d7)
> add if0
> add if1
> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
> bcn 100
> scandone
> state: 0 -> 2 (b0)
> state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 5
> cnt 
> 
> connected with xxxxxxxxx, channel 4
> dhcp client start...
> ip:192.168.0.59,mask:255.255.255.0,gw:192.168.0.1
>

kwis2, 2018-04-02:
with code revert to initial version of Basic_rBoot, except `DISABLE_SPIFFS = 1`
I got:
with work from ROM 0 it looks OK with ota I got infinite Fatal exception. I don't know reason, there is no information if it was problem with .bin or so?

> info
> 
> SDK: v2.0.0(5a875ba)
> Free Heap: 44752
> CPU Frequency: 80 MHz
> System Chip ID: 6b8d7
> SPI Flash ID: 1440e0
> Count: 2
> ROM 0: 8192
> ROM 1: 532480
> ROM 2: 0
> GPIO ROM: 0
> 
and with ota:

> ota
> Updating...
> In callback...
> Firmware updated, rebooting to rom 1...
> state: 5 -> 0 (0)
> rm 0
> del if0
> bcn 0
> del if1
> usl
> Fatal exception 28(LoadProhibitedCause):
> epc1=0x4023ac1b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 1.
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> Fatal exception (0): 
> epc1=0x402032dc, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
>

avr39-ripe, 2018-04-02:
try to enable SPIFFS I think SPIFFS is mandatory for this example

kwis2, 2018-04-03:
I have tested several times with different settings without success.
Rom0 works, but there is no way to switch Rom1. Additionally, after flashing chip there is no way to run any software on module without `flashinit`.

my chip ESP8266-01S,  last config
> SPIFF_FILES = files
> RBOOT_ENABLED ?= 1
> RBOOT_BIG_FLASH ?= 0
> RBOOT_TWO_ROMS  ?= 1
> SPI_SIZE        ?= 1M
> RBOOT_ROM_0     ?= rom0
> RBOOT_LD_0      ?= rom0.ld
> RBOOT_ROM_1     ?= rom1
> RBOOT_LD_1      ?= rom1.ld
> SPIFF_SIZE      ?= 65536
> RBOOT_SPIFFS_0  ?= 0x70000
> RBOOT_SPIFFS_1  ?= 0xf0000

after flash I got:

> trying to mount spiffs at 0x00070000, length 65536
> 
> Currently running rom 0.
> Type 'help' and press enter for instructions.
> 
> mode : sta(a0:20:a6:06:b8:d7) + softAP(a2:20:a6:06:b8:d7)
> add if0
> add if1
> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
> bcn 100
> scandone
> state: 0 -> 2 (b0)
> state: 2 -> 3 (0)
> state: 3 -> 5 (10)
> add 0
> aid 4
> cnt 
> 
> connected with xxxxxxxxx, channel 4
> dhcp client start...
> ip:192.168.0.59,mask:255.255.255.0,gw:192.168.0.1
> 
> ls
> filecount 1
> testfile.txt
> info
> 
> SDK: v2.0.0(5a875ba)
> Free Heap: 42560
> CPU Frequency: 80 MHz
> System Chip ID: 6b8d7
> SPI Flash ID: 1440e0
> Count: 2
> ROM 0: 8192
> ROM 1: 532480
> ROM 2: 0
> GPIO ROM: 0
> ota
> Updating...
> In callback...
> Firmware updated, rebooting to rom 1...
> state: 5 -> 0 (0)
> rm 0
> del if0
> bcn 0
> del if1
> usl
> Fatal exception 28(LoadProhibitedCause):
> epc1=0x4023b687, epc2=0x00000000, epc3=0x00000000, excvaddr=0x0000007c, depc=0x00000000
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 1.
> 
then at 74880 baudrate infinite reset loop

>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> tail 8
> chksum 0xf2
> csum 0xf2
> 
> rBoot v1.4.2 - richardaburton@gmail.com
> Flash Size:   8 Mbit
> Flash Mode:   QIO
> Flash Speed:  40 MHz
> 
> Booting rom 1.
> rf_cal[0] !=0x05,is 0xFF
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 1320, room 16 
> tail 8
> chksum 0x95
> load 0x3ffe8000, len 632, room 0 
> 
>

kwis2, 2018-04-26:
After long time with Sming@Windows 7 I decided to try Sming@Linux on VM. Basic_rBoot sample worked from first run @avr39-ripe example works without SPIFFS too. Problem was with choco Sming install on Windows.