SmingHub/Sming #1355: Sming 3.5.1 problem with OS identification

kwis2, 2018-04-03:
After update to Sming 3.5.1 on same Windows machines (manual update because chocolatey still don't work) there is problem with compilation. I tried to compile Basic_rBoot example and got

> CC /C/tools/sming/Sming/third-party/rboot/appcode/rboot-api.c
> c:/tools/sming/Sming/third-party/rboot/appcode/rboot-api.c:10:21: fatal error: c_types.h: No such file or directory
> compilation terminated.
> make.exe ": *** [out/build//C/tools/sming/Sming/third-party/rboot/appcode/rboot-api.o] Error 1

I found it is problem with Cygwin identificztion, after I change Makefile-rboot.mk line 114, after 

> else ifneq ($(filter CYGWIN%,$(UNAME)),)
>  Cygwin Detected

from

> UNAME := Linux

to 

> UNAME := Windows

Sming work OK

slaff, 2018-04-09:
@riban-bw Can you look at that issue?

kwis2, 2018-04-26:
System it was win 7 32 bit with not working choco install as in #1297, #1241 and #1340 
In my oppinion with manual install of mingw afrer choco unzip fail there were no MINGW uname registered, but in Windows there was Cygwin registered.
So in my opinion Cygwin is not the best way to identify OS

closed 2021-11-16 by slaff