SmingHub/Sming #1357: Already Freed

frankdownunder, 2018-04-04:
Calling downloadFile gives debug message "already freed"

Here is a small source file that demonstrates the problem
```
#include <user_config.h>
#include <SmingCore/SmingCore.h>
#include <memory>

HttpClient downloadClient;

void onTimer_checkDownloadNewFirmware();
Timer procTimerCheckDownloadNewFirmware;

void onGotIP(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	debugf("got ip");
}

void init()
{
	Serial.begin(921600);//Serial.begin(74880);		//Serial.begin(115200);	// 115200 by default
	Serial.systemDebugOutput(true);
	WifiEvents.onStationGotIP(onGotIP);
	debugf("");
	debugf("system_get_sdk_version %s", system_get_sdk_version());
	int currentRom = rboot_get_current_rom(); 
	if (currentRom == 0) {
		debugf("mount spiffs at 0x%08x, length %d", RBOOT_SPIFFS_0, SPIFF_SIZE);
		spiffs_mount_manual(RBOOT_SPIFFS_0, SPIFF_SIZE);
	}
	else {
		debugf("mount spiffs at 0x%08x, length %d", RBOOT_SPIFFS_1, SPIFF_SIZE);
		spiffs_mount_manual(RBOOT_SPIFFS_1, SPIFF_SIZE);
	}
	procTimerCheckDownloadNewFirmware.initializeMs(1 * 4 * 1000, onTimer_checkDownloadNewFirmware).start();
}

int theRequestCompletedDelegate(HttpConnection& connection, bool success)
{
	debugf("end of downloadFile");
	return 0;
}

void onTimer_checkDownloadNewFirmware()
{
	downloadClient.downloadFile("http://clarkbaycottages.com.au/buildVersion.json", theRequestCompletedDelegate);
}

/*
ÇÔÉÇÔÉÇÔÉÇÔÉÇ
system_get_sdk_version 2.0.0(5a875ba)
mount spiffs at 0x00100000, length 655360
fs.start: size:640 Kb, offset:0x100000
mount res: 0
mode : sta(60:01:94:19:2d:23) + softAP(62:01:94:19:2d:23)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt
connected with clarkbayfamily, channel 1
dhcp client start...
connect to ssid clarkbayfamily, channel 1
ip:192.168.100.100,mask:255.255.255.0,gw:192.168.100.1
ip:192.168.100.100,mask:255.255.255.0,gw:192.168.100.1
got ip
end of downloadFile
3fff13d8 already freed					<<<<<<<<<<<<<<<<<<
end of downloadFile
3fff1408 already freed					<<<<<<<<<<<<<<<<<<
--- exit ---
*/
```

frankdownunder, 2018-04-09:
@slaff Good news and bad.  Good news is the short sample I sent you now works fine without the pesky already freed messsage.  Bad news is that when I run my code it is still there.  So I put the relevant code back into theRequestCompletedDelegate and the short sample still works - phooey!
That leaves me with an "already freed" that is not easy to repro. 
Worse still, I no longer care because I realised that downloadFile was absolutely the wrong technique for what I was trying to achieve; I could see that it wrote to spiffs, not something I needed to do. All I needed was downloadString, and after making those changes the message is gone.

As part of this exercise I looked at every delete statement in sming.
1. This code looks suspicious, it lacks an assignment to nullptr:
```
int HttpServerConnection::staticOnMessageComplete(http_parser* parser)
{
	. . .
	if(connection->request.responseStream != NULL) {
		connection->request.responseStream->close();
		delete connection->request.responseStream;
	}
```
2. This code looks like a potential memory leak ( can exit function without deleting "look"), prompting me to ask is it time to start using RAII techniques ( smart pointers) to prevent that happening?:
```
bool TcpConnection::connect(String server, int port, bool useSsl /* = false */, uint32_t sslOptions /* = 0 */)
{
	. . .
	DnsLookup *look = new DnsLookup { this, port };
	err_t dnslook = dns_gethostbyname(server.c_str(), &addr, staticDnsResponse, look);
	if (dnslook != ERR_OK)
	{
		if (dnslook == ERR_INPROGRESS)
			return true;
```

slaff, 2018-04-10:
@frankdownunder  Thanks for the efforts. Point 2 is not an issue, point 1 will be fixed with the PR 1359.

PR #1359 (2018-04-10): Prevent "already-freed" error from happening.
Fixes #1357.
commit 6f3bf3e6 (2018-04-10): Prevent "already-freed" error from happening. (#1359)

Fixes #1357.
closed 2018-04-10 by slaff