SmingHub/Sming #1358: WiFi station periodic hangs

kwis2, 2018-04-06:
I have been testing Sming for months and sometimes there are problems with module.
Module ESP8266-01S, Sming version about 3.1, battery operated,  connected with router which is often reset by power loss. The only ESP work is to HTTP data send. 
For about 4 months there were hundreds of router resets. Computer connected to router always returned to connected state. Often module return to connected state without any problems, but there were two times when module failed - just hangs. First time I manualy reset module and it work OK, second time I reset router and ESP start to work. 
I think there should be any state in connection procedure when ESP thinks it is connected and is unable to send data. I am wondering if there is way to detect such state (except watch server response what is obvious, but very trafic demanding) and automatic reconnect. 
In my opinion it is some kind of bug, but it is very hard to debug because of rarely occurs. First occurence was in January, and the second two days ago.

slaff, 2018-04-11:
@kwis2 Without any error logs of information about the segfaults there is very little that we can do. You can  try and create your own crash handler that records a crash so that it can be analyzed later.

kwis2, 2018-04-26:
After next two weeks there was another hangs with Sming 3.5 - hangs of communication with internet, not module. It looks like from Sming point of view there is no need to reconnect, but http put do not return response so it looks like there were no router to internet connection, in meantime on another computer there is fully working internet connection. It looks like there was router connection state which Sming isn't able to identify (my guess).
In my opinion it is very difficult problem to identify and it happened only with one router. 
Problem is that there is no crash information. ESP module works OK, From it's response it looks that there is connection with router. I made external "WDT" circuit which resets ESP module and after reset module works OK