SmingHub/Sming #1366: Question regarding SSL cipher

icodk, 2018-04-21:
nginx default configuration(mine anyway) comes with the following line:
`ssl_ciphers ECDHE-RSA-AES256-SHA:DHE-RSA-AES256-SHA:DHE-DSS-AES256-SHA:DHE-RSA-AES128-SHA:DHE-DSS-AES128-SHA;
`
However the Basic_Ssl sample failed to connect unless I change it to:
` ssl_ciphers HIGH:!aNULL:!eNULL:!EXPORT:!CAMELLIA:!DES:!MD5:!PSK:!RC4;`
Which might be less secure
Any best practice on that issue ?
Thanks

slaff, 2018-11-19:
The SSL library that we use, axTLS has pretty limited, but well tested and fast, set of ciphers.  I am thinking about testing BearSSL and see how it behaves.

kpishere, 2019-12-21:
@slaf I see there is activity in making generic SSL library.  Paving the way for bearssl?  I really want 16k blocks for my project in esp8266 ... was there any success with using bearssl directly with rest of Sming framework?

slaff, 2019-12-31:
@icodk @kpishere Take a look at the latest state of PR #1999. With the latest changes it is possible to use `Axtls` and `Bearssl` for client and server side. You can compile Basic_Ssl and HttpServer_ConfigNetwork samples with ENABLE_SSL=BearSSL.

closed 2021-11-16 by slaff