SmingHub/Sming #1367: Questions: 1. Getting remote IP and 2. setting DNS when using lwip

icodk, 2018-04-21:
Http Client can use a URL that resolves to an IP in the stack. How can print this IP ?
How can I get remote IP when a client connect to the web server running onthe esp ?
Ho can I set the DNS server if not using DHCP ?
All these  be done with espconn  but how to do it with request/responce and Http Client ?
Thanks

slaff, 2018-04-24:
> Remote IP

The remote IP can be taken from LWIPs tcp. Example: tcp->ip->remote_ip. This is not directly accessible from one of the TcpClients/Servers but we can easily add it.  

> DNS

Use dns_setserver
`void           dns_setserver(u8_t numdns, ip_addr_t *dnsserver);`

Example:
```
#include "lwip/dns.h"

ip_addr_t dnsServerIp;
IP_ADDR4( &dnsServerIp, "1.1.1.1");
dns_setserver(0, &dnsServerIp);
```

Please, try to look at LWIP for the available options.

icodk, 2018-04-24:
@slaff Thanks
Yes please. Could be nice to have access to remote IP in both request  objects.
request remote_ip could be useful in filtering clients when used in a web server and remote IP of request or connection( as in onRequestComplete callback) in  case of httpClient to debug correct DNS resolution.
Why not to allow access to the complete tcp connection: I could also think of being able to control DNS cache or TTL (to force new DNS resoluton at will).
Hope this extra freedom is not violating the efforts to protect us from our own mistakes:-)

PR #1370 (2018-05-02): Exposed the remote port and ip for the connections.
Closes: #1367
commit 34fdcff6 (2018-05-02): Exposed the remote port and ip for the connections. (#1370)

Closes: #1367
closed 2018-05-02 by slaff