SmingHub/Sming #1372: Nextion display support

vijaygill, 2018-05-03:
Hi,
Thanks for such a great framework!
I've got this Nextion Display and I have managed to implement a few functions to use it (clear screen, show some text etc). I would like to contribute to Sming and Nextion displays could be a good option. 

Currently I am caught between writing own library from scratch or patching ITead's arduino library (using the patch mechanism in Sming).

Please share your opinions (even coding standards etc for Sming).

Thanks
Vijay

slaff, 2018-05-03:
> or patching ITead's arduino library (using the patch mechanism in Sming).

@vijaygill I would prefer the second option. Unless you can write a new library from scratch that has something to offer like: amazing speed  (much faster than the current ones), uses less resources (CPU and memory) etc.

>  ... (even coding standards etc for Sming).

That reminds of a task in my TODO list to force a coding standard using vera++ in our CI system.

vijaygill, 2018-05-03:
@slaff thanks! I will have another attempt at it. I tried once last weekend but gave up after about 15 minutes. I do not want to re-invent the wheel either. The ITead's library uses functions like "utoa" which can be patched to use "itoa". I will report my progress and ask more questions as I go along.

slaff, 2018-05-03:
Using Nextion was done before. I remember this fork here: https://github.com/ADiea/Sming/tree/work/Sming/Libraries/Nextion. @ADiea never sent his changes as PR but you can see the changes that he has made and submit a Nextion library to Sming.

vijaygill, 2018-05-10:
@slaff I have got initial compile successful (work is far from over though) with ITEAD's library added as module and a patch generated (applied using build mechanism of sming). So far all's fine. Now I need a bit more information on using second UART on WEMOS D1 I have here. does sming allow usage of both UARTs at the same time? The standard one for all debug / print messages and second one for communicating with Nextion display.

slaff, 2018-05-10:
You can send all debug information to UART1 and use the standard UART0 for bi-directional communication. What you need to add somewhere in the beginning of your init function is

```
HardwareSerial Serial1(UART1);
Serial1.begin(SERIAL_BAUD_RATE);
Serial1.systemDebugOutput(true);  // after this command all debug code will be printed to UART1

debugf("This should show up in UART1 only"); // that is not needed in your code.
``` 

A complete sample can be seen from [samples/Basic_Serial/app/application.cpp#L126](https://github.com/SmingHub/Sming/blob/develop/samples/Basic_Serial/app/application.cpp#L126)

There is something to be aware of - the boot messages will all be send using UART0.

vijaygill, 2018-05-10:
Thanks @slaff . I tried something like that but could not make it work. I was trying the other way round, using UART1 for Nextion and leave the UART0 as it is. I will try your suggestion and let you know.

slaff, 2018-05-10:
>  I was trying the other way round, using UART1 for Nextion and leave the UART0 as it is.

That will not work. Use UART0(Serial) for communication with Nextion and UART1(Serial1) for debugging.

vijaygill, 2018-05-10:
@slaff thanks again. Will follow your suggestion. Can you please share more details on why it won't work? I am learning about ESP8266 also as I go along.

slaff, 2018-05-10:
>  Can you please share more details on why it won't work?

Because the hardware does not support it :) The comments in the samples are there to help you. Make sure to read them. [samples/Basic_Serial/app/application.cpp](https://github.com/SmingHub/Sming/blob/develop/samples/Basic_Serial/app/application.cpp#L116)

vijaygill, 2018-05-10:
@slaff - thanks again! Silly me, should have looked at the code myself! I am at work and was going through github during lunch time :). I will try again when I go home in the evening.

ADiea, 2018-05-13:
@vijaygill uart1 we only have TX on esp8266 as far as i remember, you will need uart0 which has both rx and tx to communicate to nextion device. Unfortunately I abandoned my thermostat project that was going to use the nextion and have no further progress.

vijaygill, 2018-05-14:
@ADiea - Thanks for your post. I will remember that explanation of UARTs by you and @slaff. I didn't get a chance to work on the code this weekend. Will resume it in coming days of this week.

PR #1389 (2018-05-25): Add support for Nextion displays.
Fixes #1372.

Hi @slaff,
Please review the PR and let me know.
About the proof, do I need to post a video of the example (added to samples - Nextion_Button) somewhere?

Regards
Vijay
closed 2018-05-25 by slaff