SmingHub/Sming #1374: how to access mqtt.server private from outside?

dpechman, 2018-05-06:
I want to set from a configuration file the variables mqtt host and port and set MqttClient host and port in runtime but I dont figure out how to.
The MqttClient instance have to be created before it is ok to read configuration file, so I initialise it empty to then set it later.
I made a workaround changing the declaration statement from private to public in MqttClient.h where the variables are declared but its not the right way. 
```
/****
 * Sming Framework Project - Open Source framework for high efficiency native ESP8266 development.
 * Created 2015 by Skurydin Alexey
 * http://github.com/anakod/Sming
 * All files of the Sming Core are provided under the LGPL v3 license.
 ****/

/** @defgroup   mqttclient MQTT client
 *  @brief      Provides MQTT client
 *  @ingroup    tcpclient
 *  @{
 */

#ifndef _SMING_CORE_NETWORK_MqttClient_H_
#define _SMING_CORE_NETWORK_MqttClient_H_

#define MQTT_MAX_BUFFER_SIZE 1024

#include "TcpClient.h"
#include "../Delegate.h"
#include "../../Wiring/WString.h"
#include "../../Wiring/WHashMap.h"
#include "../../Services/libemqtt/libemqtt.h"

//typedef void (*MqttStringSubscriptionCallback)(String topic, String message);
typedef Delegate<void(String topic, String message)> MqttStringSubscriptionCallback;
typedef Delegate<void(uint16_t msgId, int type)> MqttMessageDeliveredCallback;

class MqttClient;
class URL;

class MqttClient: protected TcpClient
{
public:
	MqttClient(String serverHost, int serverPort, MqttStringSubscriptionCallback callback = NULL);
	MqttClient(IPAddress serverIp, int serverPort, MqttStringSubscriptionCallback callback = NULL);
	virtual ~MqttClient();

	void setKeepAlive(int seconds);			//send to broker
	void setPingRepeatTime(int seconds);            //used by client
	// Sets Last Will and Testament
	bool setWill(const String& topic, const String& message, int QoS, bool retained = false);

	bool connect(const String& clientName, boolean useSsl = false, uint32_t sslOptions = 0);
	bool connect(const String& clientName,const String& username, const String& password, boolean useSsl = false, uint32_t sslOptions = 0);

	using TcpClient::setCompleteDelegate;

	__forceinline bool isProcessing()  { return TcpClient::isProcessing(); }
	__forceinline TcpClientState getConnectionState() { return TcpClient::getConnectionState(); }

	bool publish(String topic, String message, bool retained = false);
	bool publishWithQoS(String topic, String message, int QoS, bool retained = false, MqttMessageDeliveredCallback onDelivery = NULL);

	bool subscribe(const String& topic);
	bool unsubscribe(const String& topic);

#ifdef ENABLE_SSL
	using TcpClient::addSslOptions;
	using TcpClient::setSslFingerprint;
	using TcpClient::pinCertificate;
	using TcpClient::setSslClientKeyCert;
	using TcpClient::freeSslClientKeyCert;
	using TcpClient::getSsl;
#endif

protected:
	virtual err_t onReceive(pbuf *buf);
	virtual void onReadyToSendData(TcpConnectionEvent sourceEvent);
	void debugPrintResponseType(int type, int len);
	static int staticSendPacket(void* userInfo, const void* buf, unsigned int count);

public: //original private
	String server;
	IPAddress serverIp;
	int port;
	mqtt_broker_handle_t broker;
	int waitingSize;
	uint8_t buffer[MQTT_MAX_BUFFER_SIZE + 1];
	uint8_t *current;
	int posHeader;
	MqttStringSubscriptionCallback callback;
	int keepAlive = 60;
	int PingRepeatTime = 20;
	unsigned long lastMessage = 0;
	HashMap<uint16_t, MqttMessageDeliveredCallback> onDeliveryQueue;
};

/** @} */
#endif /* _SMING_CORE_NETWORK_MqttClient_H_ */
```

slaff, 2018-05-07:
Look at the sample for usage: https://github.com/SmingHub/Sming/blob/develop/samples/MqttClient_Hello/app/application.cpp. That should be enough if you have basic C/C++ understanding.

dpechman, 2018-05-09:
I have it working, but my app need to change anytime the MQTT_HOST and MQTT_PORT. Your example initialise it before call init and use define for MQTT_HOST and MQTT_PORT. How do I set it in runtime?

slaff, 2018-05-09:
> but my app need to change anytime the MQTT_HOST and MQTT_PORT.

@dpechman That is something that should be addressed and have bothered me too. Please, create a new issue with [Feature Request] and close this one.

closed 2018-05-10 by slaff