SmingHub/Sming #1376: OTA fails for bigger firmware

frankdownunder, 2018-05-14:
I'm trying to get OTA working.  But the watchdog bites me.  
I presume this is because my firmware is non trivial size now (399,856 bytes).

So I turned on debug info by recompiling with 
`make DEBUG_VERBOSE_LEVEL=3`.  After I call otaUpdater->start(); I get the following  

_TCP received: 1390 bytes
HttpConnection::onReadyToSendData: waitingQueue.count: 0
Nothing in the waiting queue
onReadyToSendData: 1
rboot_write_flash, item.size: 36140
TCP received: 1390 bytes
HttpConnection::onReadyToSendData: waitingQueue.count: 0
Nothing in the waiting queue
onReadyToSendData: 1
rboot_write_flash, item.size: 37530
TCP received: 1390 bytes
HttpConnection::onReadyToSendData: waitingQueue.count: 0
Nothing in the waiting queue
onReadyToSendData: 1_
 
_ets Jan  8 2013,rst cause:4, boot mode:(3,7)_
 
_wdt reset_

Approximately 35K seems to be as much as it can handle. 

Its not just OTA that suffers; I have also verified the same problem occurs when downloading a file bigger than 35K.

I have tried calling system_soft_wdt_feed();  at the end of rBootItemOutputStream::write but this didnt seem to make any difference.

Strangely, I have two boards, ( a SonoffPOW and a WEMOS D1 mini); the WEMOS downloads just fine.  
The boards are obviously different; is it possible for them to have different settings for the watchdog timer?

slaff, 2018-05-14:
> I presume this is because my firmware is non trivial size now (399,856 bytes).

The size is ok, as long as you have the Flash memory space to save it and should not cause issues. I am using bigger ROM sizes and haven't noticed any issues. A problem may occur when the newly downloaded ROM overrides either the current ROM or area in Flash which is currently used from your application. OR crossed the 1MB border. For example a rom that is starting in 1MB and then ends in 2nd MB will not work.

frankdownunder, 2018-05-17:
I have spent some time looking at the flash size to see if that contributes to the problem.  @slaff  The sizes ARE different, the Wemos reports Manufacturer: ef Device: 4018 Detected flash size: 16MB, while the Sonoff POW reports Manufacturer: e0 Device: 4016 Detected flash size: 4MB. I found in my box of parts a NodeMCU that has the same size flash as the Sonoff, but it works perfectly also.  
So this problem seems restricted to the Sonoff hardware, and its most likely not an issue with Sming, Happy to close this issue. Any suggestions on what to try next ? I would be very grateful, this is a major show stopper for me, I wanted to go into beta testing on my project, can't now.

frankdownunder, 2018-05-21:
I think the Sonoff POW I have is a normal ESP8266.   Its flashes just fine; my problem is that the watchdog bites when downloading a file.  I wonder if it is because of power issues perhaps?

slaff, 2018-07-25:
Feel free to reopen or comment if you have made some progress with the issue.

closed 2018-07-25 by slaff