SmingHub/Sming #1382: Building problems

LazyMechanic, 2018-05-15:
I tried install framework with choco, but I surrendered after 3 days unsuccessful installing.
Further I cloned rep with git and called `make`, but project didn't make, again.
What have I done wrong?

```
ESP_HOME = c:\Espressif
SMING_HOME = D:\Projects\Sming\Sming
```

```
D:\Projects\Sming\Sming>make all
make[1]: Entering directory `/D/Projects/Sming/Sming'
make[2]: Entering directory `/D/Projects/Sming/Sming/third-party/esp-open-lwip'
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/def.o lwip/core/def.c
In file included from include/lwip/arch.h:43:0,
                 from include/lwip/debug.h:35,
                 from include/lwip/opt.h:46,
                 from lwip/core/def.c:39:
include/arch/cc.h:38:21: fatal error: c_types.h: No such file or directory
compilation terminated.
make[2]: *** [lwip/core/def.o] Error 1
make[2]: Leaving directory `/D/Projects/Sming/Sming/third-party/esp-open-lwip'
make[1]: *** [compiler/lib/liblwip_open.a] Error 2
make[1]: Leaving directory `/D/Projects/Sming/Sming'
make: *** [reload] Error 2
```

LazyMechanic, 2018-05-15:
Same error

1. Call `make dist-clean`

```
D:\Projects\Sming\Sming>make dist-clean
make[1]: Entering directory `/d/Projects/Sming/Sming/third-party/esp-open-lwip'
rm -f lwip/core/def.o lwip/core/dhcp.o lwip/core/dns.o lwip/core/init.o lwip/core/mem.o lwip/core/memp.o lwip/core/netif.o lwip/core/pbuf.o lwip/core/raw.o lwip/core/sntp.o lwip/core/stats.o lwip/core/sys_arch.o lwip/core/sys.o lwip/core/tcp.o lwip/core/tcp_in.o lwip/core/tcp_out.o lwip/core/timers.o lwip/core/udp.o lwip/core/ipv4/autoip.o lwip/core/ipv4/icmp.o lwip/core/ipv4/igmp.o lwip/core/ipv4/inet.o lwip/core/ipv4/inet_chksum.o lwip/core/ipv4/ip_addr.o lwip/core/ipv4/ip.o lwip/core/ipv4/ip_frag.o lwip/netif/etharp.o lwip/app/dhcpserver.o espconn_dummy.o /D/Projects/Sming/Sming/compiler/lib/liblwip_open.a
make[1]: Leaving directory `/d/Projects/Sming/Sming/third-party/esp-open-lwip'
make[1]: Entering directory `/D/Projects/Sming/samples/Accelerometer_MMA7455'
make[1]: Leaving directory `/D/Projects/Sming/samples/Accelerometer_MMA7455'
make[1]: Entering directory `/D/Projects/Sming/samples/Arducam'
make[1]: Leaving directory `/D/Projects/Sming/samples/Arducam'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_APA102'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_APA102'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Blink'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Blink'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Capsense'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Capsense'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Debug'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Debug'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Delegates'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Delegates'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_HwPWM'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_HwPWM'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Interrupts'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Interrupts'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_NFC'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_NFC'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Neopixel'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Neopixel'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_PWM'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_PWM'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_ProgMem'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_ProgMem'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_ScannerI2C'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_ScannerI2C'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Serial'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Serial'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Servo'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Servo'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_SmartConfig'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_SmartConfig'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_Ssl'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_Ssl'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_WebClient'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_WebClient'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_WebSkeletonApp'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_WebSkeletonApp'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_WiFi'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_WiFi'
make[1]: Entering directory `/D/Projects/Sming/samples/Basic_rBoot'
make[1]: Leaving directory `/D/Projects/Sming/samples/Basic_rBoot'
make[1]: Entering directory `/D/Projects/Sming/samples/CommandProcessing_Debug'
make[1]: Leaving directory `/D/Projects/Sming/samples/CommandProcessing_Debug'
make[1]: Entering directory `/D/Projects/Sming/samples/Compass_HMC5883L'
make[1]: Leaving directory `/D/Projects/Sming/samples/Compass_HMC5883L'
make[1]: Entering directory `/D/Projects/Sming/samples/DNSCaptivePortal'
make[1]: Leaving directory `/D/Projects/Sming/samples/DNSCaptivePortal'
make[1]: Entering directory `/D/Projects/Sming/samples/DS3232RTC_NTP_Setter'
make[1]: Leaving directory `/D/Projects/Sming/samples/DS3232RTC_NTP_Setter'
make[1]: Entering directory `/D/Projects/Sming/samples/Echo_Ssl'
make[1]: Leaving directory `/D/Projects/Sming/samples/Echo_Ssl'
make[1]: Entering directory `/D/Projects/Sming/samples/FtpServer_Files'
make[1]: Leaving directory `/D/Projects/Sming/samples/FtpServer_Files'
make[1]: Entering directory `/D/Projects/Sming/samples/Gesture_APDS-9960'
make[1]: Leaving directory `/D/Projects/Sming/samples/Gesture_APDS-9960'
make[1]: Entering directory `/D/Projects/Sming/samples/HttpClient_Instapush'
make[1]: Leaving directory `/D/Projects/Sming/samples/HttpClient_Instapush'
make[1]: Entering directory `/D/Projects/Sming/samples/HttpClient_ThingSpeak'
[... 164 lines cut ...]
Applied patch include/lwip/ip_addr.h cleanly.
Applied patch include/lwip/mem.h cleanly.
Applied patch lwip/app/dhcpserver.c cleanly.
Applied patch include/lwip/tcp_impl.h cleanly.
Applied patch include/lwip/igmp.h cleanly.
Fetching third-party/pwm/ ...
[... cut ...]
In file included from include/lwip/arch.h:43:0,
                 from include/lwip/debug.h:35,
                 from include/lwip/opt.h:46,
                 from lwip/core/def.c:39:
include/arch/cc.h:38:21: fatal error: c_types.h: No such file or directory
compilation terminated.
make[2]: *** [lwip/core/def.o] Error 1
make[2]: Leaving directory `/D/Projects/Sming/Sming/third-party/esp-open-lwip'
make[1]: *** [compiler/lib/liblwip_open.a] Error 2
make[1]: Leaving directory `/D/Projects/Sming/Sming'
make: *** [reload] Error 2**

```

kwis2, 2018-05-15:
If you would like to run Sming it should be in `sming/Sming` not `Sming/Sming` directory. 
`fatal error: c_types.h: No such file or directory` - in my oppinion it is known problem with OS identification check #1355

LazyMechanic, 2018-05-15:
It did not help
I changed environment variable `SMING_HOME = D:\Projects\Sming\Sming` to `SMING_HOME = D:\Projects\sming\Sming`

I changed `Makefile`
```
else ifneq ($(filter CYGWIN%,$(UNAME)),)
  # Cygwin Detected
  UNAME := Windows
```

<details>
<summary>Make result</summary>
<pre>
D:\Projects\sming\Sming>make
Fetching Arduino Library Libraries/Adafruit_BME280_Library/ ...
Submodule path 'Libraries/Adafruit_BME280_Library': checked out '321186220e1e04080bb70fa45aae6e6a5820899f'
error: unrecognized input
make: [Libraries/Adafruit_BME280_Library/library.properties] Error 128 (ignored)
Fetching Arduino Library Libraries/Adafruit_SSD1306/ ...
Submodule path 'Libraries/Adafruit_SSD1306': checked out 'ddfec78fa15f0ff8dfc8a76524077ba6bb5fc6f3'
Checking patch Adafruit_SSD1306.cpp...
Checking patch Adafruit_SSD1306.h...
Applied patch Adafruit_SSD1306.cpp cleanly.
Applied patch Adafruit_SSD1306.h cleanly.
Fetching Arduino Library Libraries/Adafruit_ST7735/ ...
Submodule path 'Libraries/Adafruit_ST7735': checked out '591fa4165d95c33ba13bfad2f09758852fe1282a'
error: unrecognized input
make: [Libraries/Adafruit_ST7735/library.properties] Error 128 (ignored)
Fetching Arduino Library Libraries/Adafruit_Sensor/ ...
Submodule path 'Libraries/Adafruit_Sensor': checked out 'e985f2253a687ef377cde3dcfb1f788830d1bc09'
error: unrecognized input
make: [Libraries/Adafruit_Sensor/library.properties] Error 128 (ignored)
Fetching Arduino Library Libraries/IR/ ...
Submodule path 'Libraries/IR': checked out '24c589f6e6a4dafd7dfe201ad8cd72d6efdd7ed0'
Submodule path 'Libraries/IR/lib/googletest': checked out '887d569eb0bac3760cb3adf5877c8284e4d7768e'
Checking patch src/IRrecv.cpp...
Checking patch src/IRrecv.h...
Checking patch src/IRremoteESP8266.h...
Checking patch src/IRsend.cpp...
Checking patch src/IRsend.h...
Checking patch src/ir_Fujitsu.h...
Checking patch src/ir_Gree.h...
Checking patch src/ir_Kelvinator.h...
Checking patch src/ir_Midea.h...
Checking patch src/ir_Mitsubishi.cpp...
Checking patch src/ir_RC5_RC6.cpp...
Checking patch src/ir_RCMM.cpp...
Checking patch src/ir_Toshiba.h...
Applied patch src/IRrecv.cpp cleanly.
Applied patch src/IRrecv.h cleanly.
Applied patch src/IRremoteESP8266.h cleanly.
Applied patch src/IRsend.cpp cleanly.
Applied patch src/IRsend.h cleanly.
Applied patch src/ir_Fujitsu.h cleanly.
Applied patch src/ir_Gree.h cleanly.
Applied patch src/ir_Kelvinator.h cleanly.
Applied patch src/ir_Midea.h cleanly.
Applied patch src/ir_Mitsubishi.cpp cleanly.
Applied patch src/ir_RC5_RC6.cpp cleanly.
Applied patch src/ir_RCMM.cpp cleanly.
Applied patch src/ir_Toshiba.h cleanly.
Fetching Arduino Library Libraries/RingBufCPP/ ...
Submodule path 'Libraries/RingBufCPP': checked out '9e091e9632bc3eba513f422ee7cb17895c565a1a'
error: unrecognized input
make: [Libraries/RingBufCPP/library.properties] Error 128 (ignored)
Fetching third-party/rboot/ ...
Submodule path 'third-party/rboot': checked out '81caa867dbfbb2447b0af66c8110bb8641d8c29d'
Checking patch Makefile...
Checking patch rboot.c...
Checking patch rboot.h...
[... 77 lines cut ...]
lwip/core/ipv4/ip_addr.c: In function 'ipaddr_aton':
lwip/core/ipv4/ip_addr.c:170:10: warning: implicit declaration of function 'isdigit' [-Wimplicit-function-declaration]
     if (!isdigit(c))
          ^
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -ID:\Projects\sming\Sming\third-party/ESP8266_NONOS_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/ipv4/ip.o lwip/core/ipv4/ip.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -ID:\Projects\sming\Sming\third-party/ESP8266_NONOS_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/ipv4/ip_frag.o lwip/core/ipv4/ip_frag.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -ID:\Projects\sming\Sming\third-party/ESP8266_NONOS_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/netif/etharp.o lwip/netif/etharp.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -ID:\Projects\sming\Sming\third-party/ESP8266_NONOS_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/app/dhcpserver.o lwip/app/dhcpserver.c
lwip/app/dhcpserver.c: In function 'parse_msg':
lwip/app/dhcpserver.c:703:11: warning: implicit declaration of function 'wifi_softap_set_station_info' [-Wimplicit-function-declaration]
       if (wifi_softap_set_station_info(m->chaddr, &client_address) == false) {
           ^
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -ID:\Projects\sming\Sming\third-party/ESP8266_NONOS_SDK/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o espconn_dummy.o espconn_dummy.c
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar rcs /D/Projects/sming/Sming/compiler/lib/liblwip_open.a lwip/core/def.o lwip/core/dhcp.o lwip/core/dns.o lwip/core/init.o lwip/core/mem.o lwip/core/memp.o lwip/core/netif.o lwip/core/pbuf.o lwip/core/raw.o lwip/core/sntp.o lwip/core/stats.o lw [... cut ...] p_frag.o lwip/netif/etharp.o lwip/app/dhcpserver.o espconn_dummy.o
make[2]: Leaving directory `/D/Projects/sming/Sming/third-party/esp-open-lwip'
AS system/irq_check.s
AS gdb/gdbstub-entry.s
In file included from system/crash_handler.c:24:0:
include/user_config.h:25:22: fatal error: c_types.h: No such file or directory
compilation terminated.
make[1]: *** [out/build/system/crash_handler.c.d] Error 1
make[1]: Leaving directory `/D/Projects/sming/Sming'
make: *** [reload] Error 2
</pre>
</details>

LazyMechanic, 2018-05-15:
Well.... I don't know what I did, but it has been built.... Half of firmware... 

<details>
<summary>Make result</summary>
<pre>
D:\Projects\sming\Sming>make all
Fetching Arduino Library Libraries/Adafruit_BME280_Library/ ...
Submodule path 'Libraries/Adafruit_BME280_Library': checked out '321186220e1e04080bb70fa45aae6e6a5820899f'
error: unrecognized input
make: [Libraries/Adafruit_BME280_Library/library.properties] Error 128 (ignored)
Fetching Arduino Library Libraries/Adafruit_SSD1306/ ...
Submodule path 'Libraries/Adafruit_SSD1306': checked out 'ddfec78fa15f0ff8dfc8a76524077ba6bb5fc6f3'
Checking patch Adafruit_SSD1306.cpp...
Checking patch Adafruit_SSD1306.h...
Applied patch Adafruit_SSD1306.cpp cleanly.
Applied patch Adafruit_SSD1306.h cleanly.
Fetching Arduino Library Libraries/Adafruit_ST7735/ ...
Submodule path 'Libraries/Adafruit_ST7735': checked out '591fa4165d95c33ba13bfad2f09758852fe1282a'
error: unrecognized input
make: [Libraries/Adafruit_ST7735/library.properties] Error 128 (ignored)
Fetching Arduino Library Libraries/Adafruit_Sensor/ ...
Submodule path 'Libraries/Adafruit_Sensor': checked out 'e985f2253a687ef377cde3dcfb1f788830d1bc09'
error: unrecognized input
make: [Libraries/Adafruit_Sensor/library.properties] Error 128 (ignored)
Fetching Arduino Library Libraries/IR/ ...
Submodule path 'Libraries/IR': checked out '24c589f6e6a4dafd7dfe201ad8cd72d6efdd7ed0'
Submodule path 'Libraries/IR/lib/googletest': checked out '887d569eb0bac3760cb3adf5877c8284e4d7768e'
Checking patch src/IRrecv.cpp...
Checking patch src/IRrecv.h...
Checking patch src/IRremoteESP8266.h...
Checking patch src/IRsend.cpp...
Checking patch src/IRsend.h...
Checking patch src/ir_Fujitsu.h...
Checking patch src/ir_Gree.h...
Checking patch src/ir_Kelvinator.h...
Checking patch src/ir_Midea.h...
Checking patch src/ir_Mitsubishi.cpp...
Checking patch src/ir_RC5_RC6.cpp...
Checking patch src/ir_RCMM.cpp...
Checking patch src/ir_Toshiba.h...
Applied patch src/IRrecv.cpp cleanly.
Applied patch src/IRrecv.h cleanly.
Applied patch src/IRremoteESP8266.h cleanly.
Applied patch src/IRsend.cpp cleanly.
Applied patch src/IRsend.h cleanly.
Applied patch src/ir_Fujitsu.h cleanly.
Applied patch src/ir_Gree.h cleanly.
Applied patch src/ir_Kelvinator.h cleanly.
Applied patch src/ir_Midea.h cleanly.
Applied patch src/ir_Mitsubishi.cpp cleanly.
Applied patch src/ir_RC5_RC6.cpp cleanly.
Applied patch src/ir_RCMM.cpp cleanly.
Applied patch src/ir_Toshiba.h cleanly.
Fetching Arduino Library Libraries/RingBufCPP/ ...
Submodule path 'Libraries/RingBufCPP': checked out '9e091e9632bc3eba513f422ee7cb17895c565a1a'
error: unrecognized input
make: [Libraries/RingBufCPP/library.properties] Error 128 (ignored)
Fetching third-party/rboot/ ...
Submodule path 'third-party/rboot': checked out '81caa867dbfbb2447b0af66c8110bb8641d8c29d'
Checking patch Makefile...
Checking patch rboot.c...
Checking patch rboot.h...
Applied patch Makefile cleanly.
Applied patch rboot.c cleanly.
Applied patch rboot.h cleanly.
Fetching third-party/spiffs/ ...
Submodule path 'third-party/spiffs': checked out 'f5e26c4e933189593a71c6b82cda381a7b21e41c'
error: unrecognized input
make: [third-party/spiffs/makefile] Error 128 (ignored)
Fetching third-party/http-parser/ ...
[... 573 lines cut ...]
	include Makefile-macos.mk      
else ifeq ($(UNAME),Linux)
	# Linux Detected
	UNAME := Linux

	# Default SMING_HOME. Can be overriden.
	SMING_HOME ?= /opt/sming/Sming

	# Default ESP_HOME. Can be overriden.
	ESP_HOME ?= /opt/esp-open-sdk

	include Makefile-linux.mk
else ifeq ($(UNAME),FreeBSD)
	# Freebsd Detected
	UNAME := FreeBSD

	# Default SMING_HOME. Can be overriden.
	SMING_HOME ?= /usr/local/esp8266/Sming/Sming

	# Default ESP_HOME. Can be overriden.
	ESP_HOME ?= /usr/local/esp8266/esp-open-sdk
	
	include Makefile-bsd.mk
endif
```

slaff, 2018-05-15:
@LazyMechanic Can you make a PR with the changes that made the compilation work for you?

LazyMechanic, 2018-05-15:
@slaff I may, but the project cannot be built :)

LazyMechanic, 2018-05-15:
What should happen after building?
What i must use for make self firmware with sming?

LazyMechanic, 2018-05-16:
I decided to make sample (e. g. `Basic_Blink`), and I don't understand what is the firmware, where is place it?
`out/firmware` is empty, it have only empty `upgrade` folder.

Make result:
```
...
LD spiffy
make[2]: Leaving directory `/D/Projects/sming/Sming'
Making esptool2
make[2]: Nothing to be done for `all'.
Making spiffy
make[2]: Nothing to be done for `all'.
make[1]: Leaving directory `/D/Projects/sming/Sming'
make[1]: Entering directory `/D/Projects/sming/Sming'
make[1]: Leaving directory `/D/Projects/sming/Sming'
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE837E|     894
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFE8908|    1416
       bss|      Uninitialized Data (RAM)|    3FFE8908|    3FFEEB30|   25128
      text|            Cached Code (IRAM)|    40100000|    401062DE|   25310
irom0_text|           Uncached Code (SPI)|    40207000|    402352B4|  189108
Total Used RAM : 27438
Free RAM : 54482
Free IRam : 7476
------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.

```

slaff, 2018-05-16:
> I decided to make sample (e. g. Basic_Blink), and I don't understand what is the firmware, where is place it?

As far as I can see you have successfully compiled the application. The next step is to connect your ESP8266 device and flash the app to it by calling:

```
make flash
```

LazyMechanic, 2018-05-16:
Oh ... It's my mistake. I had looking at `sming/Sming/out/firmware`, not at `sming/samples/Basic_Blink/out/firmware`.

Now I'm glad. It is working.

Do I understand correctly that for self firmware I'm needing main files:

> Makefile-user.mk
> Makefile
> user_config.h

and I will need edit/append folder to `MODULES` with my "main.cpp" (starting point of my app) in `Makefile-user.mk` ?