SmingHub/Sming #1394: Fatal exception 29

Marvell13, 2018-06-03:
I'm new to programming ESP8266. I made the code below for tests, using the site automacao-iot.com.br.

Example Code:

```
#define WIFI_SSID	"ssid"
#define WIFI_PWD	"password"

Timer pushTimer01,pushTimer02;

void executeEntryPoint1();
void executeEntryPoint2();

void sendHTTP(String url,String secretkey);
int callBackHttp(HttpConnection& connection, bool success);

int callBackHttp(HttpConnection& connection, bool success)
{
    if(success) {
        Serial.println(connection.getResponseString());
        Serial.println(connection.getResponseCode());
    }
    return 0;
}

void sendHTTP(String url,String secretkey)
{
    HttpClient *httpClient;
    HttpHeaders headers;

    
    headers["Authorization"] = "Bearer "+secretkey;
    headers["Connection"]="Close";
    headers["ContentType"] = "application/json";

    httpClient->send(
        httpClient->request(url)        
        ->setMethod(HTTP_GET)
        ->setHeaders(headers)
        ->onRequestComplete(callBackHttp) 
    );
    

    Serial.println("############################################");
    String heap = String(system_get_free_heap_size()); 
    Serial.println("Memory HEAP "+heap);
    Serial.println("############################################");
}

void executeEntryPoint1()
{
    sendHTTP("http://automacao-iot.com.br/api/device/A1472FBA268D778E1DBF370DE7580A7F/info","5025003AF915DC944B8ACAAFA4A0D405E87E83B6D7AE2FD4870742E86D35229D");
}

void executeEntryPoint2()
{
    sendHTTP("http://automacao-iot.com.br/api/device/ECEEFFEA77FD3CC970B87670B70809CD/info","E0F565CBE3D0FCBE4FE75B27733AD3450D16AB4F9DB6F13F291989D556C5E8BE");
}

// Connection failed
void connectionFailed(String ssid, uint8_t ssidLength, uint8_t *bssid, uint8_t reason)
{
	debugf("Not Connected");
	system_restart();
}

//Station connected
void connected(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	debugf("Connected");	
[... 238 lines cut ...]
3ffffec0:  00000000 3fffaa60 3fffbda8 3fffb918  
3ffffed0:  00000000 3fffaa60 3fffbda8 3fffff40  
3ffffee0:  3fffbda8 3fffaa60 00000000 4023750b  
3ffffef0:  3fffad20 3fffb890 3fff9708 00000000  
3fffff00:  00030003 00000000 3fff9d68 000b000b  
3fffff10:  3fff9d68 000d000d 3fffff40 40106084  
3fffff20:  3ffe89c2 00000000 402091f4 60000600  
3fffff30:  02ee4fc3 3ffea080 3ffeff60 40237650  
3fffff40:  3fffb000 00400040 3fffb838 004c004c  
3fffff50:  402091f4 00000000 00000000 40239380  
3fffff60:  40209f1c 3ffeff64 3ffea080 4010656a  
3fffff70:  60000600 3ffeff64 3ffeff64 40209f29  
3fffff80:  40209f6e 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffea0a8 3fffdab0 00000000 4020e437  
3fffffa0:  40000f49 40000f49 ffffff01 55aa55aa  

================================================================
To decode the stack dump call from command line:
   python $SMING_HOME/../tools/decode-stacktrace.py out/build/app.out
and copy & paste the text enclosed in '===='.
```

What am I doing wrong ?? Any suggestions for changing my code ??

Thanks

dpechman, 2018-06-05:
check the power supply!

slaff, 2018-06-05:
@Marvell13 If your actual code is very close to the one pasted above then it is real miracle that this code is working at all.

In `sendHTTP` you are creating every time a new pointer to an HttpClient object, never initialize that pointer or allocate space for it and directly call `send` method on the pointer. 
```
HttpClient *httpClient;
...

httpClient->send(request);
```

What would be much better is to have one global HttpClient object and inside of the sendHTTP method just create a new request object and use that object to send the request. Something like:
```c++
HttpClient httpClient;

...

void sendHTTP(String url,String secretkey)
{
    HttpHeaders headers;
    headers["Authorization"]  = "Bearer "+secretkey;
    headers["Connection"]    = "Close";
    headers["Content-Type"]  = "application/json"; // NOTICE: the right header name is 
                                                                             // Content-Type and not ContentType
   
    HttpRequest* request = httpClient.request(url);
    request->setMethod(HTTP_GET)
                ->setHeaders(headers)
                ->onRequestComplete(callBackHttp)

    httpClient.send(request);
    
    Serial.println("############################################");
    String heap = String(system_get_free_heap_size()); 
    Serial.println("Memory HEAP "+heap);
    Serial.println("############################################");
}

```

Marvell13, 2018-06-05:
@slaff 

I made the above modifications and decreasing the heap memory continues resetting the ESP8266.

code

```
#include <user_config.h>
#include <SmingCore/SmingCore.h>

#define WIFI_SSID	"xxxxx"
#define WIFI_PWD	"yyyyyy"

Timer pushTimer01,pushTimer02;
HttpClient httpClient;

void executeEntryPoint1();
void executeEntryPoint2();

void sendHTTP(String url,String secretkey);
int callBackHttp(HttpConnection& connection, bool success);

int callBackHttp(HttpConnection& connection, bool success)
{
    if(success) {
        Serial.println(connection.getResponseString());
        Serial.println(connection.getResponseCode());
    }
    return 0;
}

void sendHTTP(String url,String secretkey)
{
    HttpHeaders headers;

    headers["Authorization"] = "Bearer "+secretkey;
    headers["Connection"]="Close";
    headers["Content-Type"] = "application/json";

    //HttpRequest* request = httpClient.request(url);
    HttpRequest* request = httpClient.request(url);

    request->setMethod(HTTP_GET)
                ->setHeaders(headers)
                ->onRequestComplete(callBackHttp);

    httpClient.send(request); 

    Serial.println("############################################");
    String heap = String(system_get_free_heap_size()); 
    Serial.println("Memory HEAP "+heap);
    Serial.println("############################################");
}

void executeEntryPoint1()
{
    sendHTTP("http://automacao-iot.com.br/api/device/A1472FBA268D778E1DBF370DE7580A7F/info","5025003AF915DC944B8ACAAFA4A0D405E87E83B6D7AE2FD4870742E86D35229D");
}

void executeEntryPoint2()
{
    sendHTTP("http://automacao-iot.com.br/api/device/ECEEFFEA77FD3CC970B87670B70809CD/info","E0F565CBE3D0FCBE4FE75B27733AD3450D16AB4F9DB6F13F291989D556C5E8BE");
}

// Connection failed
void connectionFailed(String ssid, uint8_t ssidLength, uint8_t *bssid, uint8_t reason)
{
	debugf("Not Connected");
	system_restart();
}

//Station connected
void connected(IPAddress ip, IPAddress mask, IPAddress gateway)
{
	debugf("Connected");
	//pushTimer01.initializeMs(1000, executeEntryPoint1).startOnce();
	//executeEntryPoint1();
	pushTimer01.initializeMs(1000, executeEntryPoint1).start();
	pushTimer02.initializeMs(1000, executeEntryPoint2).start();
} 
 
void init()
{
    Serial.begin(SERIAL_BAUD_RATE); // 115200 by default
    Serial.systemDebugOutput(true); // Debug output to serial

    WifiStation.config(WIFI_SSID, WIFI_PWD);
    WifiStation.enable(true);
    WifiAccessPoint.enable(false);

    WifiEvents.onStationGotIP(connected);
    WifiEvents.onStationDisconnect(connectionFailed);        
}   
```

slaff, 2018-06-09:
@Marvell13 I have an idea what is happening. And I have checked it on my devices. 

Short-Story: Comment out that code:
```
    headers["Connection"]="Close";
```
and you will get the memory consumption to fluctuate up and down but remain relatively stable.

Long-Story: 
The header above  instructs the remote server to close the TCP connection after every HTTP request. If you comment it out then the TCP connection will be reused and the memory consumption will be relatively stable. Once a Tcp connection is closed it will stay for some time ( more than 2 seconds.. and more like 60 seconds) until it is deleted and its memory is reclaimed. 

In your case you send two requests every second, close the Tcp connection after the response is received and thus generate much more waiting-to-be-closed Tcp connections than the device and the Tcp stack can handle.

Marvell13, 2018-06-12:
@slaff  Thanks for the help, sorry for the questions, because I'm doing a search for my school.

I made the modifications and performed much better without loss of memory, but after some time running, it presents the message "The request queue is full at the moment" and does not execute more, something that I can do about it?

Thankful.

slaff, 2018-06-15:
> "The request queue is full at the moment" and does not execute more, something that I can do about it?

Check how much time it takes for a request to receive response. If it is, let say 2 seconds, then send new requests once per 3 or 4 seconds.

OR if you really want to send data quite often then just use the websocket client. You will have one connection to the server and send data as often as you like.

closed 2018-08-08 by slaff