SmingHub/Sming #1397: Licensing

devyte, 2018-06-05:
This repo's license is specified as LGPL-3. However, one of the 3rd party libs is ESP8266_new_pwm, which is specified as GPL-2. I understand that when you incorporate code licensed under GPL, then the entire work must be GPL. 
How is this compatible? Doesn't this mean that any Sming-based firmware that uses ESP8266_new_pwm must also be GPL? Or is my understanding of GPL wrong?
Also, what happens with the closed-source SDK parts when the whole work becomes GPL?

vijaygill, 2018-06-18:
+1 vote from me.

In my view, any code that makes Sming more restrictive must be removed / replaced.

@devyte it will be great if you take this task and raise a PR please?

devyte, 2018-06-18:
@vijaygill a PR with what? Removal of ESP8266_new_pwm? Wouldn't that break any existing applications that rely on it?
Please be aware that I'm not a user of the code of this repo. It just happens that I'm aware of the licensing of ESP8266_new_pwm due to a discussion with the author of that code in the context of using it in conjunction with our Arduino core. I just happened to notice in a google search that this Sming repo is building with that code under 3rd party, and I thought to ask the question.
Honestly, I don't even know how _any_ ESP-based code can be released under GPL, given that there is closed pre-built code distributed by Espressif. It seems completely incompatible to me, except maybe for private personal use, in which case GPL makes no sense.

vijaygill, 2018-06-18:
@devyte ok.

slaff, 2018-06-19:
There are probably many things that can be done but the following should suffice:
* Disable by default the GPL PWM library
* If someone enables it warn him/her about the GPL license requirements and consequences

But before we do this I would like to discuss this with the author of the custom PWM.

devyte, 2018-06-20:
I already tried to discuss with the author [here](https://github.com/StefanBruens/ESP8266_new_pwm/issues/16), and there are other similar issues opened.  You're welcome to try again.
There seems to be a misunderstanding about what "compatible" means in this context, both here in the comment about the compatibility matrix, and in the mind of the author of new pwm. A free license is "compatible" with GPL if that license also allows the terms of GPL. The 3-clause BSD allows it, as does LGPL.
HOWEVER: compatible does not mean equivalent. Source code that is distributed under the 3-clause BSD or LGPL licensed can get the additional restrictions imposed by GPL, and it will work, because the other license is more permissive. In the case of this repo, it's ok, because you can decide what to do with the source code contained here, and you've already done so: it is freely available online. Any user who develops app code with the code in this repo can choose to distribute the source code, or not.
But a user who builds against new pwm would be forced to license his app code under GPL, i.e.: his code would forcibly have to be made available to anyone who asks, in addition to making this repo's code available. My thinking is that any user who builds commercial applications needs to be VERY aware of that, and must fully comply with the GPL restrictions, or risk liability. In other words, a user who makes a commercial application based on sming and uses new pwm, must make display the GPL license, must make his app's source code available, and must point to the source code of this repo, and any other source code repos used. If that's not done, that user can get legal trouble if discovered.

I suspect that, because the resulting work that uses new pwm would need to be licensed under GPL, and that work includes the SDK, the SDK source code would ALSO need to be made available. But in the case of the SDK, it is distributed as binary blob where the code is not available, because Espressif decided to do it that way. That distribution is done under a license that is permissive and allows Espressif to do that.

So, as I understand it, a permissive free license may be "compatible" with GPL, but that does not necessarily mean that work distributed under that permissive license can be used with GPL-licensed work. 

I'm also not a lawyer, though, so I could very well be wrong.