SmingHub/Sming #1407: Question about SmingCore.h

Vinimuller, 2018-06-20:
Hello!
At my application for an ESP8266-12 I'm trying to keep things organized. For that I created a "sensorHandler.h" file inside "include" folder and inside "app"  folder I created the respective "sensorHandler.cpp" file.
At my header file I'm trying to create a class that contains variables of type String and Timer which are defined in SmingCore.h, but when I include SmingCore.h in my header file and compile, I keep receiving a bunch of erros saying:
`"error: template with C linkage template <class Element>"`

After some tests I realized that I always receive this error when I try to include SmingCore.h in any header file.

So my question is: How can I declare my class that uses variables of type Timer an String?

I'm running Sming on Ubuntu 16.04 LTS and the class that I am trying to create is:

[image]

slaff, 2018-06-21:
@Vinimuller It sounds like programming error from your side. Did you put C++ code inside extern "C" block(s)? May be the easiest is to publish the code of the application so that I can look over it when I have some time.

Vinimuller, 2018-06-21:
@slaff Yeah, I've read about it and found something mentioning the extern "C", but I don't have C++ code inside it.
Here is the sensorHandler.cpp

[image]

slaff, 2018-06-28:
@Vinimuller Please create a sample app where the problem is reproducible. Put that app in Github and share the link with us. After that I can try to reproduce the issue locally and analyze and fix it, if needed.

Vinimuller, 2018-06-28:
Hey @slaff, I got it fixed!
The problem was that I tried to include `#include<SmingCore/SmingCore.h>` but it actually is `#include<SmingCore.h>`
Thanks!

closed 2018-06-28 by Vinimuller