SmingHub/Sming #1412: Fatal Error 28 using WT8266

giorgiogalantegit, 2018-07-10:
I would like to adopt this Framework for some project I am working with. I have setup the environment in widows 10 and set SPI size to 2M accordingly to the module specification.

I try several samples (HTTP Server AJAX, Network config, etc. but any of this sample show me a Fatal error 28.
any suggestion is more  than welcome!
kind regards

slaff, 2018-07-12:
@giorgiogalantegit Make sure that the Flash size is correct. Then follow [that article](https://github.com/SmingHub/Sming/wiki/Application-prints-random-symbols-and-restarts) and `flashinit` the device. It has to be done once. Make sure to power-off and the power-on the device. Then try to flash again the application.  If the error persists again then paste here the complete log and the decoded stack.

giorgiogalantegit, 2018-07-12:
Hello, I switch to the EspressIf SDK 1.54 and all work fine. before I were using the SDK 2.0

slaff, 2018-07-12:
@giorgiogalantegit I guess SDK 2.0 will work also fine for you, but you should `flashinit` if you move between SDKs.

giorgiogalantegit, 2018-07-12:
@staff if i ry to use flashinti i get this error:
================
make flashinit 
Flash init data default and blank data.
/c/Espressif/utils/ESP8266/esptool.exe -p COM3 -b 115200 erase_flash
Connecting...
head: 0 ;total: 0
erase size : 0
/c/Espressif/utils/ESP8266/esptool.exe -p COM3 -b 115200 write_flash -ff 40m -fm qio -fs 16m 0x1fc000 /c/Espressif/ESP8266_SDK/bin/esp_init_data_default.bin 0x1fe000 /c/Espressif/ESP8266_SDK/bin/blank.bin 0x100000 /C/tools/sming/Sming/compiler/data/blankfs.bin
Connecting...
Traceback (most recent call last):
  File "esptool.py", line 567, in <module>
  File "esptool.py", line 160, in connect
Exception: Failed to connect
make: *** [flashinit] Error 255

================

slaff, 2018-07-12:
Hmm... Try the following sequence of commands:

* Power-Off -> Power-On the device. 
* Put the device in FLASH mode
* /c/Espressif/utils/ESP8266/esptool.exe -p COM3 -b 115200 erase_flash
* Power-Off -> Power-On the device
* Put the device in FLASH mode
* c/Espressif/utils/ESP8266/esptool.exe -p COM3 -b 115200 write_flash -ff 40m -fm qio -fs 16m 0x1fc000 /c/Espressif/ESP8266_SDK/bin/esp_init_data_default.bin 0x1fe000 /c/Espressif/ESP8266_SDK/bin/blank.bin 0x100000 /C/tools/sming/Sming/compiler/data/blankfs.bin

giorgiogalantegit, 2018-07-12:
Yes in this way work perfectelly!
thank you.. 
BTW.. we find a bug and a solution in the websocket library I am going to submit a new issue

slaff, 2018-07-12:
> BTW.. we find a bug and a solution in the websocket library I am going to submit a new issue

Thanks in advance. Meanwhile I will close this ticket because the original issue seems to be solved.

closed 2018-07-12 by slaff