SmingHub/Sming #1414: Refactoring: Move ArduinoJson from Services to Libraries

slaff, 2018-07-12:
The ArduinoJson library should be moved from Services to Libraries. Libraries is better suited as it contains the Arduino compatible libraries and we have mechanism there that can help us keep the third-party Arduino libraries up-to-date.

vijaygill, 2018-07-12:
A quick question - will it become a git-module also as part of refactoring?

Thanks.
Vijay

slaff, 2018-07-12:
> A quick question - will it become a git-module also as part of refactoring?

If there are no big differences then a volunteer, for example you :), can add it as a git submodule.

vijaygill, 2018-07-12:
Sure! No problem! I will be glad to do it :)

slaff, 2018-07-12:
@vijaygill Get the latest `develop` version and give it a try. ArudinoJson is now moved to Libraries.

PR #1415 (2018-07-12): Moved ArduinoJson from Services to Libraries
Closes #1414.
closed 2018-07-12 by slaff