SmingHub/Sming #1416: Issue on the WebSocket library

giorgiogalantegit, 2018-07-12:
We start to test the websocket library with enabled SSL by using  simple websocket server make by using node js 8.11.3 and express-ws 4.0.0.

when the client try to complete the connection the server crash with this error code:

RangeError: Invalid WebSocket frame: FIN must be set

Looking into the library I see that during the connecting a ping message is sent.
The  ping can be send only after the connection has been establish. We just comment this line and the connection complete without error..
We start to receive message from the client, but as soon as the client close the connection the server crash again. we see the close method send 2 byte 0x87, 0x00. For my knowing for close the connection, the code is just a single byte equal to 0x8. We made also this update and also the close work properly.

the remain issue that I not was able to figure out are that if I send a ping or a message from the server to the client this replay in a manner that let crash again the server.
We are using nodejs 
Kind regard

slaff, 2018-07-12:
You can try the still unfinished websocket refactoring in that PR: #1387.

Morbden, 2018-10-27:
https://github.com/SmingHub/Sming/issues/1324#issue-296067050

This is the solution to the FIN error. But it should be correct already in the release.

PR #1514 (2018-11-07): Refactoring websocket client and server.
## Improvements
* Simplified and decreased the code base responsible for the Websocket functionality. 
Refactored the WebsocketConnection to be reused in server and client mode.

* Documentation for the updated connections.

## Breaking Changes (Impact: High)

If you use Websocket functionality make sure to read the notes below. 

* Renamed Websocket related types, classes and methods. We used to have `WebSocket` and `Websocket`.  From now on only `Websocket` will be used. Example ` WebsocketConnection::getActiveWebSockets` is now ` WebsocketConnection: [... cut]
closed 2018-11-07 by slaff