SmingHub/Sming #1417: Enabling SSL in HttpServer

icodk, 2018-07-12:
Is http over wifi is encrypted ? 
As fa as I know only wifi key exchange is encrypted. The actual traffic is readable to any wifi adapter in Monitor mode that can capture the traffic.
or what ?

slaff, 2018-07-12:
> Is http over wifi is encrypted ?

Short answer: 
If you have WiFi Access Point with WPA protection or better and this access point is dedicated to your devices then knowing the WPA passphrase can help hackers in their efforts to decrypt the traffic.

Long answer:
I would take the liberty and generalize the question to "Is any communication data over WIFI encrypted?". To answer this question we need to answer one additional question: Is the WIFI access point with open access, WEP, WPA or better? 

If you use open WIFI access point (AP) or one which is protected with WEP then the answer is - nothing is really encrypted, because it is possible and relatively easy for other parties to sniff your WIFI traffic and decode the communication data.

On the other hand if you use WPA or better and you will need the WPA passphrase to decode the content  + the individual session keys that WPA created during the initial handshake.  So if you don't have the WPA passphrase + session keys you will be need time and luck to dig them and decrypt the traffic. Until then your traffic will be safe and unreadable for the WIFI sniffers.

icodk, 2018-07-12:
> So if you don't have the WPA passphrase + session keys you will be need time and luck to dig them and decrypt the traffic. Until then your traffic will be safe and unreadable for the WIFI sniffers.

OK thanks I think I got it