SmingHub/Sming #1425: Sming app compiled and flashed, but doesn't run on NodeMCU board

chall3ng3r, 2018-08-04:
Hi, I'm starting to get back in using Sming for my ESP based projects. I had previous manual install with pretty old version of Sming and ESP SDK.

I updated Sming to latest dev build, installed latest beta UDK. Was getting an error, but after setting ESP_HOME and SMING_HOME environment vars, Sming samples started to compile fine.

I was able to only copile and flash Basic_Wifi sample once. After I made some changes to sample, and re-flashed, it started showing garbage in serial. Checked baudrate, recompiled Sming but no use.

Made a quick test using PlatformIO Arduino framework, and it flashed and worked perfectly.

Now I'm not sure why my Sming (develop) samples compiles perfectly, flash without errors, but doesn't run on NodeMCU board.

Anyone pointers, what's I'm missing to check? 

System:
Windows 10 Pro
ESP UDK 2.2.2
Sming (latest develop)

This is output from building "Basic_Serial" sample

```
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app.out

#Memory / Section info:
------------------------------------------------------------------------------
   Section|                   Description| Start (hex)|   End (hex)|Used space

------------------------------------------------------------------------------

      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8390|     912

    rodata|           ReadOnly Data (RAM)|    3FFE8390|    3FFE91D8|    3656

       bss|      Uninitialized Data (RAM)|    3FFE91D8|    3FFEF490|   25272

      text|            Cached Code (IRAM)|    40100000|    401068DB|   26843

irom0_text|           Uncached Code (SPI)|    40208000|    40238FEA|  200682

Total Used RAM : 29840

Free RAM : 52080

Free IRam : 5943

------------------------------------------------------------------------------
# Generating image...
Generate firmware images successully in folder out/firmware.
Done
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
Killing Terminal to free COM3
/C/Espressif/utils/ESP8266/esptool.exe -p COM3 -b 115200 write_flash -ff 40m -fm qio -fs 4m 0x00000 out/firmware/0x00000.bin 0x8000 out/firmware/0x8000.bin
WARNING: Flash size arguments in megabits like '4m' are deprecated.
Please use the equivalent size '512KB'.
Megabit arguments may be removed in a future release.
esptool.py v2.0.0
Connecting....
Detecting chip type... ESP8266
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Flash params set to 0x0000
Compressed 31456 bytes to 22427...

Writing at 0x00000000... (50 %)
Writing at 0x00004000... (100 %)
Wrote 31456 bytes (22427 compressed) at 0x00000000 in 2.0 seconds (effective 126.3 kbit/s)...
Hash of data verified.
Compressed 200684 bytes to 147562...

Writing at 0x00008000... (10 %)
Writing at 0x0000c000... (20 %)
Writing at 0x00010000... (30 %)
Writing at 0x00014000... (40 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (60 %)
Writing at 0x00020000... (70 %)
Writing at 0x00024000... (80 %)
Writing at 0x00028000... (90 %)
Writing at 0x0002c000... (100 %)
Wrote 200684 bytes (147562 compressed) at 0x00008000 in 13.2 seconds (effective 121.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
"pause"

```

slaff, 2018-08-06:
@chall3ng3r Try this https://github.com/SmingHub/Sming/wiki/Application-prints-random-symbols-and-restarts

chall3ng3r, 2018-08-06:
Okay, trying out other samples. HTTP_AJAX works. Its something with ConfigNetwork I think, maybe I have sample from old version mixed up. Going to delete all samples and download again.

Will share how it goes.

slaff, 2018-08-06:
> Now just joined back in, and some new features look amazing.

Welcome back and hope to see you sending PRs soon :).

> working, and shows random characters in serial output.

If the crash is caused due to an error in Sming then you should be able to see a stack trace, that you can later decode. Try to reset the device after flashing. Then capture the serial output and paste it here from the start till the crash to see what is actually happening.

chall3ng3r, 2018-08-06:
Thanks you, yes started working with node-red, and didn't find any ESP8266 framework to work with it to my liking. Sming is my favorite, so might be contributing back as I develop on it.

After multiple tries, and setting export SPI_SIZE=4M, I was able to boot my ESP with HTTP_Bootstrap sample, it showed progress of initializing FS on serial, and connected to my home WiFi. However, it disconnects automatically and reconnects to WiFi in 10 seconds time.

After restarting few times with reset button on board, now it started to crash again. Here's the output from serial:

`;l[00]dœß|[00]Œ$à|[03][04][0C][04]Ä[0C]dì[04]c|Çƒ[02]ä[1B]“{’cÄ[0C]bŒóogŸl'oÜãì[0C]#[1C]xŒ${l{$pûgà[10][02][04][0C]ƒ[04]l[04][0C][0C][04][0C][04]c[0C]nâ|[03]Œì[0C][04]‡cÄò'oï[00]$ŒŽl`[03][12][1B]o'$Œd`[02][0E][03]gsÛ“o[0C][04]Ûc$`[03]`û'[04][0C][04]ŸàbƒgdŒ[04]Äónoï[00][04]Ç[07]lxóoà[18][03][0C][04]{Œœœãà[04][0C][04][0C]#[04]gã|[03]d[04][0C]c„ûgnç[00]dÄ[0C]d`[03][1B][13]ngdŒl`[03][07][03]orÇ›’'[0C][0C][03]l›`[02] ûg[04][0C][0C]ŸàcƒolÄ[0C]„ûgoî[00][04]‡[0F]$pò'à[18][03][04][0C]{ŒÜœâàäl[0C][04]c[04]oã<[03]ldŒÇ[04]c„ûgoî[00]dŒ[0C]$`[03][1B][13]ngdÄl`[03][07][03]orÇ›’'[04][0C][1B]â“`[02][07]{’“o[0C][04][12]ãÛ`[03]„ãû“p[03]Ã'g€l[1B]ld[04][0C][03]lÇcìÇ[03][0C][04]ãâ›`[02]Œã[03]ä[12]ƒoì›rƒûg|ì[0C]d[0C]l$l [03]„ã{“dŒl[12]Œ[04]ŒÜœ[0E]dŒãs’läl[1B]Ä[0C]„œÜ[07]l„ã;›$[0C]Žß[00]Œ[04]Œœœ[0F]$Ÿò[1B]œgà[02]'[07]d;l‡ƒÿ[1C]{â#|˜[13]c‚c[0C]ì[1B]g[12]#ìä‡Ž[1C][00]ìl8x[04]ìgœœ|[12]Ãc[04][0C]ƒ[0C]Ü[04][0C]„`[02];ûgó|Ã[03]Ä’s#[0C][04]Œ [03]‡ƒ[1B]‚[13]ognìóogœì[1F]“#[13]ƒ[12]'og$g|ì[0C]ìo<c{ûgo[03]|›û[03]ä[02]Û›Œœ<[1B][1B]Û“ãgpã[03]lo<ädŒÇ$`[03]sûgœâ[00]„[0C]ìƒ[12][03]càÀÄŒ[04]ŒlãŒä[0C]Ç[1F]ŸŒŽ[1F]Ÿl$c[03][03]Ãƒ[03]l'œŒ`[03]dlŒc[1B]lŒ#ŒØà[12]ƒolœr›[03]ƒ[1B]gno„[04]Œ`;“[04][0C]ƒ[0C]„Äìäû„[04]ì[03]Û[04][04]r„œÜœœ|à[0C][04]$œ{ã[13]ƒ‚cÛ[04][04][02][04][0C][04][04][0C][04][0C]„`r›[0C][04][03][0C][04][0C][04][04]ì[04][0C]ä[1B]b›[0C][04][03][0C][04][0C][04][04]Œ[04][0C]„Ÿb[1B]o€[04][03][04][0C][04][04][0C][04]Œ„`<À[04]Œ[04][04][0C]ƒ[0C]„Äìäû[04]$l[04]ìŽ{ã[0C]d‡sŒ[04][0C]„[03]dÄ[0C][04][0C]sÄÜœœœ|à[0C]dl[04]äÇ[1C]Ÿ[0C]Ž$Œ[0C]„[03]Œ„[0C][04][04][03][04][0C][04][04][0C][04][0C]d[04][0C][04][0C][04][04][0C][04][0C][04][04]Œ`<€äÄ[0C][04][0C]ƒ[04]Œ„ìdœ|[07]›l[04]äÇ{ã[0C]„ì‡„[03][0C][1F][00][04][04][03][04][0C][04][04][0C][04][0C]d[04][0C][04][0C][04][04][0C][04][0C][04][04]Œ[03]lŽ[1F][00][0C][04][02][04][04][0C][04][0C][04][04]l[04][0C][04][04][0C][04][0C][04][04][0C][04]Œ |€Œ[1F][00][04][04]ããâãããã‚“[0C][0C][04][0C][04][0C][0C][04]l#[1C][00]Œ[03]ìŽ[1F][00][0C][04][02][04][0C][0C][04][0C][04]ll[04][0C][04][0C][0C][04][0C][04][0C]l[04]Œ[03][0C]Œ[04][0C][04][03][0C][04][0C][04][0C][0C][04]l[04][0C][0C][04][0C][04][0C][0C][04][0C][04]Œ`|€l„[0C][0C][04]‚[04]ŒŒÄ‡[1C][0C]Çƒ[0C]ä‡sã$lŽ[1C]€[0C]Œ[1B][0C]Œ[04][0C][04][03][0C][04][0C][04][0C][0C][04]l[04][0C][0C][04][0C][04][0C][0C][04][0C][04]Œ[13]$Œ[04][0C][0C][03][0C][04][0C][0C][04][0C][04]l[0C][04][0C][04][0C][0C][04][0C][04][0C][0C]Ä`|ƒŒŒ[04][0C][04];Œœœã[1C]Ã[13][04]‡ƒ[0C]ì[1C]œä‡sÜÄ‡[04]Œ[13]äŒ[04][0C][04][03][0C][04][0C][04][04][0C][04]l[04][04][0C][04][0C][04][04][0C][04][0C][04]Ä[12][04]‡[1E][00][04][0C][03][0C][04][04][0C][04][0C][04]$[0C][04][0C][04][04][0C][04][0C][04][04][0C]„`|‚$‡[1F][00][0C][04]{Œœœã[1C]ãl[04]‡‚[04]ì[1C]œä[1C]àä‡ÇÄ`[03]Ã[13]{rn[0C]Üûs[03]cŒ “g[04]œã[0E]ûgœã[00][04]Ÿgã[00]ìƒ[1B]s{o[04]œò;[03][03]ä[1B]ƒno„œ<?›oäûgçs[1B]ãoÜ[04]oì|›c„`[04][04][0C][1B]Ã[03]ogn[0C]Œds[13]ll[12]dlÇ[1B]óccbcžogn|[04]œâ[07]ûoœãsìƒ[12]s;gŸ<[03]sÛãb[12][03]ägûò#x›[13]ngœcàÀ[03]â#oóØc`à|Ã[03]äò'[1B][03]„„[04][1B]sƒ’[03][04]ßnã[00][04]Ÿãàƒ[02]dœgppò'|[1B]ƒ[03]l'oìŒ?ûd$ìŒc[03]ò$ldld$ldld$ldldll$ldll$ldll$ldll$ldll$ldll$ldll$ldll$ldll$ldll$Œ [1C][1C]ããâ[13][0C][0C]Ä[0C][04][0C]ÇÄŒäìóÄ[0C][04][0C][0C][04][0C][04]ì[0C][04][0C][04][0C][0C][04][0C]„[0C][0C][04][0C][04][0C][0C][04][0C]„[0C][0C]Ä`[1C][1C]ããã[12]d[0C]Œ[04][0C][04]‡ŒÄìäû[0C][04][0C][04][0C][0C][04][0C][04][0C][0C][04]‡„Œì$œ|[0F][03][0C][04][0C][04][0C][0C][04][0C][04][0C]Œ`[1C][1C]ããã[1B]l[03]Œ[0C][04][0C][04][0C][0C][04][0C][04][0C][0C]ÄœœœÜœœœœ[0C][04][0C][04][0C][04][04]l[04][0C][04][04][0C][04][0C][04][04][0C][04]Œ [1C][1C]âãã[13]Ä[02]„[0C][04][04]‡„Œ„[1C]Ø[04]‡[04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C]äŒÜœâ[1C]ƒ[1B][04][0C]„`[1C][1C]ãâã[1B]ä[03]Œ[04][0C][04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C]äŒœœâ[1C]ãl[04][0C][04][0C][04][04][0C]dì[03][04]Œ`[1C][1C]ããã[1B][0C]œÄ[0C][04][0C][04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C][04][0C][04]äŒœœœœ|Üd[0C][04]Ä`[1C][1C]ããã[13]dœ„[04][0C]äŒœœÜx[1C]Û[03][04][0C]l[04]$läb[04][04][0C][04][0C][04][04][0C]ä[02][04]Œäl„œÇ|[00][0C]„`[1C][1C]ãâã[1B]ŒœŒ[04][04][0C]Œ[04]ÄÜà[10]ò[04][04]Ç[04][0C][04][1C][18]c[02][04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C][04][0C][04][04][0C][04]Œ [1C][1C]ããâ[13][03][0C]„[0C][04][04]Ç[04][0C][04]‡pÇ$[0C]äŒÜœœp[0C]Ã[0C][04]ì„œÜœœàŽ[0C][0C]äŒœœÜœà‡[04][0C][0C]Ä`[1C][1C]ããã[12][13][0C]Œ[04][0C][04][0C][0C][04][0C][04][0C][0C][04]ŒŽìÇÄ‡pl[0C][04][0C][04][0C][0C][04]ŒŽ8[00][0C][04][0C][04][0C][0C][04][0C][04][0C]Œ`[1C][1C]ããã[1B]b[04]Œ[0C][04]ì„œÜœœà[0C][0C]äŒœœÜœà‡[04][0C][0C][04][0C][04][0C][0C][04][0C][04][0C]ÇÄ[0C][04]œÜ[1C][00][04]Œ`[1C][1C]âãã[13]{[0C]„[0C][0C]äŒœœÜx[1C]r[03][0C][0C][04][0C][04][0C][0C][04][0C][04]ìŒœœããlŒ[03][0C][04]‡ŒÄìcûlœ[0C]„`[1C][1C]ãâã[1B]ƒ[04]Œ[04][0C]äÄœœœÜàà[0C][04]ì„œœœœàà[04][0C][04][0C][04][04][0C][04][0C][04][0C]Œ[04] [... cut ...] ãããã[1B][03]„[0C][04][04]Ç[04][0C][04]älœ<[00][04][04][0C][04]Œ„âà[04]ŒœœÜœœœ[04][0C]d‡s[02][1B]“d‡à[02][03][0C]„`[03]ò$ldld$ldld$ldld$ldld$ldld$ldld$ldld$ldld$ldld$ldld$ldld$ldld$ldŒ “g[04]œã[0E]ûgœã[00][04]Ÿgã[00]ìƒ[1B]s{g[04]œó;[03][02]ä[1B]ƒng„œ|?›gäûgçs[12]ãoÜ[04]gì|ÛcÄ`[04][0C][0C][1B]‚[03]ogn[0C]Œd{[13]ld[13]dlŽ[1B]óccbcžogn<[04]œã[07]ûgœã{ìƒ[12]s;gŸ<[03]sÛãb[12][03]ìgûò#x›[13]ngœcàÀ[03]â#oóØc`à|Ã[03]äò'[1B][03]Ä„[04][1B]sƒ’[03][0C]ßoâ[00][04]Ÿãàƒ[02]dœgxpò'|[1B]ƒ[03]l'oìŒ~ûdlìŒc[03]`

chall3ng3r, 2018-08-06:
CPU frequency was set to 160MHz in HTTP samples, commented that line. But still same issue.

slaff, 2018-08-06:
> installed latest beta UDK.

May be that is the reason. We don't support Espressif's NON OS SDK > 2.0. Try with some older UDK version. 

> However, it disconnects automatically and reconnects to WiFi in 10 seconds time.
> After restarting few times with reset button on board, now it started to crash again. Here's the output from serial:

Let me see if I understood this correctly - the sample actually works but with frequent disconnects. And after multiple restarts it just stops working... That is utterly strange... Try an older UDK version and they we can talk. @ADiea what is the UDK version that works for you?

chall3ng3r, 2018-08-07:
Alright, that makes sense as upgraded ESP SDK 1.5 and Sming 2.x to 3.6.

Yes, you got it right. The sample flashed and worked, and then started crashing after some restarts. It never recover to working again, until I flashinit and flash again.

chall3ng3r, 2018-08-08:
Okay, I got it working perfectly now.

@slaff thank you for your help on this, highly appreciate it.

I uninstalled beta UDK, and installed ESP UDK 2.2.1 which contains ESP SDK 2.0. Afterwards, I did rebuild of Sming, and samples. esptool is different in this one, as write progress is shown differently.

I did make sure that before doing `flashinit`, I set SPI_SIZE=4M, so memory table is set properly. Tried Blink sample, it worked, then tried NetworkConfig sample which never worked after I updated Sming. It worked perfectly after I set SPI_SIZE=4M in its makefile, and flashed it.

Now going to continue some R&D on adding node-red support.

closed 2018-08-08 by chall3ng3r