SmingHub/Sming #1426: *** Preparation for Release 3.6.2 ***

slaff, 2018-08-06:
This issue is created to collect all the tasks we need to do before the 3.6.2.

### Source Code
- [ ] Update Sming/SmingCore/SmingCore.h
- [ ] Update Readme.md
- [ ] Merge develop into master
- [ ] Tag the source code in the master branch

### New features
TBD

### Bug fixes
TBD

All PRs scheduled for this release can be seen from [here](https://github.com/SmingHub/Sming/milestone/18)

kwis2, 2018-11-14:
I think it is high time to verify quickstart manual. It is very hard to start with Sming because of small mistakes in quickstart.
Linux:
First of all question which version of esp-open-sdk should be preferred. For now it is `VENDOR_SDK=1.5.4` isn't it high time to switch to 2.0.0, especially because of known WiFi bugs?
in Linux quickstart there are minor bugs, please verify if am I right:
Set ENV Variables there should be `export SMING_HOME=/opt/sming/Sming` instead of `export SMING_HOME=/opt/Sming/Sming `
In Get and Build Sming Core there is `cd $SMING_HOME/../..` while on fresh system, before we use `git clone ...` there is no `$SMING_HOME` dir 
Cloning Sming. question if there should be sming/Sming or Sming/Sming dir. Maybe it could be good idea to use command `git clone https://github.com/SmingHub/Sming.git sming`
Question about install esptool - it is already installed with  esp-open-sdk. Should it be installed once again?
Windows:
`choco install sming -y -force -source 'https://www.myget.org/F/sming/'` worked. But I received a lot of errors 

> …  *** ERROR ***  …  extraction failed
> … *** ERROR *** …  probable package conflict; existing file not overwritten

 so I am not sure if my install procedure didn’t failed - any solution?
Just in case I use Linux version.

`

slaff, 2018-11-15:
@mikee47 The release is scheduled for the end of the week. Do you have anything small to add or all new goodies will be for the next release?

mikee47, 2018-11-16:
@slaff That's it from me for this release

closed 2018-11-17 by slaff