SmingHub/Sming #1430: Windows Installation  - Package not found in source

orel-corporation-microbit, 2018-08-13:
Hi,

I previously was able to install and run the Sming framework on my machine around a year ago. I recently wanted to return to the programming of the ESP, so I cleared out the old version and am attempting to install the latest version of Sming.

I managed to install chocolatey, but every time I try to install Sming I get the below message:

"
2018-08-13 13:46:46,900 10200 [INFO ] - Installing the following packages:
2018-08-13 13:46:46,915 10200 [INFO ] - sming
2018-08-13 13:46:46,915 10200 [INFO ] - By installing you accept licenses for the packages.
2018-08-13 13:46:47,150 10200 [WARN ] - [NuGet] An error occurred while loading packages from 'https://chocolatey.org/api/v2/': Object reference not set to an instance of an object.
2018-08-13 13:46:47,150 10200 [WARN ] - [NuGet] An error occurred while loading packages from 'https://www.myget.org/F/sming/': Object reference not set to an instance of an object.
2018-08-13 13:46:47,150 10200 [ERROR] - sming not installed. The package was not found with the source(s) listed.
 Source(s): 'https://chocolatey.org/api/v2/;https://www.myget.org/F/sming/'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more 
 assistance.
2018-08-13 13:46:47,181 10200 [WARN ] - 
Chocolatey installed 0/1 packages. 1 packages failed.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).
2018-08-13 13:46:47,197 10200 [INFO ] - 
2018-08-13 13:46:47,197 10200 [ERROR] - Failures
2018-08-13 13:46:47,197 10200 [ERROR] -  - sming - sming not installed. The package was not found with the source(s) listed.
 Source(s): 'https://chocolatey.org/api/v2/;https://www.myget.org/F/sming/'
 NOTE: When you specify explicit sources, it overrides default sources.
If the package version is a prerelease and you didn't specify `--pre`,
 the package may not be found.
Please see https://chocolatey.org/docs/troubleshooting for more 
 assistance.
2018-08-13 13:46:47,197 10200 [DEBUG] - Sending message 'PostRunMessage' out if there are subscribers...
2018-08-13 13:46:47,212 10200 [DEBUG] - Exiting with 1
"
Tried explicitly calling the sources with no luck. Read through all the documentation I could find but still haven't found a solution to the problem.  The suggestions at https://github.com/chocolatey/choco/wiki/Troubleshooting#install-chocolateypath-doesnt-seem-to-work seemed to be my best bet, but none of the solutions seemed to address my problem.

Any support with the matter would be greatly appreciated.

slaff, 2018-08-13:
I guess the problem lies in choco

```
2018-08-13 13:46:47,150 10200 [WARN ] - [NuGet] An error occurred while loading packages from 'https://chocolatey.org/api/v2/': Object reference not set to an instance of an object.
2018-08-13 13:46:47,150 10200 [WARN ] - [NuGet] An error occurred while loading packages from 'https://www.myget.org/F/sming/': Object reference not set to an instance of an object.
```

Can you run `choco --v` and paste here the output?

orel-corporation-microbit, 2018-08-13:
Hi slaff,

The output from the logs are as follows:

"
2018-08-13 14:41:27,912 168 [INFO ] - ============================================================
2018-08-13 14:41:28,443 168 [INFO ] - Chocolatey v0.10.11
2018-08-13 14:41:28,474 168 [DEBUG] - Chocolatey is running on Windows v 6.3.9600.0
2018-08-13 14:41:28,474 168 [DEBUG] - Attempting to delete file "C:/ProgramData/chocolatey/choco.exe.old".
2018-08-13 14:41:28,474 168 [DEBUG] - Attempting to delete file "C:\ProgramData\chocolatey\choco.exe.old".
2018-08-13 14:41:28,506 168 [DEBUG] - Command line: "C:\ProgramData\chocolatey\choco.exe" --v
2018-08-13 14:41:28,506 168 [DEBUG] - Received arguments: --v
2018-08-13 14:41:28,646 168 [DEBUG] - RemovePendingPackagesTask is now ready and waiting for PreRunMessage.
2018-08-13 14:41:28,662 168 [DEBUG] - Sending message 'PreRunMessage' out if there are subscribers...
2018-08-13 14:41:28,677 168 [DEBUG] - [Pending] Removing all pending packages that should not be considered installed...
2018-08-13 14:41:28,787 168 [DEBUG] - Sending message 'PostRunMessage' out if there are subscribers...
2018-08-13 14:41:28,803 168 [DEBUG] - Exiting with 1
"
I do get a bunch of log4net warnings everytime I run a chocolately command though.

slaff, 2018-08-13:
The following error message hints to an error in .NET code.
```
2018-08-13 13:46:47,150 10200 [WARN ] - [NuGet] An error occurred while loading packages from 'https://chocolatey.org/api/v2/': Object reference not set to an instance of an object.
```
My bold guess is that the version of .NET that you have is too old. Try updating to a newer one. Also this problem is choco related and will be better addressed if you report it in the [choco issue list](https://github.com/chocolatey/choco/issues).

orel-corporation-microbit, 2018-08-13:
I don't think it's a .NET problem because I did check if it was the latest version and I apparently have version 4.7 involved.
[image]

Will look for help with this on the choco repo too. If you, or anyone, has any suggestions on getting this working please let me know though.

chall3ng3r, 2018-08-14:
choco install also failed for me, but it gets stuck. 

My manual solution is simple,
1. Install python 2.7
2. Install ESP UDK 2.2.1
3. Clone Sming dev or master branch to your desired location
4. Set environment vars

That's it, worked for me when I recently also returned to Sming after some months and needed to update to latest version.

closed 2018-11-07 by slaff