SmingHub/Sming #1432: error: too few arguments to function '_PyImport_FixupBuiltin'

DanielMazurkiewicz, 2018-08-18:
Have below issue while:
make VENDOR_SDK=2.2.1 STANDALONE=y

esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/python/python.c:1757:3: error: too few arguments to function '_PyImport_FixupBuiltin'

Any ideas?

DanielMazurkiewicz, 2018-08-18:
[INFO ]  Installing cross-gdb
[ERROR]    /home/ESP82xx/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/python/python.c:1757:3: error: too few arguments to function '_PyImport_FixupBuiltin'
[ERROR]    make[5]: *** [Makefile:2517: python.o] Error 1
[ERROR]    make[4]: *** [Makefile:8755: all-gdb] Error 2
[ERROR]    make[3]: *** [Makefile:845: all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 48:24.48)

slaff, 2018-08-26:
@DanielMazurkiewicz  The [esp-open-sdk project](https://github.com/pfalcon/esp-open-sdk) is the proper place to report and discuss this issue.

closed 2018-08-26 by slaff