SmingHub/Sming #1434: nRF24L01 example Fatal Exception 3, related to PROGMEM, PSTR

kmihaylov, 2018-08-27:
Hi again,

SDK: 2.0.0
Sming branch - develop

I'm trying to run the nRF24L01 example for a too much time already. Actually I'm able to run it if I comment the following lines (313-323) in RF24.cpp

```
print_address_register(PSTR("RX_ADDR_P0-1"),RX_ADDR_P0,2);
print_byte_register(PSTR("RX_ADDR_P2-5"),RX_ADDR_P2,4);
print_address_register(PSTR("TX_ADDR"),TX_ADDR);

print_byte_register(PSTR("RX_PW_P0-6"),RX_PW_P0,6);
print_byte_register(PSTR("EN_AA"),EN_AA);
print_byte_register(PSTR("EN_RXADDR"),EN_RXADDR);
print_byte_register(PSTR("RF_CH"),RF_CH);
print_byte_register(PSTR("RF_SETUP"),RF_SETUP);
print_byte_register(PSTR("CONFIG"),CONFIG);
print_byte_register(PSTR("DYNPD/FEATURE"),DYNPD,2);
```

I'm almost sure that the problem is related to the PSTR macro, unfortunately my knowledge isn't enough to solve the problem.

Any feedback is appreciated.

Thank you.

mikee47, 2018-09-11:
The 'name' parameter for print_byte_register() and print_address_register() is handled strangely. Internally the methods use strlen_P, but then pass name in the argument list for printf_P() in which case it MUST be a RAM pointer, not a flash pointer.

I'd call this a bug in RF24, prototypes should be:

  void print_address_register(PGM_P name, uint8_t reg, uint8_t qty);

note that PGM_P doesn't do anything special, it's just a hint that we're talking about a flash string pointer not a RAM pointer.

Implementation should be something like:

  char extra_tab = strlen_P(name) < 8 ? '\t' : 0;
  char buf[32]; // arbitrary length
  strcpy_P(buf, name); // would prefer to use strncpy_P but not (yet) available
  printf_P(PSTR(PRIPSTR"\t%c ="),buf,extra_tab);

slaff, 2018-09-19:
@mikee47 Can you suggest a PR with a fix for the problem once you have updated the Basic_ProgMem as part of PR #1438 and the PR has been merged?

mikee47, 2018-09-19:
@slaff Sure, simple enough - don't appear to be any external dependencies :-)

mikee47, 2018-09-21:
@slaff I've submitted a PR which should allow the sample to work, though I haven't had time to test it myself.

PR #1443 (2018-09-22): Fix issues with NRF24 library progmem string handling
Exception caused by passing progmem reference as argument to printf_P in the print_byte_register() and print_address_register() methods.
The methods have been revised to accept simply a const char* as advertised, not a PGM_P.

Also print_status() method uses printf_P() with RAM string - PSTR() wrapper missing on format argument.

Closes: #1434.
closed 2018-09-22 by slaff