SmingHub/Sming #1439: Any plans to support the new SDK 3.0.0

stokersoft, 2018-09-17:
Hi,

There are some severe bugs in all espressif SDK's less than 3.0.0. They are not compatible with some routers, like Jensen Omni and Nova mesh. Its fixed in 2.2.1 but, that version has some other flaws when downloading larger content via tcp.

Any plans to support this new SDK? First thing must be to let opensdk be based on 3.0.0. right?

slaff, 2018-09-17:
> First thing must be to let opensdk be based on 3.0.0. right?

Right. @stokersoft what you can do is create a PR in the [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) project with the 3.0.0 version or open an issue in that project.

vijaygill, 2018-09-19:
There is branch in a fork of the esp-open-sdk https://github.com/nabovarme/esp-open-sdk/tree/sdk-v3.0.0. I have not tried it myself (just found it after reading your post), might do it this weekend.

slaff, 2018-09-23:
@stokersoft You can try this unofficial and untested branch  (https://github.com/slaff/Sming/tree/update/sdk-3.0) which supports compilation with the v3.0 + updates of the NON-OS SDK. 

The commands to do this are:
```
cd /tmp
git clone -b update/sdk-3.0 https://github.com/slaff/Sming.git
cd Sming/Sming
export SMING_HOME=`pwd`
export SDK_BASE="$SMING_HOME/third-party/ESP8266_NONOS_SDK"
make 
```

Since you are switching between SDK versions you might need to issue this command, only once, before flashing:

```
make flashinit
```

Give it a try and let us know if it works for you and if there is more free heap.

vijaygill, 2018-09-24:
> @stokersoft You can try this unofficial and untested branch (https://github.com/slaff/Sming/tree/update/sdk-3.0) which supports compilation with the v3.0 + updates of the NON-OS SDK.
> 
@slaff - great stuff!!! If you need tester, I can be one :) Do you keep that branch up-to-date with the develop of main repo (I notice that you do, just asking)?

Vijay

slaff, 2018-09-24:
> great stuff!!! If you need tester, I can be one :)

@vijaygill Give it a try. As long as there is interest and progress I will keep it up to date until we merge it into develop at some later moment.

slaff, 2018-10-08:
With the new commits this branch of Sming (https://github.com/slaff/Sming/tree/update/sdk-3.0) adds support for SDK 3.0 for rBoot applications. So far all rBoot samples seem to work as expected.

PR #1470 (2018-10-12): Experimental support for SDK 3.0 for rBoot apps.
This PR is here to allow developers test Sming with NON-OS SDK 3.0. Although most of the things work it is not complete yet.

Things to do before declaring the PR ready for merging:
- [ ] (Optional) Non-rBoot application support is missing and is nice to have.

## Wanna test? 

Type the following commands

on Linux, MacOS and FreeBSD
```
cd $SMING_HOME
export SDK_BASE="$SMING_HOME/third-party/ESP8266_NONOS_SDK"
```

on Windows
```
cd %SMING_HOME%
set SDK_BASE="%SMING_HOME%/third-party/ESP8266_NONOS_SDK"
```

followed by the following commands (f [... cut]
closed 2018-10-12 by slaff