SmingHub/Sming #1444: AWS iot with Sming Certificate Problem

Vinimuller, 2018-09-24:
Hey! I'm trying to use AWS iot via MQTT, but it requires a certificate to make the connection. I have already used MQTT without certification, but I think that I'm having problems with the way that I'm passing the certificate and key to the function.
When I try to connect to the broker I get :

> please start sntp first !
Error: Invalid X509 ASN.1 file (X509 not ok)

Here is how I'm passing the certificate:

`	mqtt->addSslOptions(SSL_SERVER_VERIFY_LATER);

	unsigned int private_key_len = 1191, certificate_len = 861;
	unsigned char private_key[2000], certificate[2000];
	file_t cert,key;

	if(fileExist("cert.der")){

		cert = fileOpen("cert.der", eFO_ReadOnly);
		fileRead(cert,certificate,certificate_len);

	}else{
		Serial.println("CERT NOT FOUND");
	}

	if(fileExist("key.der")){
		
		key = fileOpen("key.der", eFO_ReadOnly);
		fileRead(key,private_key,private_key_len);

	}else{
		Serial.println("KEY NOT FOUND");
	}

	mqtt->setSslClientKeyCert(private_key, private_key_len,
							  certificate, certificate_len, NULL, true);

	fileClose(cert);
	fileClose(key);`

I'm running sming on Ubuntu16.04 and using ESP12F

slaff, 2018-09-25:
I guess your certificate is NOT in DER format. Open it with text editor and if you can read the content then it is in PEM (text format). Convert it to DER(binary format). Take a look here how this can be done: https://github.com/igrr/axtls-8266/blob/0c3a9f722f11799fbeda1f99f9d9ab77a82a4489/tools/make_certs.sh#L127

slaff, 2018-09-25:
> Do you know if I'm calling the functions in the right sequence?

Move the `connect` after you set the client certificate. In the code above this means to move `mqtt->connect("sdk-nodejs-2ada784d-7605-4f75-a3f2-c46dd3c84f40",true);` after `mqtt->setSslClientKeyCert...`

Vinimuller, 2018-09-25:
Moving the connect gives me the same output still.
Function:
````
void startMqttClient()
{
	procTimer.stop();

	if(!fileExist("cert.der")){
		Serial.println("cert not found");
	}

	if(!fileExist("key.der")){
		Serial.println("key not found");
	}
	
	mqtt->addSslOptions(SSL_SERVER_VERIFY_LATER);

	unsigned int private_key_len = fileGetSize("key.der"), certificate_len = fileGetSize("cert.der");
	unsigned char private_key[2000], certificate[2000];
	
	file_t certFile,keyFile;

	certFile = fileOpen("cert.der", eFO_ReadOnly);
	fileRead(certFile,certificate,certificate_len);
	
	keyFile = fileOpen("key.der", eFO_ReadOnly);
	fileRead(keyFile,private_key,private_key_len);

	mqtt->setSslClientKeyCert(private_key, private_key_len,
							  certificate, certificate_len, NULL, true);

	fileClose(certFile);
	fileClose(keyFile);

	mqtt->connect("sdk-nodejs-2ada784d-7605-4f75-a3f2-c46dd3c84f40",true);

	// Assign a disconnect callback function
	mqtt->setCompleteDelegate(checkMQTTDisconnect);
	mqtt->subscribe("thing/fish/teste");
}

```
Output:

> please start sntp first !
> please start sntp first !
> Error: Invalid X509 ASN.1 file (X509 not ok)
> MQTT Broker Unreachable!!
> 3fff1d00 already freed
>

slaff, 2018-09-25:
> Error: Invalid X509 ASN.1 file (X509 not ok)

Well make sure to upload the DER certificate and key to the device. You have to fix this issue before going to the next issues.  Take a look at the sample that we have: https://github.com/SmingHub/Sming/blob/develop/samples/MqttClient_Hello/app/application.cpp#L96.

Vinimuller, 2018-09-27:
Ok, I just realized that my version of of sming is not the same as the example.
The function "mqtt.setSslKeyCert" is actually called "mqtt.setSslClientKeyCert" and I don't have the setCallback function.

Do you think that the problem is that I'm not updated with the repo? Or the version that I am should work too?
Another question: Is it wrong to put the DER file into ESP in the compilation stage and read the file as I'm doing instead of using the "ssl/cert.h" and "ssl/private_key.h" headers with my certificate?

Vinimuller, 2018-09-30:
You can create a free account here https://aws.amazon.com/free/?nc1=h_ls.
It has some limitations but should work for exchanging some messages with aws iot.

In AWS iot each sensor that you have that is sending data to their broker is called a "thing". So in order to send data from a sensor to AWS iot, you first need the create this thing in aws iot console, which will give you the certificate and keys to access the broker. 

To create a thing you should go to:
console.aws.amazon.com -> IoT Core -> Onborad -> Configure a device.
Just give any name to the thing and you will receive the certificate and keys. The address to the broker you will find at settings in aws iot console.

If you think it's easier to use my certificate and keys tho access my broker I can send you an email with the files.

Vinimuller, 2018-09-30:
I was also wondering if it would be a problem the private key having a size of 1190 bytes.

Thanks for your help so far @slaff

slaff, 2018-10-01:
> If you think it's easier to use my certificate and keys tho access my broker I can send you an email with the files.

Nooo! Keep you private keys safe ;). 

> I was also wondering if it would be a problem the private key having a size of 1190 bytes.

Probably not.

Ok, I will try to find some time this week to send `mqtt` message to AWS IoT and share my progress with you. And thanks for the instructions.

Vinimuller, 2019-01-08:
@slaff I'm trying to solve this again and turns out that you were right, I my certificate wasn't in DER format hehe, sorry for that.
Now I'm only getting the following message after running the connect function.

> please start sntp first !

Any tips on that?

slaff, 2019-01-09:
>> please start sntp first !

> Any tips on that?

This is just a warning saying that you haven't set the actual time. If you want to do a full SSL validation you should have also set the time so that you can compare the time validity of the certificate that is sent to you.

Vinimuller, 2019-01-09:
Ok, makes sense. So to do a full validation I should first initialize a ntp server and then call startMqttClient?
Something like this?
> void onNtpReceive(NtpClient& client, time_t timestamp)
> {
> 	Serial.println("\n=== ON NTP RX ===");
> 	
> 	SystemClock.setTime(timestamp, eTZ_UTC); //System timezone is LOCAL so to set it from UTC we specify TZ
> 	ntpClient->setAutoQueryInterval( 3600 );
> 
> 	startMqttClient();
> }

Vinimuller, 2020-05-11:
@slaff 
Found on AWS documentation that they offer a C++ lib to implement AWS iot. It is made thinking about porting to different platforms, so I think it would be the case to make it work with sming. I would like to contribute with this, but it looks way easier to someone that has already worked with openSSL and has more experience on C++. What do you think?

Here is the link to the lib: https://github.com/aws/aws-iot-device-sdk-cpp/tree/release

slaff, 2020-11-09:
@Vinimuller Take a look at the sample in this PR: https://github.com/SmingHub/Sming/pull/2149.

closed 2021-02-04 by slaff