SmingHub/Sming #1449: NTPClient not retrying synchronization if DNS resolution fails

lrodorigo, 2018-09-29:
Hi all,
I noticed that the NTP synchronization is not retrying if the DNS resolution fails.

I solved by moving `timeoutTimer.startOnce()` before `dns_gethostbyname`  instead of in `internalRequestTime` method.:

`

	// Start timeout timer, if no response is recieved within NTP_RESPONSE_TIMEOUT
	// a new request will be sent.
	timeoutTimer.startOnce();

	int result = dns_gethostbyname(this->server.c_str(), &resolvedIp, staticDnsResponse, (void*)this);

`

This seems to solve the issue. I don't know if it's the right thing to do.

slaff, 2018-09-29:
Hm... this is more of a workaround than a proper fix. What we can do is probably improve the NtpClient to 
* listen to Wifi connects and disconnects events.
* if there is a WIFI connection but the sntp host is not resolvable (due to missing connection to Internet or timeout) to run a timer that retries the DNS resolution...

slaff, 2018-09-29:
Revert your changes to the file. Replace the NtpClient::staticDnsResponse method with the one below and give it a try.  
```
void NtpClient::staticDnsResponse(const char* name, LWIP_IP_ADDR_T* ip, void* arg)
{
	// DNS has been resolved

	NtpClient* self = (NtpClient*)arg;

	if(ip != NULL) {
		// We do a new request since the last one was never done.
		self->internalRequestTime(*ip);
	}
	else {
		timeoutTimer.startOnce();
	}
}
```

lrodorigo, 2018-09-30:
I think it should be `self->timeoutTimer.startOnce();`, but unfortunately this doesn't seems to fix the issue (`staticDnsResponse` not invoked at all).

slaff, 2018-10-01:
> I think it should be self->timeoutTimer.startOnce();

You are absolutely right. I answered hastily without really checking. Can you give me hints  how I can reproduce the problem that you are having?

lrodorigo, 2018-10-02:
Simply writing a program that waits for NTP sync before starting to work
(eg. a led blinks in the callback of the NTP sync) .

Disconnect your router from the internet, power on the ESP, re-connect your
router to internet and check if the timeout timer fires and then the sync
callback is invoked.

Thanks

slaff, 2018-10-11:
@lrodorigo Can you test PR #1482?

mikee47, 2018-10-11:
The PR doesn't address this particular issue (I wrote it a couple of months ago). Taking another look, it's hard to see exactly what's going on with 3 timers so I ditched two of them and added some debug statements. (Note that all the timers invoked `requestTime` anyway). Now I can see what's going on you had the right idea @lrodorigo, the latest commit should fix the problem. Let us know how you get on with that?

slaff, 2018-10-12:
@lrodorigo can you check PR #1482 ? I am waiting for your results in order to merge the PR.

PR #1482 (2018-10-12): Various improvements to NtpClient class
Tested with SystemClock_NTP sample, but note that time will be wrong as we don't yet have any timezone management in place. The sample also sets local time, but NTP provides time in UTC. No point in fixing that yet as without timezone it just makes things worse.

* Fix #include guard name
* Tidy #includes
* Put NTP_DEFAULT_SERVER into Flash
* Use const String& for parameters
* Use SimpleTimer instead of Timer as appropriate (efficiency, saves RAM)
* timeoutTimer doesn't need to be initialised until a request has been queued, so moved out of constructor
* [... cut]