SmingHub/Sming #1451: Strange bug with Strings

slaff, 2018-10-01:
@mikee47 Strangely enough the following code produces incorrect result:

```
http_method_str(request->method) + String(" ") + request->uri.getPathWithQuery() + " HTTP/1.1\r\n"
```

Instead of the complete HTTP request line I am getting just "HTTP/1.1\r\n". The issue is happening after your recent PRs were merged (related to the String class). Can you check if you can reproduce it or it is just me and my compiler with `mforce-l32`?

slaff, 2018-10-01:
I narrowed down the problem to the following code:
```
        String path = "/path/to";
	String query;

	if(path.length() + query.length() > 0) {
		String tmpString = path + query;
		debugf("[%s]", tmpString.c_str());
	}
```

Expected output: [/path/to].
Actual output: [(null)]

Which is not correct and might break the code in multiple places.
@mikee47 when you have time please submit a fix for this issue.

mikee47, 2018-10-01:
@slaff That is weird. I tested it with:
```
String content = http_method_str(request.method) + String(" ") + request.uri.pathWithQuery() + " HTTP/1.1\n";
debug_i("request: %s", content.c_str());
```

And it's fine.

mikee47, 2018-10-01:
Fix is in WString.cpp, swap lines 279 & 280 to read:
```
  if (length == 0) return true;
  if (!cstr) return false;
```

I'll review the rest of it see if there are any similar conditions then submit a PR.

mikee47, 2018-10-01:
diff --git a/Sming/Wiring/WString.cpp b/Sming/Wiring/WString.cpp
index 6a87294f..c6e5e873 100644
--- a/Sming/Wiring/WString.cpp
+++ b/Sming/Wiring/WString.cpp
@@ -276,8 +276,8 @@ bool String::concat(const String &s)
 bool String::concat(const char *cstr, unsigned int length)
 {
   unsigned int newlen = len + length;
-  if (!cstr) return false;
   if (length == 0) return true;
+  if (!cstr) return false;
   if (!reserve(newlen)) return false;
   memmove(buffer + len, cstr, length);
   buffer[newlen] = '\0';

mikee47, 2018-10-02:
Another thing, which may be of concern. As an example, consider:
```
String tmp = String("test ") + String(54) + String(" again") + anotherString;
```
If any of the allocations in the expression fail, then the result, tmp, should be an invalid string. However, with the changes I made this will not happen and the result will be unpredictable.

Thoughts?

slaff, 2018-10-02:
> result will be unpredictable.

Unpredictable results is bad and is completely negating any good results of not setting initially the string to be "" instead of nullptr.  Especially for a class that is one of the most used core classes.

Mike can you make another PR, alongside the current PR #1452, where you change the code so that a new String object has always "" instead of nullptr + change accordingly concat and the other methods.

mikee47, 2018-10-02:
That's unpredictable in the event of heap allocation failure - #969 addresses this in a wider context.
With the original unmodified version of String, the result would be empty strings being used, which I guess is a little more predictable, though still undesirable, but still requires the caller to check that a String is valid before using it; I cannot recall seeing that done anywhere.

@slaff I could do with some steering on how Sming should deal with heap allocation failures.

slaff, 2018-10-02:
@mikee47 Can you rephrase and add 

`If any of the allocations in the expression fail, then the result, tmp, should be an invalid string. However, with the changes I made this will not happen and the result will be unpredictable.` 

as comment in the code of PR #1452 that mentions this unlikely-but-possible situation.

PR #1452 (2018-10-03): Fix/string concat
Fix inconsistent behaviour in String::concat(). Fixes #1451

* Result String should only be invalidated if a memory allocation error occurs
* An invalid string is considered equivalent to an empty string
closed 2018-10-03 by slaff