SmingHub/Sming #1453: Building in linux

zdebel, 2018-10-02:
Tried building sming in linux because I didn't have good experience with the framework on windows (had to go the manual route, keep getting crashes on samples when I restart the board). This is what I've got:
`CC system/flashmem.c
In file included from include/user_config.h:33:0,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
system/include/esp_systemapi.h:48:13: error: conflicting types for 'ets_timer_arm_new'
 extern void ets_timer_arm_new(ETSTimer *ptimer, uint32_t milliseconds, bool repeat_flag, int isMstimer);
             ^
In file included from system/include/esp_systemapi.h:7:0,
                 from include/user_config.h:33,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
/opt/esp-open-sdk/sdk/include/osapi.h:65:6: note: previous declaration of 'ets_timer_arm_new' was here
 void ets_timer_arm_new(os_timer_t *ptimer, uint32_t time, bool repeat_flag, bool ms_flag);
      ^
In file included from include/user_config.h:33:0,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
system/include/esp_systemapi.h:57:13: error: conflicting types for 'ets_delay_us'
 extern void ets_delay_us(uint32_t us);
             ^
In file included from system/include/esp_systemapi.h:7:0,
                 from include/user_config.h:33,
                 from system/flashmem.h:11,
                 from system/flashmem.c:1:
/opt/esp-open-sdk/sdk/include/osapi.h:33:6: note: previous declaration of 'ets_delay_us' was here
 void ets_delay_us(uint16_t us);
      ^
Makefile:568: recipe for target 'out/build/system/flashmem.o' failed
make[1]: *** [out/build/system/flashmem.o] Error 1
make[1]: Leaving directory '/opt/Sming/Sming'
Makefile:512: recipe for target 'reload' failed
make: *** [reload] Error 2`

slaff, 2018-10-03:
Can you output your ESP_HOME and SMING_HOME env. variables? 
```
echo $SMING_HOME
echo $ESP_HOME
```

zdebel, 2018-10-03:
`zdebel@DESKTOP-85KGDT2:~$ echo $SMING_HOME
/opt/Sming/Sming
zdebel@DESKTOP-85KGDT2:~$ echo $ESP_HOME
`/opt/esp-open-sdk`

I was following https://github.com/SmingHub/Sming/wiki/Linux-Quickstart

Browsing through various issues I managed to find info that when building esp-open-sdk I had to use the 2.0.0 version and indeed it helped with this issue. Going through the examples, all seem to work properly.

slaff, 2018-10-04:
> Going through the examples, all seem to work properly.

If the issue is solved can you please close it?

zdebel, 2018-10-04:
No problem, although I think a doc update would be needed so people don't have this issue in the future.

closed 2018-10-04 by zdebel

slaff, 2018-10-04:
> No problem, although I think a doc update would be needed so people don't have this issue in the future.

Thanks for noticing this.  I have added the following text changes: https://github.com/SmingHub/Sming/wiki/Linux-Quickstart/_compare/21a88a0d35050bffb2b4aa47878819fb52326546...b3e4f910b6b169e2894e1a92c53bbdd6a36567ca