SmingHub/Sming #1454: max macro - compiler errors

frankdownunder, 2018-10-03:
My source file has 4 lines:
#include <user_config.h>
#include <SmingCore/SmingCore.h>
#include <Libraries/OneWire/OneWire.h>
#include <algorithm>

It doesnt build.

Errors are:

In file included from /mnt/c/opt/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.5/bits/stl_algo.h:60:0,
                 from /mnt/c/opt/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.5/algorithm:62,
                 from app/temperatureDS18B20.cpp:12:
/mnt/c/opt/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.5/bits/algorithmfwd.h:360:41: error: macro "max" passed 3 arguments, but takes just 2
     max(const _Tp&, const _Tp&, _Compare);

frankdownunder, 2018-10-03:
Workaround is to rearrange the headers and to undefine the max and min macros
````
#include <user_config.h>
#include <SmingCore/SmingCore.h>
#include <algorithm>
#include <Libraries/OneWire/OneWire.h>
#undef max
#undef min
````

mikee47, 2018-10-04:
@frankdownunder could you see if the PR fixes your issue?

frankdownunder, 2018-10-05:
I would try it out but I couldnt find the code - have you done a push?
The latest commit on https://github.com/mikee47/Sming was a lot earlier than the commit referenced above

mikee47, 2018-10-05:
Sorry, I'm a bit clumsy with git still! I removed that branch and created a new one for the PR - hopefully you've found it.

PR #1458 (2018-10-05): Better handling of min() and max() in C++
Fixes #1454.
closed 2018-10-05 by slaff