SmingHub/Sming #1464: TcpServer in TLS mode leaks memory

Informatic, 2018-10-07:
Hey!

Every connection to TcpServer running in SSL/TLS mode seems to leak approximately 7344 bytes of memory.

I've looked around, and it seems like `TcpConnection::closeSsl` only frees `sslContext`, which seems to be a `nullptr` when `TcpConnection` is instantiated from `TcpServer`. Adding `ssl_free(ssl);` in `TcpConnection::closeSsl` seems to reduce leaking to less than 100 bytes per request, though I'm not sure if it won't break anything else.

Very minimal example: https://git.io/fxs8O

slaff, 2018-10-07:
@Informatic TcpServer running in SSL/TLS mode is not officially supported. In server mode the server keeps SSL session data to allow SSL session resumption (https://github.com/SmingHub/Sming/blob/develop/Sming/SmingCore/Network/TcpServer.cpp#L99). Thus it may seem that it leaks memory.  Also the heap consumption will go up and down after some seconds or minutes down due to the TCP stack. 
With that said you can try to set the `sslSessionCacheSize` to 0 and see if something changes.

Informatic, 2018-10-07:
Ah, this makes sense. So far the issue was that even that minimal TcpServer in SSL/TLS just crashed after 5 requests. I'll check if changing `sslSessionCacheSize` works for me later today.

slaff, 2018-11-19:
@Informatic I will close the ticket for now. If you have more information the feel free to post it in the ticket and to reopen it, if needed.

closed 2018-11-19 by slaff