SmingHub/Sming #1465: error: 'unhex' redeclared

frankdownunder, 2018-10-07:
third-party/http-parser/http_parser.c:230:21: error: 'unhex' redeclared as different kind of symbol
 static const int8_t unhex[256] PROGMEM_L32 =
                     ^
In file included from system/include/esp_systemapi.h:23:0,
                 from include/user_config.h:33,
                 from third-party/http-parser/http_parser.c:23:
system/include/stringutil.h:47:27: note: previous definition of 'unhex' was here
 static inline signed char unhex(char c)

I think it is because of this new code which went into http_parser.c
#ifdef __ets__
    #include <user_config.h>
#endif

mikee47, 2018-10-07:
http parser needs re-patching. Delete third-party/http_parser then make should pull it back in and apply the new patch.

slaff, 2018-10-08:
@frankdownunder Make dist-clean or delete the  `third-party/http_parser` directory.

```
cd $SMING_HOME
make dist-clean
```

or as @mikee47 suggested just delete the `third-party/http_parser` directory and the new patch will be applied correctly.

closed 2018-10-09 by slaff