SmingHub/Sming #1501: Notice: GitHub is unstable - No new PR will be merged...

slaff, 2018-10-22:
GitHub seems to have [problems](https://developers.slashdot.org/story/18/10/22/0437245/githubs-website-remains-broken-after-a-data-storage-system-failed-earlier-today)

No new PR will be merged until those problems are fixed and we are sure that only the latest version of a PR will get merged.

slaff, 2018-10-22:
@mikee47 the GitHub servers seem to be stable again. Can you try again to send the latest changes to your PRs and/or re-build again Travis and Appveyor builds?

mikee47, 2018-10-22:
@slaff I've pushed a change to #1498. I've only got Travis set up on my fork and it's failing to start.

mikee47, 2018-10-22:
Looks like Travis is working again, bless him :-)

closed 2018-10-23 by slaff