SmingHub/Sming #1517: Serial.commandProcessing(true) has no effect

ematt, 2018-11-06:
Hi,

There is a strange bug that prevents the commandHandler to get any character from Serial unless Serial.setCallback(nullptr); is called. 

```
#include <user_config.h>
#include "../include/user_config.h"

#include "Services/CommandProcessing/CommandProcessingIncludes.h"

#include <Debug.h>

#include <SmingCore/SmingCore.h>
#include <HardwarePWM.h>

extern "C" {
  #include "user_interface.h"
}

// Will be called when WiFi hardware and software initialization was finished
// And system initialization was completed
void ready()
{
	debugf("READY!");

}

void init()
{
    wifi_set_sleep_type(NONE_SLEEP_T);
    System.setCpuFrequency(eCF_160MHz);

    Debug.start();

    Serial.begin(SERIAL_BAUD_RATE);
    Serial.systemDebugOutput(true); // Allow debug print to serial
    Serial.println("Sming. Let's do smart things!");

    commandHandler.setCommandEOL('\n');

    Serial.commandProcessing(true);

    //Comment this line
    Serial.setCallback(nullptr);

    Debug.setDebug(Serial);
    Debug.initCommand();
    Debug.start();

	// Set system ready callback method
	System.onReady(ready);
}
```

slaff, 2018-11-07:
@mikee47 Can you take a look at this issue?

mikee47, 2018-11-13:
I've been unable to replicate this problem, using the current `Develop` branch.
Tested by pasting the above code into the `CommandProcessing_Debug` sample and it works as expected.
Also, looking at the code I can see no reason why `setCallback(nullptr)` would have the indicated effect.

Can you elaborate at all?

closed 2018-11-17 by slaff