SmingHub/Sming #1518: Faild to run project

Evgsurf, 2018-11-13:
Hello, 
I found my dream! ESP8266 with C++!
But, I can't run any project from samples :( 

I use UBUNTU 18.04 64bit.
I have installed esp-open-sdk, and I were able to build, flash and run projects.

So, I clone Sming and try to run Blink project. (some problems with compilation, but "issues" helped me)
I have 3 modules esp8266 and no one starts ...

Try Blink ...

Big NODEMCU
[image]
make flashinit -> OK
make flash
```
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.8 kbit/s)...
Writing 229376 @ 0x2000... 229376 (100 %)
Wrote 229376 bytes at 0x2000 in 19.9 seconds (92.3 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Booting rom 0.
emapg
map 0 err
system_partition_table_regist: failed
size_map = 0
partition[0]: 1, 0x00000000, 0x00001000
partition[1]: 100, 0x00001000, 0x00001000
partition[2]: 101, 0x00002000, 0x000fe000
partition[3]: 4, 0x0007b000, 0x00001000
partition[4]: 5, 0x0007c000, 0x00001000
partition[5]: 6, 0x0007d000, 0x00003000
```

Mini with "silver" chip 
[image]

make flashinit -> OK
make flash
```
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (90.0 kbit/s)...
Writing 229376 @ 0x2000... 229376 (100 %)
Wrote 229376 bytes at 0x2000 in 19.8 seconds (92.5 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
Fatal exception (28): 
epc1=0x4000228b, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000b5, depc=0x00000000
```

Mini with "black" chip 
[image]
make flashinit -> OK
make flash
```
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0000
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.7 kbit/s)...
Writing 229376 @ 0x2000... 229376 (100 %)
Wrote 229376 bytes at 0x2000 in 19.9 seconds (92.3 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
de:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Writing default boot config.
Booting rom 0.
emapg
map 0 err
system_partition_table_regist: failed
size_map = 0
partition[0]: 1, 0x00000000, 0x00001000
partition[1]: 100, 0x00001000, 0x00001000
partition[2]: 101, 0x00002000, 0x000fe000
partition[3]: 4, 0x0007b000, 0x00001000
partition[4]: 5, 0x0007c000, 0x00001000
partition[5]: 6, 0x0007d000, 0x00003000
```

I have no ideas what to do ...

mikee47, 2018-11-13:
Try adding these lines to the project's `Makefile-user.mk` file:

```
SPI_SPEED = 80
SPI_MODE = dio
SPI_SIZE = 4M
```

These work for me with my NodeMCU. The crucial one is `SPI_SIZE` which I suspect is causing your problems.

Evgsurf, 2018-11-13:
Thank you very much! All three board started. 
Whith some trash 
```
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x024f
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.6 kbit/s)...
Writing 229376 @ 0x2000... 229376 (100 %)
Wrote 229376 bytes at 0x2000 in 19.9 seconds (92.3 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
fault boot config.
Booting rom 0.
ota1 not set
ota2 not set
>> 0x3&J,R�,���␃�␒�o�;��g|�␌l␄ld$`␃␜#�␛␂␄�|␃{�␃$�n�␄�o�␀␄l`␃��s�l�l␛�␌␄␌l`␃��;�d�d␒�␄␌␌$`␃��s�l␄��␀�␌␌$l ␃��␛�'�␂'␇d;l�DK ver: 3.0.0(e27cf60) compiled @ Sep  5 2018 15:35:05
phy ver: 1136_0, pp ver: 10.2

mode : softAP(82:7d:3a:6e:3e:3b)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

```
, but started. I can try more now.

closed 2018-11-13 by Evgsurf

mikee47, 2018-11-13:
Add `COM_SPEED_SERIAL = 74880` will get rid of the trash...

Evgsurf, 2018-11-13:
> Add `COM_SPEED_SERIAL = 74880` will get rid of the trash...
No luck

mikee47, 2018-11-13:
Oh, OK. I get trash too, funny I hadn't noticed that before!