SmingHub/Sming #1520: problem with OTA in Basic_rBoot

petrynchyn, 2018-11-13:
I have problem with OTA in sample Basic_rBoot, please check someone this example in yourself.

```
make flash
Killing Terminal to free /dev/ttyUSB0
/opt/Sming/Sming/Makefile-rboot.mk:614: recipe for target 'flash' failed
make: [flash] Вбито (ігнорується)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/rboot.bin 0x02000 out/firmware/rom0.bin 0x100000 out/firmware/spiff_rom.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v2.2-dev
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 2032 bytes to 1429...
Wrote 2032 bytes (1429 compressed) at 0x00000000 in 0.1 seconds (effective 122.6 kbit/s)...
Hash of data verified.
Compressed 319168 bytes to 220389...
Wrote 319168 bytes (220389 compressed) at 0x00002000 in 19.5 seconds (effective 131.1 kbit/s)...
Hash of data verified.
Compressed 65536 bytes to 191...
Wrote 65536 bytes (191 compressed) at 0x00100000 in 0.0 seconds (effective 22285.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
��␃�␛�g�s��o|�␌l␌lll ␃��{�l��: 079069 trying to mount spiffs at 0x00100000, length 65536
79378 fs.start: size:64 Kb, offset:0x100000

81006 mount res: 0

81624 Station configuration is: MKVL

Currently running rom 0.
Type 'help' and press enter for instructions.

mode : sta(60:01:94:2b:ae:0b)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with MKVL, channel 2
dhcp client start...
ip:192.168.1.69,mask:255.255.255.0,gw:192.168.1.1
pm open,type:2 0
Updating...
147084586 current_rom = 0
147084776 slot = 1
147085052 Download file:
    (0) http://192.168.1.223/fw/Basic_rBoot/rom0.bin -> 202000
147086789 Creating new httpConnection
147089702 HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
147096029 HttpConnection::connecting ...
147099673 +TCP connection
147101829 connect to: 192.168.1.223
147105682 TcpConnection::connect result:, 0
147108828 Download file:
    (0) http://192.168.1.223/fw/Basic_rBoot/spiff_rom1.bin -> 300000
147117582 HttpConnection::connect: TCP state: 2, isStarted: 1, isActive: 1
147223402 OnConnected
147223584 TCP connected
147223709 HttpConnection::onReadyToSendData: waitingQueue.count: 2
147224162 Storing 39 bytes in stream
147225781 realloc 39 -> 189
147227903 Storing 21 bytes in stream
147231156 Storing 19 bytes in stream
147234327 Storing 2 bytes in stream
147237458 onReadyToSendData: 0
147240279 Written: 81, Available: 81, isFinished: 1, PushCount: 1 [TcpBuf: 2699]
147247159 TcpClient stream finished
147262322 TCP sent: 81
147262545 HttpConnection::onReadyToSendData: waitingQueue.count: 1
147262960 Storing 45 bytes in stream
147263315 realloc 45 -> 195
147264858 Storing 21 bytes in stream
147268105 Storing 19 bytes in stream
147271283 Storing 2 bytes in stream
147274392 onReadyToSendData: 2
147277192 Written: 87, Available: 87, isFinished: 1, PushCount: 1 [TcpBuf: 2693]
147284109 TcpClient stream finished
147289608 The headers are complete

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 1344, room 16 
tail 0
chksum 0x9c
load 0x3ffe8000, len 660, room 8 
tail 12
chksum 0xbd
csum 0xbd

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
��␃�␛�o�s��n|�␄$␌lll`␃��s�d��: 086188 trying to mount spiffs at 0x00100000, length 65536
86496 fs.start: size:64 Kb, offset:0x100000

88125 mount res: 0

88743 Station configuration is: MKVL

Currently running rom 0.
Type 'help' and press enter for instructions.

mode : sta(60:01:94:2b:ae:0b)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with MKVL, channel 2
dhcp client start...
ip:192.168.1.69,mask:255.255.255.0,gw:192.168.1.1
pm open,type:2 0

SDK: v2.0.0(5a875ba)
Firmware version: v18.1113.1325
Free Heap: 47888
CPU Frequency: 80 MHz
System Chip ID: 2bae0b
SPI Flash ID: 1640ef
SPI Flash Size: 4194304
561879786 Count: 2
561881383 ROM 0: 8192
561883290 ROM 1: 2105344
561885463 ROM 2: 0
561887115 GPIO ROM: 0

```

slaff, 2018-11-21:
@petrynchyn I see some problems in the areas that are allocated for the ROMS and the actual location where they are downloaded. Can you check with
```
make clean
make 
make flashinit
make flash
make flashconfig
```

And see if it works for you. The last command `make flashconfig` takes care to reset the rboot configuration so that a correct one will be generated on the next boot.

petrynchyn, 2018-11-26:
result is similar, I will try later on another ESP module

etmmahi, 2018-11-26:
Unfortunately I see the Basic_rBoot OTA also failing, even with the make flashconfig

petrynchyn, 2018-11-26:
Problems begin at the stage of establishing a connection with WiFi. I can connect to WiFi only after flash  sample Basic_WiFi and after connecting with WiFi i flash Basic_rBoot

```
petrynchyn@ubuntu:/mnt/hgfs/Projects/Basic_rBoot$ make clean
petrynchyn@ubuntu:/mnt/hgfs/Projects/Basic_rBoot$ make DEBUG_VERBOSE_LEVEL=3
make[1]: Entering directory '/home/petrynchyn/Sming/Sming/Sming'
make[1]: Leaving directory '/home/petrynchyn/Sming/Sming/Sming'
OC out/build/libmain2.a
make -C /home/petrynchyn/Sming/Sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0 RBOOT_SILENT=0
make[1]: Entering directory '/home/petrynchyn/Sming/Sming/Sming/third-party/rboot'
CC rboot-stage2a.c
LD /mnt/hgfs/Projects/Basic_rBoot/out/build/rboot-stage2a.elf
E2 /mnt/hgfs/Projects/Basic_rBoot/out/build/rboot-hex2a.h
CC rboot.c
LD /mnt/hgfs/Projects/Basic_rBoot/out/build/rboot.elf
E2 /mnt/hgfs/Projects/Basic_rBoot/out/firmware/rboot.bin
make[1]: Leaving directory '/home/petrynchyn/Sming/Sming/Sming/third-party/rboot'
CC /home/petrynchyn/Sming/Sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
CC /home/petrynchyn/Sming/Sming/Sming/third-party/rboot/appcode/rboot-api.c
CC /home/petrynchyn/Sming/Sming/Sming/appspecific/rboot/overrides.c
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8371|     881
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFE8DA4|    2596
       bss|      Uninitialized Data (RAM)|    3FFE8DA8|    3FFEF850|   27304
      text|            Cached Code (IRAM)|    40100000|    401064AB|   25771
irom0_text|           Uncached Code (SPI)|    40202010|    40248C8C|  289916
Total Used RAM : 30781
Free RAM : 51139
Free IRam : 7015
E2 out/firmware/rom0.bin
Checking for spiffs files
files directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x10000 (65536) bytes.
Adding files in directory 'files'.
Added 'testfile.txt' to spiffs (40 bytes).
petrynchyn@ubuntu:/mnt/hgfs/Projects/Basic_rBoot$ make flashinit
Flash init data default and blank data.
/home/petrynchyn/Sming/esp-toolkit/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 erase_flash
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Erasing flash (this may take a while)...
Erase took 7.1 seconds
/home/petrynchyn/Sming/esp-toolkit/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 32m 0x3fc000 /home/petrynchyn/Sming/esp-toolkit/sdk/bin/esp_init_data_default.bin 0x3fe000 /home/petrynchyn/Sming/esp-toolkit/sdk/bin/blank.bin 0x100000 /home/petrynchyn/Sming/Sming/Sming/compiler/data/blankfs.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Writing 4096 @ 0x3fc000... 4096 (100 %)
Wrote 4096 bytes at 0x3fc000 in 0.4 seconds (89.2 kbit/s)...
Writing 4096 @ 0x3fe000... 4096 (100 %)
Wrote 4096 bytes at 0x3fe000 in 0.4 seconds (89.2 kbit/s)...
Writing 16384 @ 0x100000... 16384 (100 %)
Wrote 16384 bytes at 0x100000 in 1.4 seconds (91.8 kbit/s)...
Leaving...

petrynchyn@ubuntu:/mnt/hgfs/Projects/Basic_rBoot$ make flash
Killing Terminal to free /dev/ttyUSB0
/home/petrynchyn/Sming/Sming/Sming/Makefile-rboot.mk:632: recipe for target 'flash' failed
make: [flash] Killed (ignored)
/home/petrynchyn/Sming/esp-toolkit/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm qio -fs 32m 0x00000 out/firmware/rboot.bin 0x02000 out/firmware/rom0.bin 0x100000 out/firmware/spiff_rom.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.2 kbit/s)...
[... 58 lines cut ...]
tail 0
chksum 0x9c
load 0x3ffe8000, len 660, room 8
tail 12
chksum 0xbd
csum 0xbd

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
��␃�␒�g�s��o|�␌l␌lld`␃��{�l��: 085837 trying to mount spiffs at 0x00100000, length 65536
86146 fs.start: size:64 Kb, offset:0x100000

87776 mount res: 0

Currently running rom 0.
Type 'help' and press enter for instructions.

mode : null

```

mikeg197, 2018-12-10:
Ive got the same problem. Did you found any fix for this? 

RBOOT_ENABLED ?= 1
RBOOT_BIG_FLASH ?= 1
SPI_SIZE        ?= 4M
SPIFF_SIZE      ?= 65536

```
mike@mike-desktop > /opt/Sming/samples/Basic_rBoot $ make clean
mike@mike-desktop > /opt/Sming/samples/Basic_rBoot $ make 
OC out/build/libmain2.a
make -C /opt/Sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0 RBOOT_SILENT=0
make[1]: Verzeichnis „/opt/Sming/Sming/third-party/rboot“ wird betreten
CC rboot-stage2a.c
LD /opt/Sming/samples/Basic_rBoot/out/build/rboot-stage2a.elf
E2 /opt/Sming/samples/Basic_rBoot/out/build/rboot-hex2a.h
CC rboot.c
LD /opt/Sming/samples/Basic_rBoot/out/build/rboot.elf
E2 /opt/Sming/samples/Basic_rBoot/out/firmware/rboot.bin
make[1]: Verzeichnis „/opt/Sming/Sming/third-party/rboot“ wird verlassen
CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c
CC /opt/Sming/Sming/appspecific/rboot/overrides.c
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8371|     881
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFE8DB0|    2608
       bss|      Uninitialized Data (RAM)|    3FFE8DB0|    3FFEF850|   27296
      text|            Cached Code (IRAM)|    40100000|    401064AB|   25771
irom0_text|           Uncached Code (SPI)|    40202010|    40247B60|  285520
Total Used RAM : 30785
Free RAM : 51135
Free IRam : 7015
E2 out/firmware/rom0.bin
Checking for spiffs files
files directory exists. Creating out/firmware/spiff_rom.bin
Creating rom 'out/firmware/spiff_rom.bin' of size 0x10000 (65536) bytes.
Adding files in directory 'files'.
Added 'testfile.txt' to spiffs (40 bytes).
mike@mike-desktop > /opt/Sming/samples/Basic_rBoot $ make flashinit
Flash init data default and blank data.
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 erase_flash
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Erasing flash (this may take a while)...
Erase took 1.4 seconds
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 32m 0x3fc000 /opt/esp-open-sdk/sdk/bin/esp_init_data_default.bin 0x3fe000 /opt/esp-open-sdk/sdk/bin/blank.bin 0x100000 /opt/Sming/Sming/compiler/data/blankfs.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Writing 4096 @ 0x3fc000... 4096 (100 %)
Wrote 4096 bytes at 0x3fc000 in 0.4 seconds (89.7 kbit/s)...
Writing 4096 @ 0x3fe000... 4096 (100 %)
Wrote 4096 bytes at 0x3fe000 in 0.4 seconds (89.8 kbit/s)...
Writing 16384 @ 0x100000... 16384 (100 %)
Wrote 16384 bytes at 0x100000 in 1.4 seconds (91.7 kbit/s)...
Leaving...
mike@mike-desktop > /opt/Sming/samples/Basic_rBoot $ make flash
Killing Terminal to free /dev/ttyUSB0
/opt/Sming/Sming/Makefile-rboot.mk:634: recipe for target 'flash' failed
make: [flash] Getötet (ignoriert)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 32m 0x00000 out/firmware/rboot.bin 0x02000 out/firmware/rom0.bin 0x100000 out/firmware/spiff_rom.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0240
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.7 kbit/s)...
Writing 315392 @ 0x2000... 315392 (100 %)
Wrote 315392 bytes at 0x2000 in 27.3 seconds (92.3 kbit/s)...
Writing 65536 @ 0x100000... 65536 (100 %)
Wrote 65536 bytes at 0x100000 in 5.7 seconds (92.2 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Writing default boot config.
Booting rom 0.
��␃�␛�n�{��n|�␌d␌d$l ␃��s�l�$␓�␄␌␄$`␃��s�l�d␓�␄␌␌d`␂��;�d␌��␀�␄␄ld`␃��␒�'�␂'␇l{l�␑-�ver: 222378046 trying to mount spiffs at 0x00100000, length 65536
22378485 fs.start: size:64 Kb, offset:0x100000

22380403 mount res: 0

22442720 Station configuration was updated to: FRITZ!Box 6490 Cable

Currently running rom 0.
Type 'help' and press enter for instructions.

mode : sta(5c:cf:7f:3c:5c:0f)
add if0
scandone
no FRITZ!Box 6490 Cable found, reconnect after 1s
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with FRITZ!Box 6490 Cable, channel 11

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

```

Evgsurf, 2019-02-07:
I have the same problem with Basic_rBoot.

before I used  "make flashinit" system was able to connect to WiFi, but did not download and update on "ota" commans (it just out "Updating...." send request to rom0.bin and nothing more (did not hungs, just nothing more))

now I have the same situation like wrote another guys ...

P.S. I can restore system for WiFi by flashing another sample (HttpServer_WebSocket for exapmle) and flash again rBoot and system gets ip, but Updating does not work :(

andr2000, 2019-02-07:
I traced it down to spi_flash_erase_sector which hangs and makes the WDT fire.
I also tried to erase the same sector that OTA wants right before enabling WiFi and it works ok...
My 12E module was externally powered with 2A power supply, so not sure if power is an issue here.
flash{init|config}, erasing, writing all 4MiB with 0xff didn't help either.
@anakod @raburton do you by chance have any inputs on this?
The issue is open since Nov 2018 and we still suffer from this....

etmmahi, 2019-02-07:
Hi, I have already traced it down to the rboot library. The issue is that during the update of the flash sectors (erase or write) the interrupts (except flash interrupt) should be disabled. Maybe the faulty ISR do not have ICACHE_RAM_ATTR set and will crash because the flash is not accessible.

Updates to the file rboot-api.c (actually the patch files must be updated)

somewhere in the top of he file add:
```
#define FLASH_INT_MASK ((1 << ETS_SDIO_INUM) | (1 << ETS_GPIO_INUM) | (1 << ETS_UART_INUM) | (1 << ETS_FRC_TIMER1_INUM))
```

around line 70:

```
	spi_flash_read(BOOT_CONFIG_SECTOR * SECTOR_SIZE, (uint32*)((void*)buffer), SECTOR_SIZE);
	memcpy(buffer, conf, sizeof(rboot_config));
	ets_isr_mask(FLASH_INT_MASK);
	spi_flash_erase_sector(BOOT_CONFIG_SECTOR);
	spi_flash_write(BOOT_CONFIG_SECTOR * SECTOR_SIZE, (uint32*)((void*)buffer), SECTOR_SIZE);
	ets_isr_unmask(FLASH_INT_MASK);

```

in begin of function rboot_write_flash add:
```
	SpiFlashOpResult fop;
```

around line 160:

```
		// erase any additional sectors needed by this chunk
		lastsect = ((status->start_addr + len) - 1) / SECTOR_SIZE;
		while (lastsect > status->last_sector_erased) {
			status->last_sector_erased++;
			ets_isr_mask(FLASH_INT_MASK);
			spi_flash_erase_sector(status->last_sector_erased);
			ets_isr_unmask(FLASH_INT_MASK);
		}

		// write current chunk
		//os_printf("write addr: 0x%08x, len: 0x%04x\r\n", status->start_addr, len);
		ets_isr_mask(FLASH_INT_MASK);
		fop = spi_flash_write(status->start_addr, (uint32 *)((void*)buffer), len);
		ets_isr_unmask(FLASH_INT_MASK);
		if (fop == SPI_FLASH_RESULT_OK) {
```

I have tested this and it makes my OTA rock solid.

etmmahi, 2019-02-07:
diff version:

```

--- ./Sming/Sming/third-party/rboot/appcode/rboot-api.c	2017-02-20 11:54:45.879688993 +0100
+++ ./projects3/Basic_rBoot/rboot-api.c	2019-02-07 16:59:56.000000000 +0100
@@ -28,6 +28,8 @@
 extern "C" {
 #endif
 
+#define FLASH_INT_MASK ((1 << ETS_SDIO_INUM) | (1 << ETS_GPIO_INUM) | (1 << ETS_UART_INUM) | (1 << ETS_FRC_TIMER1_INUM))
+
 #if defined(BOOT_CONFIG_CHKSUM) || defined(BOOT_RTC_ENABLED)
 // calculate checksum for block of data
 // from start up to (but excluding) end
@@ -66,9 +68,11 @@
 	
 	spi_flash_read(BOOT_CONFIG_SECTOR * SECTOR_SIZE, (uint32*)((void*)buffer), SECTOR_SIZE);
 	memcpy(buffer, conf, sizeof(rboot_config));
+	ets_isr_mask(FLASH_INT_MASK);
 	spi_flash_erase_sector(BOOT_CONFIG_SECTOR);
 	//spi_flash_write(BOOT_CONFIG_SECTOR * SECTOR_SIZE, (uint32*)((void*)buffer), SECTOR_SIZE);
 	spi_flash_write(BOOT_CONFIG_SECTOR * SECTOR_SIZE, (uint32*)((void*)buffer), SECTOR_SIZE);
+	ets_isr_unmask(FLASH_INT_MASK);
 	
 	os_free(buffer);
 	return true;
@@ -120,7 +124,8 @@
 	bool ret = false;
 	uint8 *buffer;
 	int32 lastsect;
-	
+	SpiFlashOpResult fop;
+
 	if (data == NULL || len == 0) {
 		return true;
 	}
@@ -153,12 +158,17 @@
 		lastsect = ((status->start_addr + len) - 1) / SECTOR_SIZE;
 		while (lastsect > status->last_sector_erased) {
 			status->last_sector_erased++;
+			ets_isr_mask(FLASH_INT_MASK);
 			spi_flash_erase_sector(status->last_sector_erased);
+			ets_isr_unmask(FLASH_INT_MASK);
 		}
 
 		// write current chunk
 		//os_printf("write addr: 0x%08x, len: 0x%04x\r\n", status->start_addr, len);
-		if (spi_flash_write(status->start_addr, (uint32 *)((void*)buffer), len) == SPI_FLASH_RESULT_OK) {
+		ets_isr_mask(FLASH_INT_MASK);
+		fop = spi_flash_write(status->start_addr, (uint32 *)((void*)buffer), len);
+		ets_isr_unmask(FLASH_INT_MASK);
+		if (fop == SPI_FLASH_RESULT_OK) {
 			ret = true;
 			status->start_addr += len;
 		}
```

slaff, 2019-02-07:
@petrynchyn @andr2000 @Ph3r0x @Evgsurf Guys, can you check the suggested change and report if it is working for you?

petrynchyn, 2019-02-08:
OTA seems to work, but problem remained with connection WiFi after make flashinit

andr2000, 2019-02-08:
I can confirm what @petrynchyn says. OTA does work, but WiFi is still an issue.
Frankly, I am not surprised, as we workaround interrupts during OTA, but still WifiStation class wants to
write its settings to flash (auto-reconnect and credentials):
```
-                       WifiStation.config(CONFIG_WIFI_SSID, CONFIG_WIFI_PWD);
+                       WifiStation.config(CONFIG_WIFI_SSID, CONFIG_WIFI_PWD, false, false);
```
With the change above I effectively disable wifi_set_opmode and wifi_station_set_config both willing to write to flash (if I understood the SDK documentation correctly).
And this is at least, but not at last. As per my understanding any flash write/erase may cause an issue if interrupts are enabled. So, I would consider the above patch from @etmmahi (who did a really good job!!!) to be just a particular workaround and in order to fix the issue we do need understanding what is the real root cause for this.
IMO, it is likely as @etmmahi said, that interrupt handlers should be examined if they still reside in RAM, because flash erase/write effectively unmaps QSPI from the CPU which may lead to hang and WDT reset...

andr2000, 2019-02-08:
guys at Arduino finally managed to ["Remove interrupt disable around flash operations"](https://github.com/esp8266/Arduino/pull/5577)

> "All interrupt service routines are supposed to be in IRAM now, so there
> is no need to keep interrupts disabled while doing flash operations.
> Remove the IRQ disable/enable from the ESP.flash* methods."

So, indeed, there is something trying to run from flash in some interrupt handler....

PR #1610 (2019-02-11): Fix: `HardwareSerial` interrupt callback code not in IRAM
Supersedes #1604, fixes #1520 and possibly #1602

Lambda inherits code segment of enclosing statement, i.e. flash. Adding IRAM_ATTR doesn't work, even if it's separated out into a different statement. However, placing the entire `updateUartCallback` method into IRAM_ATTR _does_ work, but is wasteful of RAM. Therefore reverted to regular static callback handler.
closed 2019-02-11 by slaff