SmingHub/Sming #1523: HttpServer_WebSocket sample and broadcast before server.shutdown()

Evgsurf, 2018-11-15:
Hello.

I build and run HttpServer_WebSocket sample.
I changed index.html code and add text field and button to send data to server
This code added before output div
`<div><input id='sendtext'><input type="button" id="disconnect" onclick="doSend(document.getElementById('sendtext').value)" value="Send"/></div>
`
So, when I open serveral session (different tabs in browser) server sends messages to each session by broadcast command
```
	String message = "New friend arrived! Total: " + String(totalActiveSockets);
	socket.broadcast(message.c_str(), message.length());
```

`RESPONSE: New friend arrived! Total: 2`

But when I type **shutdown** message/command, only active session get message 
`The server is shutting down...`

by code 
```
		String message = "The server is shutting down...";
		socket.broadcast(message.c_str(), message.length());

```
Other sessions do not get this broadcasted message.

mikee47, 2018-11-15:
Sounds probable that the HTTP server is getting shut down before all messages can be sent. Try commenting out line 58 `server.shutdown();` to see if that fixes it. It would be better anyway to call `server.shutdown()` using a timer to delay by a few seconds.

mikee47, 2018-11-15:
Try replacing `server.shutdown();` with this:
```
	  auto timer = new SimpleTimer;
	  timer->setCallback([](void* timer) {
		delete static_cast<SimpleTimer*>(timer);
		server.shutdown();
	  }, timer);
	  timer->startMs(3000);
```

Evgsurf, 2018-11-15:
Of couse it does work correctly!
But hardcode 3 secs is not very good by me.

I guess, it will better to user another way

  ```
auto timer = new SimpleTimer;
	  timer->setCallback([](void* timer) {
                if (server. READY()) { // I am not sure what is function is neccessary to use
		      delete static_cast<SimpleTimer*>(timer);
		      server.shutdown();
                }
	  }, timer);
	  timer->startMs(100);
```

PR #1598 (2019-02-06): HttpServer_Websockets sample: wait 1 second before calling `server.shutdown()`
Closes #1523
closed 2019-02-06 by slaff