SmingHub/Sming #1525: Windows install problem Sming 3.7.0

kwis2, 2018-11-18:
As noticed in #1426 in Windows 7 64 bit, in new 3.7.0 version there is still problem with Sming install.
I have updated Sming with `choco install sming -y -force -source 'https://www.myget.org/F/sming/'` and tried to make. as result I received 

> make[1]: Entering directory `/c/tools/sming/Sming'
> C+ system/m_printf.cpp
> make[1]: *** [out/build/system/m_printf.o] Error 1
> make[1]: Leaving directory `/c/tools/sming/Sming'
> make: *** [reload] Error 2

and 

> cc1plus.exe has stopped

any solution?

slaff, 2018-11-19:
@kwis2 Rerun the make command with V=1
```
cd %SMING_HOME%
make V=1 
```

kwis2, 2018-11-19:
```
C:\tools\sming\Sming>make V=1
if [ 1 -eq 1 ]; then \
        make -C /c/tools/sming/Sming  RELOAD_MKFILE=0; \
    fi
make[1]: Entering directory `/c/tools/sming/Sming'
if [ 0 -eq 1 ]; then \
        make -C /c/tools/sming/Sming  RELOAD_MKFILE=0; \
    fi
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -Isystem -Isystem/helpers -IWiring -Iappinit -ISmingCore/ -ISmingCore/Data/ -ISmingCore/Data/Stream/ -ISmingCore/Network/ -IS
mingCore/Network/Http/ -ISmingCore/Network/Http/Websocket/ -ISmingCore/Platform/ -IServices/CommandProcessing/ -IServices/DateTime/ -IServices/FATFS/ -IServices/HexDump/ -IServices
/Profiling/ -IServices/SpifFS/ -IServices/WebHelpers/ -IServices/Yeelight/ -IServices/libb64/ -IServices/libemqtt/ -ILibraries/AM2321/ -ILibraries/APA102/ -ILibraries/Adafruit_BME2
80_Library/ -ILibraries/Adafruit_GFX/ -ILibraries/Adafruit_ILI9341/ -ILibraries/Adafruit_NeoPixel/ -ILibraries/Adafruit_PCD8544/ -ILibraries/Adafruit_SSD1306/ -ILibraries/Adafruit_
ST7735/ -ILibraries/Adafruit_Sensor/ -ILibraries/ArduCAM/ -ILibraries/ArduinoJson/ -ILibraries/BH1750FVI/ -ILibraries/BMP180/ -ILibraries/Bounce/ -ILibraries/CapacitiveSensor/ -ILi
braries/DFRobotDFPlayerMini/ -ILibraries/DHTesp/ -ILibraries/DS18S20/ -ILibraries/DS3232RTC/ -ILibraries/HMC5883L/ -ILibraries/I2Cdev/ -ILibraries/IR/ -ILibraries/ITEADLIB_Arduino_
Nextion/ -ILibraries/LiquidCrystal/ -ILibraries/MCP23017/ -ILibraries/MCP23S17/ -ILibraries/MFRC522/ -ILibraries/MMA_7455/ -ILibraries/Mirf/ -ILibraries/OneWire/ -ILibraries/RCSwit
ch/ -ILibraries/RF24/ -ILibraries/RingBufCPP/ -ILibraries/SDCard/ -ILibraries/SI7021/ -ILibraries/Servo/ -ILibraries/SparkFun_APDS9960/ -ILibraries/TFT_ILI9163C/ -ILibraries/TM1637
/ -ILibraries/Ultrasonic/ -ILibraries/WS2812/ -ILibraries/si4432/ -ILibraries/ArduinoJson/src/ -ILibraries/IR/src/ -Ithird-party/spiffs/src -Ithird-party/http-parser/ -Ithird-party
/ws_parser/ -Igdb -Isystem/include -Isystem/helpers/include -IWiring/include -Iappinit/include -ISmingCore//include -ISmingCore/Data//include -ISmingCore/Data/Stream//include -ISmi
ngCore/Network//include -ISmingCore/Network/Http//include -ISmingCore/Network/Http/Websocket//include -ISmingCore/Platform//include -IServices/CommandProcessing//include -IServices
/DateTime//include -IServices/FATFS//include -IServices/HexDump//include -IServices/Profiling//include -IServices/SpifFS//include -IServices/WebHelpers//include -IServices/Yeelight
//include -IServices/libb64//include -IServices/libemqtt//include -ILibraries/AM2321//include -ILibraries/APA102//include -ILibraries/Adafruit_BME280_Library//include -ILibraries/A
dafruit_GFX//include -ILibraries/Adafruit_ILI9341//include -ILibraries/Adafruit_NeoPixel//include -ILibraries/Adafruit_PCD8544//include -ILibraries/Adafruit_SSD1306//include -ILibr
aries/Adafruit_ST7735//include -ILibraries/Adafruit_Sensor//include -ILibraries/ArduCAM//include -ILibraries/ArduinoJson//include -ILibraries/BH1750FVI//include -ILibraries/BMP180/
/include -ILibraries/Bounce//include -ILibraries/CapacitiveSensor//include -ILibraries/DFRobotDFPlayerMini//include -ILibraries/DHTesp//include -ILibraries/DS18S20//include -ILibra
ries/DS3232RTC//include -ILibraries/HMC5883L//include -ILibraries/I2Cdev//include -ILibraries/IR//include -ILibraries/ITEADLIB_Arduino_Nextion//include -ILibraries/LiquidCrystal//i
nclude -ILibraries/MCP23017//include -ILibraries/MCP23S17//include -ILibraries/MFRC522//include -ILibraries/MMA_7455//include -ILibraries/Mirf//include -ILibraries/OneWire//include
 -ILibraries/RCSwitch//include -ILibraries/RF24//include -ILibraries/RingBufCPP//include -ILibraries/SDCard//include -ILibraries/SI7021//include -ILibraries/Servo//include -ILibrar
ies/SparkFun_APDS9960//include -ILibraries/TFT_ILI9163C//include -ILibraries/TM1637//include -ILibraries/Ultrasonic//include -ILibraries/WS2812//include -ILibraries/si4432//include
 -ILibraries/ArduinoJson/src//include -ILibraries/IR/src//include -Ithird-party/spiffs/src/include -Ithird-party/http-parser//include -Ithird-party/ws_parser//include -Igdb/include
 -Iinclude -Isystem/include -IWiring -ILibraries -ISmingCore -I/c/Espressif/ESP8266_SDK/../include -Ithird-party/rboot -Ithird-party/rboot/appcode -Ithird-party/spiffs/src -Ithird-
party/http-parser/ -Ithird-party/ws_parser/ -Ithird-party/esp-open-lwip/include -I/c/Espressif/ESP8266_SDK/include -I/c/Espressif/ESP8266_SDK/include/json -DSDK_PWM_PERIOD_COMPAT_M
ODE=1 -Wall -Wundef -Wpointer-arith -Wno-comment -Wl,-EL -nostdlib -mlongcalls -mtext-section-literals -finline-functions -fdata-sections -ffunction-sections -D__ets__ -DICACHE_FLA
SH -DUSE_OPTIMIZE_PRINTF -DARDUINO=106 -DCOM_SPEED_SERIAL=115200 -DENABLE_CMD_EXECUTOR=1 -DESP8266=1 -DSMING_INCLUDED=1  -Werror -Wno-sign-compare -Wno-parentheses -Wno-unused-vari
able -Wno-unused-but-set-variable -Wno-strict-aliasing -Wno-return-type -Wno-maybe-uninitialized -Os -g -DPROGMEM_L32="" -DCUST_FILE_BASE=m_printf -DDEBUG_VERBOSE_LEVEL=2 -DDEBUG_P
RINT_FILENAME_AND_LINE=0 -fno-rtti -fno-exceptions -std=c++11 -felide-constructors -Wno-reorder  -c system/m_printf.cpp -o out/build/system/m_printf.o
make[1]: *** [out/build/system/m_printf.o] Error 1
make[1]: Leaving directory `/c/tools/sming/Sming'
make: *** [reload] Error 2

C:\tools\sming\Sming>
```

slaff, 2018-11-19:
Hm.. I don't see the actual error that gcc reported?! Is `system/m_printf.cpp` the only file does not compile? Can you paste the output from the command below?
```
/c/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++  -v
```
 
Is that directory existing `/c/Espressif/ESP8266_SDK/include` ? If yes, what is the version of the SDK that you have in there?

kwis2, 2018-11-19:
Sming make failed with system pop-up titled `cc1plus.exe` nad message (in translation) `cc1plus.exe has stoped working`, With `search soution online` and `close program` so in log there is no message.

all dirs look ok. I asked google and there is problem with stack or so...

> C:\tools\sming\Sming>c:/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++  -v
> Using built-in specs.
> COLLECT_GCC=c:/Espressif/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++
> COLLECT_LTO_WRAPPER=c:/espressif/xtensa-lx106-elf/bin/../libexec/gcc/xtensa-lx106-elf/5.2.0/lto-wrapper.exe
> Target: xtensa-lx106-elf
> Configured with: ../configure --prefix=/d/Neo/crosstool/xtensa-lx106-elf --target=xtensa-lx106-elf --enable-multilib --disable-nls --disable-shared --disable-threads --with-gnu-as
> --with-gnu-ld --with-gmp=/d/Neo/crosstool/build/gmp --with-mpfr=/d/Neo/crosstool/build/mpfr --with-mpc=/d/Neo/crosstool/build/mpc --enable-languages=c,c++ --with-newlib --disable-l
> ibssp --disable-__cxa_atexit --enable-decimal-float=yes --enable-cxx-flags='-mlongcalls -mtext-section-literals'
> Thread model: single
> gcc version 5.2.0 (GCC)
> 
> C:\tools\sming\Sming>

slaff, 2018-11-19:
I can confirm the issue. The issue is caused when the compiler tries to use <algorithm>. But that fact that it builds on our CI system means that either we don't have the right compiler or there are some missing environment variables. When I have more time I will try to see how to solve this on Windows.

slaff, 2018-11-20:
@mikee47 Were you also using Windows for development? Are you experiencing the same problem as reported above?

mikee47, 2018-11-20:
@slaff Yes, but manually installed... Haven't mucked about with Choco

closed 2019-05-24 by kwis2