SmingHub/Sming #1527: AtClient sample

Evgsurf, 2018-11-20:
Hello, is some sample for ATClient ? I did not find .

slaff, 2018-11-20:
There is no AtClient sample but you are free to add one. From other projects I can give you the following code (incomplete) samples
```c++
void a6Callback(AtClient& client, Stream& source, char arrivedChar,
                uint16_t availableCharsCount)
{
    // Do something useful here...
}

void a20_example()
{
    AtSerial modem(&Serial);
    modem.send("ATE0");
    modem.send("AT+CAMSTART");

    modem.send(
        "AT+CAMCAP\r", "+CAMCAP:",
        (AtCallback)([&](AtClient& client, Stream& source, char arrivedChar,
                         uint16_t availableCharsCount) {
            printf("Ch: 0x%x, Av: %d\n", arrivedChar, availableCharsCount);
        }),
        NULL, 4000);

    AtCommand command;
    command.name = "AT+CAMRD\r";
    command.breakOnError = false;
    command.callback = a6Callback;
    modem.send(command);
   // ...
}
```

I have a PhotoBooth project which is gathering dust for quite some time. When I find the time I can try to revive it and add it here as a sample or as an application based on Sming Framework,

Evgsurf, 2018-11-20:
@slaff BTW your code is not Sming, because AtCommand struct has not field callback and name ...

Evgsurf, 2018-11-20:
@slaff The world has not any warranty!
I try to modify code, but looks like my mode neoway 590 is die. 
I need to get new one ....