SmingHub/Sming #1534: FTP not accessing file system for some ftp clients

riban-bw, 2018-11-24:
I have enabled FTPServer and can connect from a Linux ftp client. Commands such as 'ls' work fine. If I connect from a Windows or Cygwin ftp client the the connection seems to establish (username and password are requested then welcome message displayed). I can type 'help' and get the help message and type some other commands but file system commands such as 'ls' or 'put' show

> 220 OK
> 150 Connecting

then just hang.

slaff, 2018-11-26:
@riban-bw You should try to use passive mode and check. Also if the problem is still there, please, describe the steps to repeat it.

riban-bw, 2018-11-26:
I have just tried on a different Windows 10 machine and after it asked to set firewall rules it worked so it may be some security constraints on the Windows 7 machine that I am using.

closed 2018-11-26 by riban-bw