SmingHub/Sming #1535: Default NTP query interval to short

riban-bw, 2018-11-25:
The default NTP query interval is set to 30 seconds which puts a load on the NTP server which may lead to requests being blocked. (Some NTP servers block requests from clients that exceed certain limits. 30s seems a rather short default period. Should it be longer, like 3600 (one hour)?

slaff, 2018-11-26:
> Some NTP servers block requests from clients that exceed certain limits

Can you provide us with a document stating the best time for auto-refresh? Then we can quite easily change the default settings.

riban-bw, 2018-11-26:
Actually, I seem to remember we looked at this a couple of years ago and thought it would be sensible to have a fairly fast poll to start with and to reduce the poll rate once connected to allow NTP client to be updated promptly on start-up. Each user can do this by adding

> ntpClient.setAutoQueryInterval(SLOW_RATE);

to their ntp callback. 
It may be advantageous to allow the ntp client polling to be started and stopped which may allow us to trigger the first poll after we are connected to the network.

closed 2018-11-26 by riban-bw