SmingHub/Sming #1545: Refactoring of DateTime

slaff, 2018-12-07:
The following refactoring changes should be done:
- Move DateTime from Services to SmingCore.
- Rename the old methods according to the rules above.
- In the DateTime.h header added the old methods that should redirect to the new ones, plus mark them as deprecated. 
Example:

```C++
class DateTime {
public: 
    /** @brief  Parse a HTTP full date and set time and date
     *  @param  httpDate HTTP full date in RFC 1123 format, e.g. Sun, 06 Nov 1994 08:49:37 GMT
     *  @retval bool True on success
     *  @note   Also supports obsolete RFC 850 date format, e.g. Sunday, 06-Nov-94 08:49:37 GMT where 2 digit year represents range 1970-2069
     *  @note   GMT suffix is optional and is always assumed / ignored
     *
     *   @deprecated Use `fromHttpDate` instead
     */
     bool parseHttpDate(const String& httpDate)   __attribute__ ((deprecated))
     {
            return fromHttpDate(httpDate);
     }
}
```

riban-bw, 2018-12-09:
As discussed, you can allocate this to me if you want. I will pick it up over the next couple of weeks.

riban-bw, 2018-12-10:
Related to #1540.

riban-bw, 2018-12-10:
@slaff suggested adding:

format(const String& format), e.g.
```
DateTime d;
String httpDate = d.format("D, d M Y H:i:s GMT");
```
I wonder whether something more like strftime would be better, e.g.
```
DateTime d;
String httpDate = d.format("%a, %d %b %Y %H:%M:%S GMT");
```
If so, we may want to limit the implementation or add pre-processor conditional compilation blocks to avoid bloat.

I have done the refactoring (PR imminent) and will work up a strftime style _format_ function for review.

slaff, 2018-12-10:
> I wonder whether something more like strftime would be better, e.g.

Better not for the moment.

closed 2018-12-17 by slaff