SmingHub/Sming #1547: gdbstub (not esp-gdbstub) inappropriately named

riban-bw, 2018-12-09:
There is a third-party provided module called, "esp-gdbstub" which implements and interface with gdb debugger. There is also a module called, "gdbstub" which seems to catch exceptions and print some debug info to serial but does not implement a gdb interface. Naming this "gdbstub" and describing it within source code as, "A stub to make the ESP8266 debuggable by GDB over the serial port." is misleading and confusing. May I suggest this module be renamed and the description changed to be more appropriate?
Maybe this module should be called, "exception_handler" or similar. I note there is also code, "crash_handler" which may be appropriate to consolidate with this.

slaff, 2018-12-10:
> "gdb" which seems to catch exceptions and print some debug info to serial but does not implement a gdb interface. 

`stack_dumper` is probably a better name. But I am open to suggestions. Stack dumper should be activated only in non-RELEASE mode. 

`crash_handler` is something that is allowing user-based code to be executed after an exception was raised and do some minimal operations like save it to flash memory or just increase a number. the latter could be used in RELEASE and non-RELEASE mode.

slaff, 2019-04-10:
Fixed with the merge of PR #1655.

closed 2019-04-10 by slaff