SmingHub/Sming #1552: Basic_rBoot rom 2 wont start

Mike-Gre, 2018-12-12:
Hello everyone,

im struggling with the Basic_rBoot sample. I want to switch between 2 roms in the flash without spiffs on a 4MB flash. Im building the first rom (rom0.bin) with following Makefile settings:

```
COM_SPEED	= 115200
SPI_MODE = dio
RBOOT_ENABLED ?= 1
RBOOT_BIG_FLASH ?= 1
#RBOOT_TWO_ROMS  ?= 1
SPI_SIZE        ?= 4M
RBOOT_ROM_0     ?= rom0
RBOOT_LD_0      ?= rom0.ld
DISABLE_SPIFFS  = 1
```
and following rom0.ld:

`  irom0_0_seg :                         org = 0x40202010, len = (1M - 0x2010)
`

i added two lines to the init method:
```
	WifiAccessPoint.enable(false);

	WifiStation.enable(true);
	WifiStation.config(WIFI_SSID, WIFI_PWD);

	Serial.printf("\r\nCurrently running rom %d.\r\n", slot);
```

make output:

```
OC out/build/libmain2.a
make -C /opt/Sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0 RBOOT_SILENT=0
make[1]: Verzeichnis „/opt/Sming/Sming/third-party/rboot“ wird betreten
CC rboot-stage2a.c
LD /opt/Sming/samples/Basic_rBoot/out/build/rboot-stage2a.elf
E2 /opt/Sming/samples/Basic_rBoot/out/build/rboot-hex2a.h
CC rboot.c
LD /opt/Sming/samples/Basic_rBoot/out/build/rboot.elf
E2 /opt/Sming/samples/Basic_rBoot/out/firmware/rboot.bin
make[1]: Verzeichnis „/opt/Sming/Sming/third-party/rboot“ wird verlassen
CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c
CC /opt/Sming/Sming/appspecific/rboot/overrides.c
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8371|     881
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFE8CF8|    2424
       bss|      Uninitialized Data (RAM)|    3FFE8CF8|    3FFEEFB0|   25272
      text|            Cached Code (IRAM)|    40100000|    40105CD3|   23763
irom0_text|           Uncached Code (SPI)|    40202010|    40244F5C|  274252
Total Used RAM : 28577
Free RAM : 53343
Free IRam : 9023
E2 out/firmware/rom0.bin
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
```

after the make i upload the rom with make flash:

```
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
Killing Terminal to free /dev/ttyUSB0
/opt/Sming/Sming/Makefile-rboot.mk:634: recipe for target 'flash' failed
make: [flash] Getötet (ignoriert)
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 32m 0x00000 out/firmware/rboot.bin 0x02000 out/firmware/rom0.bin
esptool.py v1.2
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0240
Writing 4096 @ 0x0... 4096 (100 %)
Wrote 4096 bytes at 0x0 in 0.4 seconds (89.7 kbit/s)...
Writing 303104 @ 0x2000... 303104 (100 %)
Wrote 303104 bytes at 0x2000 in 26.3 seconds (92.3 kbit/s)...
Leaving...
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Writing default boot config.
Booting rom 0.
��;�d�d␒�␄␄�d`␃��{�d�l␛�␌␄␄l ␃��{�l␄��␀�␌␌dl`␃sd�␑-�ver: 117778931 spiffs disabled

Currently running rom 0.
Type 'help' and press enter for instructions.

mode : sta(5c:cf:7f:3c:5c:0f)
add if0
f r0, scandone
no FRITZ!Box 6490 Cable found, reconnect after 1s
reconnect
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with FRITZ!Box 6490 Cable, channel 11
dhcp client start...
ip:192.168.178.43,mask:255.255.255.0,gw:192.168.178.1
ip: 192.168.178.43 mac: 5c:cf:7f:3c:5c:0f

SDK: v1.5.4(baaeaebb)
Free Heap: 50920
CPU Frequency: 80 MHz
System Chip ID: 3c5c0f
SPI Flash ID: 1640c8
28867055 Count: 2
28867201 ROM 0: 8192
28868455 ROM 1: 2105344
28870537 ROM 2: 0
28872115 GPIO ROM: 0
pm open,type:2 0

```

After that i am changing the Makefile to make the rom1.bin like this:

RBOOT_ROM_0     ?= rom1
RBOOT_LD_0      ?= rom1.ld

with rom1.ld:

`  irom0_0_seg :                         org = 0x40402000, len = (1M - 0x2010)
`
make clean

make:

```
OC out/build/libmain2.a
make -C /opt/Sming/Sming/third-party/rboot RBOOT_GPIO_ENABLED=0 RBOOT_SILENT=0
make[1]: Verzeichnis „/opt/Sming/Sming/third-party/rboot“ wird betreten
CC rboot-stage2a.c
LD /opt/Sming/samples/Basic_rBoot/out/build/rboot-stage2a.elf
E2 /opt/Sming/samples/Basic_rBoot/out/build/rboot-hex2a.h
CC rboot.c
LD /opt/Sming/samples/Basic_rBoot/out/build/rboot.elf
E2 /opt/Sming/samples/Basic_rBoot/out/firmware/rboot.bin
make[1]: Verzeichnis „/opt/Sming/Sming/third-party/rboot“ wird verlassen
CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-bigflash.c
CC /opt/Sming/Sming/third-party/rboot/appcode/rboot-api.c
CC /opt/Sming/Sming/appspecific/rboot/overrides.c
C+ app/application.cpp
AR out/build/app_app.a
LD out/build/app_0.out
   Section|                   Description| Start (hex)|   End (hex)|Used space
------------------------------------------------------------------------------
      data|        Initialized Data (RAM)|    3FFE8000|    3FFE8371|     881
    rodata|           ReadOnly Data (RAM)|    3FFE8380|    3FFE8CF8|    2424
       bss|      Uninitialized Data (RAM)|    3FFE8CF8|    3FFEEFB0|   25272
      text|            Cached Code (IRAM)|    40100000|    40105CD3|   23763
irom0_text|           Uncached Code (SPI)|    40402000|    40444F4C|  274252
Total Used RAM : 28577
Free RAM : 53343
Free IRam : 9023
E2 out/firmware/rom1.bin
(!) Spiffs support disabled. Remove 'DISABLE_SPIFFS' make argument to enable spiffs.
```

and flashing rom1.bin with:

`/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 -b 115200 write_flash -ff 40m -fm dio -fs 32m 0x202000 out/firmware/rom1.bin`

thats my output when trying to switch:

```
Swapping from rom 0 to rom 1.
Restarting...

state: 5 -> 0 (0)
rm 0
del if0
usl

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```

btw sorry fot that long text

slaff, 2019-02-27:
@Ph3r0x Can you try with the latest code in `develop` to see if the issue you were experiencing is already fixed?

Ph3r0x, 2019-02-27:
This issue can be closed, this issue is fixed.

closed 2019-02-27 by Ph3r0x