SmingHub/Sming #1559: cmath not compilable after upgrade

ematt, 2018-12-21:
Hello,

I had a working setup until I have decided to upgrade to 3.7.0. There seems to be a problem related to the <cmath> header that prevents my application code to compile.

### Expectation
The application code should compile without errors

### Problem
The application code does not compile

### Update process:
1. `cd $ESP_HOME`
2. Update esp-open-sdk as stated in the [documentaion](https://github.com/pfalcon/esp-open-sdk#pulling-updates)
`$ make clean
$ git pull
$ git submodule sync
$ git submodule update --init`
3. `make VENDOR_SDK=1.5.4 STANDALONE=y`
4. `cd $SMING_HOME`
5. `git pull`
6. `make`

The host OS is an Ubuntu x86_64

make output: [https://pastebin.com/VMmvEB34](https://pastebin.com/VMmvEB34)

slaff, 2018-12-24:
@ematt  In order to be able to help you you should provide a sample app where the problem is reproducible.

ematt, 2019-02-22:
Apparently I got the wrong inspiration to name one of my files "math.h". It seems that the errors are caused by it. So this is not a bug. Thank you!

closed 2019-02-22 by slaff