SmingHub/Sming #1560: Servo sample wdt reset

kmihaylov, 2018-12-23:
Hello. I'm trying to use the Servo driver, but I'm getting wdt reset.
I tried flashinit with no luck.
The only change in the Basic_Servo example I made is to put
`SPI_SIZE`        ?= 4M
in Makefile-user.mk

I'm using the develop branch, commit cc7af2263e41e24c12c047878c95f13489d5c81c
I'll be glad if someone tries to build the example in order to confirm whether the issue is dedicated to my build environment or not.

Thank you!

mikee47, 2018-12-23:
The sample does appear to be missing 'Serial.begin(SERIAL_BAUD_RATE);' at the start of `init()`, but not convinced that would case wdt resets...

kmihaylov, 2018-12-23:
I'll try it tomorrow despite that I have the SERIAL_BAUD_RATE option in my project.

I'm seeing that the Servo driver uses HardwareTimer. Just tried Basic_HwPWM and it works. I set the frequency and period as per Servo's specs and I believe HwPWM can be a replacement option. I didn't test it with AP/Sta enabled, but I hope it works. Currently the duty cycle isn't precise much (I'm using the euqation provided in the nonos reference).

Informatic, 2019-05-17:
Hey, experiencing the same issue while directly using HardwareTimer in nonmaskable mode (I need 5-bit 50bps uart for a project).

It looks like it broke somewhere between 0b646a584e9c44f71f45e5fc431145e355619107 (2018/08/28; not 100% sure, but it looks like it was the last time my magic uart code worked...) and current release.

mikee47, 2019-05-17:
Have you tried running it under the debugger ?

```
make clean
make ENABLE_GDB=1
```

Informatic, 2019-05-17:
```trap '' INT && /home/informatic/Projects/esp-open-sdk-2/xtensa-lx106-elf/bin/xtensa-lx106-elf-gdb -x /home/informatic/Projects/Sming/Sming/gdb/gdbcmds -b 115200 -ex "target remote /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0"
GNU gdb (crosstool-NG crosstool-ng-1.22.0-55-gecfc19a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-lx106-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
add symbol table from file "out/build/bootrom.elf" at
        .text_addr = 0x40000000

Welcome to SMING!
Type 'c' (continue) to run application

Remote debugging using /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
gdbstub_init () at /home/informatic/Projects/Sming/Sming/gdb/gdbstub.cpp:914
914                     gdb_do_break();
(gdb) c
Continuing.
Init Basic Servo Sample

Program received signal SIGTRAP, Trace/breakpoint trap.
gdbstub_init () at /home/informatic/Projects/Sming/Sming/gdb/gdbstub.cpp:914
914                     gdb_do_break();
(gdb) bt
#0  gdbstub_init () at /home/informatic/Projects/Sming/Sming/gdb/gdbstub.cpp:914
#1  0x4020798a in gdb_init () at /home/informatic/Projects/Sming/Sming/appspecific/gdb/gdb_hooks.cpp:231
#2  0x40232571 in user_init () at appinit/user_main.cpp:37
#3  0x40206d61 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) c
Continuing.
Init Basic Servo Sample
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
gdbstub_init () at /home/informatic/Projects/Sming/Sming/gdb/gdbstub.cpp:914
914                     gdb_do_break();
(gdb) bt
#0  gdbstub_init () at /home/informatic/Projects/Sming/Sming/gdb/gdbstub.cpp:914
#1  0x4020798a in gdb_init () at /home/informatic/Projects/Sming/Sming/appspecific/gdb/gdb_hooks.cpp:231
#2  0x40232571 in user_init () at appinit/user_main.cpp:37
#3  0x40206d61 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

Seems like nothing interesting, after wdt reset I just end up on initial breakpoint, doing Ctrl-c during hang does nothing (and then wdt kicks in as expected with the same result)

I'll try stepping into `attach()` and will report what I get... Otherwise, I'll just do a bisect between these versions.

Informatic, 2019-05-17:
As (somewhat) expected, ESP halts after this line: https://github.com/SmingHub/Sming/blob/develop/Sming/Arch/Esp8266/Core/HardwareTimer.cpp#L117

I'll do the bisect... later today.

mikee47, 2019-05-17:
I wonder if the problem is if the timer interval is too short - try setting the interval first?

` `
__`RTC_REG_WRITE(FRC1_LOAD_ADDRESS, usToTimerTicks(interval));`__
`TM1_EDGE_INT_ENABLE();`
`ETS_FRC1_INTR_ENABLE();`
` `
~~`RTC_REG_WRITE(FRC1_LOAD_ADDRESS, usToTimerTicks(interval));`~~

mikee47, 2019-10-05:
@kmihaylov @Informatic Did you guys make any progress on this?

kmihaylov, 2019-10-07:
HwPWM did the job for me, therefore I didn't test with HardwareTimer. I can give it a try later today.