SmingHub/Sming #1561: HwPWM duty calculation

kmihaylov, 2018-12-24:
I'm using the HwPWM to drive a servo motor.

Setting the frequency with 
```
HW_pwm.setPeriod( 20000 ); 
```
gives an excellent frequency of 50 Hz (20000 microseconds).
According to the NONOS reference the equation to calculate the maximum duty cycle is
```
period*1000/45
```
So to get a 1 ms duty cycle the calculation should be duty*(period*1000/45)/period

1000*(20000*1000/45)/20000

This calculation doesn't give the expected duty. If i change 45 to 40, however, I can see a nice timing on the scope.
Respectiverly with values 25000 for 1 ms and 50000 for 2 ms (which sounds pretty round).

Otherwise the values should be 22222 and 44444.

Do you have any clue where this mismatch comes from?
I'm using the develop branch, commit cc7af2263e41e24c12c047878c95f13489d5c81c
I'm using SDK 2.0

mikee47, 2018-12-27:
Does this help? It's the library Sming uses for HW PWM:
https://github.com/StefanBruens/ESP8266_new_pwm/

kmihaylov, 2018-12-28:
Yes, it helps. Thank you. I'm closing the issue.

Although I knew that Sming uses different library I didn't notice the differences. Maybe I'll put some changes to the comments in HardwarePWM.cpp to make the function arguments more clear (for the auto-generated documentation) and prepare a PR.

All the best!

----
In Sming/third-party/pwm/README.md

> To have full compatibility with the SDK, you have to set the
SDK_PWM_PERIOD_COMPAT_MODE define to 1. If set, the code will use 1us for PWM
period and 40ns for the duty cycle. E.g. 10% duty cycle at 1kHz is set by a
period value of 1000 and a duty cycle value of 2500, full duty at 25000 and
above

closed 2018-12-28 by kmihaylov