SmingHub/Sming #1562: Custom headers are not sent correctly

slaff, 2018-12-26:
Setting custom header results in invalid HTTP request.
```
	HttpHeaders headers;
	headers[F("X-Powered-By")] = _F("Sming");				  // Use text for non-standard field names
	getRequest->setHeaders(headers);
```

[image]

As can be seen from the snapshot above the header name is missing.

slaff, 2018-12-26:
@mikee47 Can you take a look at that issue, please?

mikee47, 2018-12-26:
@slaff Just had a quick look, problem is with `HttpHeaders::setMultiple()` which needs over-riding. I'll submit a PR later today.

mikee47, 2018-12-26:
@slaff This should sort the problem, can you check?

PR #1563 (2018-12-26): Add HttpHeaders::setMultiple() method
Using inherited HashMap version doesn't work. Fixes #1562
commit 30f705e7 (2018-12-26): Add HttpHeaders::setMultiple() method (#1563)

* Add HttpHeaders::setMultiple() method, using inherited HashMap version doesn't work. Fixes #1562
* Make HashMap private ancestor
closed 2018-12-26 by slaff