SmingHub/Sming #1570: Recommended SDK? esp-open-sdk with SDK 2.0.0?

verybadsoldier, 2019-01-06:
I have a basic question:
Which is the recommended SDK you would propose for production use?

In current `esp-open-sdk` they default to SDK `2.1.0` but I get compile errors with that:
```
vbs@ubuntu:~/Projects/esp_rgbww_firmware/Sming/Sming$ make
make[1]: Entering directory '/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming'
make[2]: Entering directory '/home/vbs/Projects/esp_rgbww_firmware/Sming/Sming/third-party/esp-open-lwip'
/home/vbs/Projects/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/home/vbs/Projects/esp-open-sdk/sdk/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/def.o lwip/core/def.c
/home/vbs/Projects/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -I/home/vbs/Projects/esp-open-sdk/sdk/include -Wl,-EL -mlongcalls -mtext-section-literals -I../../system/include -I../../Wiring   -c -o lwip/core/dhcp.o lwip/core/dhcp.c
In file included from include/lwip/stats.h:37:0,
                 from lwip/core/dhcp.c:73:
lwip/core/dhcp.c: In function 'dhcp_start':
lwip/core/dhcp.c:676:52: error: too many arguments to function 'pvPortMalloc'
     dhcp = (struct dhcp *)mem_malloc(sizeof(struct dhcp));
                                                    ^
include/lwip/mem.h:58:36: note: in definition of macro 'mem_malloc'
 #define mem_malloc(s) pvPortMalloc(s, "", __LINE__, false)
```

It compiles fine when setting SDK to `2.0.0`. So I conclude `2.1.0` is not supported.

As support for `3.0` is experimental currently I guess the recommended way is to use current `esp-open-sdk` and set SDK version back to `2.0.0`?

Just to be sure: `esp-open-sdk` is the way to go and It is not meant to use vanilla `espressif ESP8266 NONOS SDK`, right?

Thanks alot.

mikee47, 2019-01-06:
I would definitely focus on SDK 3.0, keeping the codebase as current as possible. I don't see why it shouldn't be fit for 'production use', subject to your own pre-release testing of course!

slaff, 2019-01-07:
> As support for 3.0 is experimental currently I guess the recommended way is to use current esp-open-sdk and set SDK version back to 2.0.0?

The esp-open-sdk can be used just to build the compiler for you.  After that it would be best if you use SDK 3.0.0 and instruct Sming (as described in the [README](https://github.com/SmingHub/Sming/blob/develop/Readme.md)) to compile against it. The Sming build mechanism will do the needed patching and stitching.

verybadsoldier, 2019-01-07:
Thanks, I got Sming compiling now with SDK 3.0.0. It failed for me with every other version when using SDK from `esp-open-sdk` expect for `2.0.0`.

Your comment was the missing hint for me (I am sure it is written somewhere but I could not find it):
`Sming` actually brings an embedded SDK (which is `3.0.0` in the directory `third-party`). If you want to use it then you have to set env variable `SDK_BASE`.
In that case it does not matter which SDK you specified when you compiled `esp-open-sdk` cause it won't be used anyway.

I really tried hard to grab some extra commits into `esp-open-sdk` to get `3.0.0` there. That worked but I only got compile errors in `Sming`.

closed 2019-01-12 by verybadsoldier

icodk, 2019-02-23:
@verybadsoldier I am facing the same issues with the SDK ; building successfully  with 2.0.0 and 1.5.4 but failing with with  2.2.1 and 3.0.0 .
Could you please list the steps you used to get it working with SDK 3.0.0 and the latest Sming version ?
I am using the following dir structure according to the readme:
```
/opt/
     esp-open-sdk/
     Sming/
```

verybadsoldier, 2019-02-23:
For me it worked after setting this env variable:
```
SDK_BASE=/home/vbs/Projects/Sming/Sming/third-party/ESP8266_NONOS_SDK/
```

icodk, 2019-02-23:
@verybadsoldier Thanks,  This is regarding sming but what did you do to get  esp-open-sdk  to 3.0.0 ?
Did you build it with make VENDOR_SDK=3.0.0 STANDALONE=n ?
or mak  VENDOR_SDK=3.0.0 STANDALONE=y ?
or didn't build it at all just downloaded sming after the esp_open-sdk  , set SDK_BASE (as you mentioned above) and run make in Sming ?

verybadsoldier, 2019-02-23:
The SDK comes with Sming. You should find it if you take a look at the path I posted.

icodk, 2019-02-24:
@verybadsoldier Your tip probably got me over this step and I managed to build sming without errors
including Basic_Blink(see below all my steps after the error) .
I am not sure if  `VENDOR_SDK=3.0.0`  in step 5 is doing anything positive.
However Basic_Blink will not run and instead came with the following error:
Again any help will be appreciated.
```
python -m serial.tools.miniterm /dev/ttyUSB0 115200 
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
emapg
map 0 err
system_partition_table_regist: failed
size_map = 0
partition[0]: 1, 0x00000000, 0x00001000
partition[1]: 100, 0x00001000, 0x00001000
partition[2]: 101, 0x00002000, 0x000fe000
partition[3]: 4, 0x0007b000, 0x00001000
partition[4]: 5, 0x0007c000, 0x00001000
partition[5]: 6, 0x0007d000, 0x00003000
```
Steps for building the sdk and sming:
```

1. Make a dir /opt
    mkdir -p /opt/

    
    
2. Change owner to ico:
    chown -R ico:ico /opt
    cd /opt/

3. Get the esp-open-sdk tool chain
     git clone --recursive https://github.com/pfalcon/esp-open-sdk.git

    cd /opt/esp-open-sdk     
4. Build the toolchain only
     cd /opt/esp-open-sdk     
     make STANDALONE=n

When finished:     
        Xtensa toolchain is built, to use it:
        export PATH=/opt/esp-open-sdk/xtensa-lx106-elf/bin:$PATH
        Espressif ESP8266 SDK is installed. Toolchain contains only Open Source components
        
        To link external proprietary libraries add:
        xtensa-lx106-elf-gcc -I/opt/esp-open-sdk/sdk/include -L/opt/esp-open-sdk/sdk/lib
5. Get Sming
    cd /opt
    git clone https://github.com/SmingHub/Sming.git
     
6. Add the following var: ( also to the ~.profile file )
    export ESP_HOME=/opt/esp-open-sdk
    export SMING_HOME=/opt/Sming/Sming
    export SDK_BASE=/opt/Sming/Sming/third-party/ESP8266_NONOS_SDK/
    export PATH=$PATH:$ESP_HOME/esptool2

7. Build sming
        cd /opt/Sming/Sming
        make ENABLE_SSL=1 ENABLE_CMD_EXECUTOR=0 VENDOR_SDK=3.0.0
    
```

verybadsoldier, 2019-02-24:
Did you try to double-check `SPI_SPIZE`? You have to set it according to your hardware.

icodk, 2019-02-24:
@verybadsoldier This helped. Thanks