SmingHub/Sming #1573: SDK RF Config & OTA Update

verybadsoldier, 2019-01-14:
One question please gentlemen:

The ESP has to be initialized with the RF configuration data which is specific to the used SDK, as far as I understand it (`make flashinit`).
So when switching SDKs (e.g. from `2.0.0` to `3.0.0`) this RF configuration data has to be updated in the device.

I assume that when using OTA update then this RF configuration is not updated accordingly (at least I would not understand how). So am I right that it is not possible to use OTA update to update from a firmware that uses SDK `2.0.0` to a firmware that uses SDK `3.0.0`? So in that case before installing the new firmware the device has to be initialized with the new RF config?

verybadsoldier, 2019-01-14:
Ok thank you. But for now am I correct that switching SDK versions using OTA update is not supported and therefore this transition has to be done using the serial bootloader?

slaff, 2019-01-14:
> SDK versions using OTA update is not supported

Yes, that is correct. I would say that it will most probably work if you have old RF and use it with apps compiled with SDK 1.5, 2.0 and 3.0 but there is no guarantee.