SmingHub/Sming #1583: {StationClass,AccessPointClass}::getMAC - undocumented change in behaviour

q3k, 2019-01-28:
https://github.com/SmingHub/Sming/commit/483f3bfb34bea8cf1247c73be016455673c0e42d changed the behaviour of the getMAC methods on `AccessPointClass` and `StationClass` - they now return a MAC address as `00:11:22:aa:bb:cc`, while previously they returned it as `001122aabbcc`. This is a sudden API change on the `master` branch, seemingly undocumented.

In my opinion, the behaviour change should be rolled back, as it breaks applications that rely on a fixed format of the MAC for identification. I have discovered this due to an application breaking in this exact way.

Preferably, an alternative/extended method would return a MAC object that can be rendered as a string in different formats...

mikee47, 2019-01-28:
Agreed, have submitted a proposed solution.

slaff, 2019-01-29:
@q3k Can you check if the fix is working for you? The updated code is **only** in the `develop` branch until we release a new version.

PR #1584 (2019-01-29): Revert to default MAC string format for `getMAC` method...
...in `AccessPointClass` and `StationClass`

Fixes #1583, so by default reverts to previous behaviour, but supports alternative formats.
closed 2019-01-29 by slaff