SmingHub/Sming #1602: Basic_rBoot skip updating

Evgsurf, 2019-02-07:
Hello.
What I did today
1. Install ubuntu 18.04 x64
2. Install esp-open-sdk
3. Install Sming
4. Install esptool
5. Install esptool2
6. Build Sming
7. Build&Flash&Test Basic_Blink
8. Build&Flash&Test HttpServer_AJAX
9. Build&Flash&Test HttpServer_WebSocket
there are no any problems at any steps 
All samples were build with 
SPI_MODE = dio
SPI_SIZE    = 4M
SPI_SPEED = 80

10. Build&Flash Basic_rBoot - passed
```

Booting rom 0.
��;�d�d␒�␄␌�$`␃��s�l�$␓�␌␌␄l ␃��{�d␄��␀�␌␄ld`␂{l�DK ve22145004 trying to mount spiffs at 0x00100000, length 65536
22145310 fs.start: size:64 Kb, offset:0x100000

22146900 mount res: 0

Currently running rom 0.
Type 'help' and press enter for instructions.

mode : sta(80:7d:3a:6e:3e:3b)
add if0
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f -180, scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2

connected with start, channel 1
dhcp client start...
cnt 
ip:172.16.1.153,mask:255.255.254.0,gw:172.16.0.1
pm open,type:2 0

```
>info
```
SDK: v1.5.4(baaeaebb)
Free Heap: 48888
CPU Frequency: 80 MHz
System Chip ID: 6e3e3b
SPI Flash ID: 164020
38870640 Count: 2
38870751 ROM 0: 8192
38872212 ROM 1: 2105344
38874292 ROM 2: 0
38875859 GPIO ROM: 0

```

>ip

`ip: 172.16.1.153 mac: 807d3a6e3e3b
`

>help
```
available commands:
  help - display this message
  ip - show current ip address
  connect - connect to wifi
  restart - restart the esp8266
  switch - switch to the other rom and reboot
  ota - perform ota update, switch rom and reboot
  info - show esp8266 info
  ls - list files in spiffs
  cat - show first file in spiffs
```

>ota
`Updating...
`

>help
```
available commands:
  help - display this message
  ip - show current ip address
  connect - connect to wifi
  restart - restart the esp8266
  switch - switch to the other rom and reboot
  ota - perform ota update, switch rom and reboot
  info - show esp8266 info
  ls - list files in spiffs
  cat - show first file in spiffs
```
ip&info&help do work well, but when I call ota, system call to http server (I use SimpleHTTPServer) and looks like hang. But system woks, i able to with commans like ip ip&info&help ....

I tested with 2 boards.
I have not idea what is wronf and how to find out the problem ....

slaff, 2019-02-08:
@Evgsurf Can you check that PR #1604 ?

Evgsurf, 2019-02-12:
Gyus , I updated Sming from git by (git pull)
Rebuild all, and still have a problem.
Here is video https://youtu.be/_oyLnV4gNl0

mikee47, 2019-02-12:
Try rebuilding Sming and app with DEBUG_VERBOSE_LEVEL=3

Evgsurf, 2019-02-12:
@mikee47 in Makefile-user.mk ? I tied no effect
I tied to add in application.cpp #define DEBUG_VERBOSE_LEVEL 3 - no effect
Just Updating ... :(

Evgsurf, 2019-02-12:
Oh I tied but forget to rebuild Sming it self.

I get result
```

Updating...
47529719 Download file:
    (0) http://172.16.1.183:8080/rom0.bin -> 202000
47530644 Creating new httpConnection
47530953 HttpConnection::connect: TCP state: -1, isStarted: 0, isActive: 0
47535875 HttpConnection::connecting ...
47539411 +TCP connection
47541498 connect to: 172.16.1.183
47545072 TCP connect result: 0
47547187 Download file:
    (0) http://172.16.1.183:8080/spiff_rom.bin -> 300000
47554807 HttpConnection::connect: TCP state: 2, isStarted: 1, isActive: 1
47566559 TCP connected
47566706 TCP connected
47566830 HttpConnection::onReadyToSendData: waitingQueue.count: 2
47570641 Storing 24 bytes in stream
47573815 realloc 24 -> 173
47575878 Storing 20 bytes in stream
47579033 Storing 19 bytes in stream
47582124 Storing 2 bytes in stream
47585171 TCP onReadyToSendData: 0
47588261 TCP Written: 65, Available: 65, isFinished: 1, PushCount: 1 [TcpBuf: 2715]
47595391 TcpClient stream finished
47603434 TCP sent: 65
47603619 HttpConnection::onReadyToSendData: waitingQueue.count: 1
47606284 Storing 29 bytes in stream
47609421 realloc 29 -> 178
47611535 Storing 20 bytes in stream
47614689 Storing 19 bytes in stream
47617780 Storing 2 bytes in stream
47620807 TCP onReadyToSendData: 2
47623869 TCP Written: 70, Available: 70, isFinished: 1, PushCount: 1 [TcpBuf: 2710]
47631037 TcpClient stream finished
47635221 TCP received: 17 bytes
47636877 HttpConnection::onReadyToSendData: waitingQueue.count: 0
47642576 Nothing in the waiting queue
47645871 TCP onReadyToSendData: 1
47650507 The headers are complete
47695173 rboot_write_flash: item.size: 1200
47695327 TCP received: 1390 bytes
47695471 HttpConnection::onReadyToSendData: waitingQueue.count: 0
47696661 Nothing in the waiting queue
47699965 TCP onReadyToSendData: 1
47706938 rboot_write_flash: item.size: 2590
47707144 TCP received: 1390 bytes
47709712 HttpConnection::onReadyToSendData: waitingQueue.count: 0
47715402 Nothing in the waiting queue
47718713 TCP onReadyToSendData: 1
47725164 rboot_write_flash: item.size: 3980
47725482 TCP received: 1390 bytes
47728447 HttpConnection::onReadyToSendData: waitingQueue.count: 0
47734135 Nothing in the waiting queue
47737444 TCP onReadyToSendData: 1
47782098 rboot_write_flash: item.size: 5353
47782293 TCP received: 1373 bytes
47782429 HttpConnection::onReadyToSendData: waitingQueue.count: 0
47783626 Nothing in the waiting queue
47786942 TCP onReadyToSendData: 1
47790897 TCP sent: 70
47791819 HttpConnection::onReadyToSendData: waitingQueue.count: 0
47797510 Nothing in the waiting queue
47800820 TCP onReadyToSendData: 2
47806582 rboot_write_flash: item.size: 6743
47807588 TCP received: 1390 bytes
47810551 HttpConnection::onReadyToSendData: waitingQueue.count: 0
[... 867 lines cut ...]
54830541 rboot_write_flash: item.size: 290303
54831212 TCP received: 1390 bytes
54834177 HttpConnection::onReadyToSendData: waitingQueue.count: 0
54839868 Nothing in the waiting queue
54843180 TCP onReadyToSendData: 1
54895421 rboot_write_flash: item.size: 291693
54898370 rboot_write_flash: item.size: 293083
54901238 rboot_write_flash: item.size: 294473
54940787 rboot_write_flash: item.size: 295863
54940936 TCP received: 5560 bytes
54941075 HttpConnection::onReadyToSendData: waitingQueue.count: 0
54942444 Nothing in the waiting queue
54945756 TCP onReadyToSendData: 1
54955550 rboot_write_flash: item.size: 297253
54955771 TCP received: 1390 bytes
54955909 HttpConnection::onReadyToSendData: waitingQueue.count: 0
54961356 Nothing in the waiting queue
54964668 TCP onReadyToSendData: 1
54971092 rboot_write_flash: item.size: 298643
54971609 TCP received: 1390 bytes
54974573 HttpConnection::onReadyToSendData: waitingQueue.count: 0
54980263 Nothing in the waiting queue
54983573 TCP onReadyToSendData: 1
54987529 TCP connection error: -9
```

mikee47, 2019-02-12:
Error -9 is ' connection reset', any help?

Evgsurf, 2019-02-12:
@mikee47 I understand that problem is connection, after you say about DEBUG, and install apache (Maybe problem in Python HTTP)

Now I have 
```
Firmware download finished!
41462684  - item: 0, addr: 202000, len: 311728 bytes
41467275  - item: 1, addr: 300000, len: 65536 bytes
In callback...
Firmware updated, rebooting to rom 1...
41517508 Firmware updated.
41517784 TCP connected
41517897 HttpConnection::onReadyToSendData: waitingQueue.count: 0
41518040 Nothing in the waiting queue
41519841 TCP onReadyToSendData: 0
41523065 TCP received: 1071 bytes
41525762 HttpConnection::onReadyToSendData: waitingQueue.count: 0
41531453 Nothing in the waiting queue
41534757 TCP onReadyToSendData: 1
state: 5 -> 0 (0)
rm 0
pm close 7
41541061 TCP connection error: -8
del if0
usl

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Evgsurf, 2019-02-12:
System hangs on "wdt reset", but if I restart board manually, I can switch between ROM 0 and ROM 1

mikee47, 2019-02-12:
In sample_rBoot, in OtaUpdate_CallBack, could you change `System.restart()` to `System.restart(1000)` and see if that helps?

mikee47, 2019-02-12:
Also, see #1611, slaff mentioned about setting GPIO#0 high

Evgsurf, 2019-02-12:
@mikee47 System.restart(1000) does not help. Also I have no external devices now at board esp8266.

mikee47, 2019-02-12:
I suspect it's SDK related - try building on SDK 3.0 see #1470 - I use this all the time without issue.

Evgsurf, 2019-02-12:
@mikee47  I will try, but it will take a time, because I am newbie at ESP.
BTW the problem is not Restart, system hangs on rboot_set_current_rom.
Thank You very much!

mikee47, 2019-02-12:
@Evgsurf You're welcome. Apologies, though, the SDK 3.0 post is a little out of date. To build with SDK 3.0 support all you need to do is:

```
cd $SMING_HOME
export SDK_BASE="$SMING_HOME/third-party/ESP8266_NONOS_SDK"
cd $SMING_HOME
make dist-clean
make
```

mikee47, 2019-02-12:
https://bbs.espressif.com/viewtopic.php?t=574

mikee47, 2019-02-12:
Something else to try. After programming, do a hard reset _before_ running an update.

Evgsurf, 2019-02-12:
Ohh, so many ways :)
I tried SDK 3.0 by https://github.com/SmingHub/Sming/issues/1439#issuecomment-423799397 and your patch https://github.com/SmingHub/Sming/pull/1470#issuecomment-429315841
Fatal Exception 28 with long DUMP

```
TCP connection error: -8
del if0
usl
Firmware updated.

***** Fatal exception 28
pc=0x40243e8a sp=0x3ffff9f0 excvaddr=0x00000074
ps=0x00000030 sar=0x00000008 vpri=0x40100fa2
r00: 0x40243e80=1076117120 r01: 0x3ffff9f0=1073740272 r02: 0x00000000=         0 

.......
```

Evgsurf, 2019-02-12:
> Something else to try. After programming, do a hard reset _before_ running an update.

This is "handmade" solution, but works :)

mikee47, 2019-02-12:
Yaaay! The above two posts should help to explain what's happening - the `System.restart()` remembers the previous boot mode (i.e. programming mode) and uses that. The hard reset clears this. I've never had an issue with this but I guess it depends on hardware versions, etc.

Evgsurf, 2019-02-12:
@mikee47 I have tested 3 esp8266 https://github.com/pfalcon/esp-open-sdk/issues/330#issuecomment-417303260 There is not difference. All three work well after handmade reboot after load new room from PC. And all three do not work without first habdmade reboot.

Evgsurf, 2019-02-12:
> Error -9 is ' connection reset', any help?

I guess in is not bad if method void rBootHttpUpdate::start() will be int rBootHttpUpdate::start() and return an error code (Error -9 is ' connection reset )

mikee47, 2019-02-12:
> I guess in is not bad if method `void rBootHttpUpdate::start()` will be `int rBootHttpUpdate::start()` and return an error code (Error -9 is 'connection reset')
I agree, needs some way of reporting failure back. In this instance the error occurs a long time after `start()` has finished; Sming is event-driven so doesn't hang around, kicks off the transfer and leaves it running.

@slaff I think `rBootHttpUpdate` needs some attention here. In general probably call the user-supplied callback on failure in `start()` or on a TCP error. It may need some work on `HttpClient` as well - I have a PR lined up for that so could take a look.

`Or C++ Exception ?` 
LOL. Nice idea! Sming doesn't use exceptions though, exception frames eat RAM for one thing.

Evgsurf, 2019-02-12:
I guess it is will very good, if it will be possible to have property eventHandler at TcpConnection (as I see this is "main" class) to where set *func(int event,int val)

for example

void eventHandler(int event,int val)
{
Serial.printf("Event %d value %d\r\n", event,val);
}

....

if(otaUpdater)
	delete otaUpdater;
otaUpdater = new rBootHttpUpdate();

otaUpdater->setTcpEventHandler(eventHandler);
...

I will try if will not die in class structure :)

slaff, 2019-02-20:
> `Or C++ Exception ?`
LOL. Nice idea! Sming doesn't use exceptions though, exception frames eat RAM for one thing.

Forget it for now. We try to run on an embedded device.

> I think `rBootHttpUpdate` needs some attention here. In general probably call the user-supplied callback on failure in `start()` or on a TCP error. It may need some work on `HttpClient` as well - I have a PR lined up for that so could take a look.

Have you done  some changes related to this?

mikee47, 2019-02-20:
@slaff I didn't make any functional changes to `rBootHttpUpdate` (or `HttpClient`) and haven't specifically looked at this issue yet.